botletics/SIM7000-LTE-Shield #2: QMI

Ben1902, 2018-01-23:
Hello!

Thank you for sharing your valuable insight on this new, rather new chip! I am looking to use this SIM7000(E) as a modem on my OpenWRT router. I bought the chip on a mPCIE board, and its listed with PPP capability. But when installed, it lists a QMI dev, but I can not get it to work. I can not find the promised "Linux" driver for this chip either. So my question is: Do you know if the SIM7000 supports a QMI modem interface (with time, and kernel support), or is this some common simmcom interface, and not supported? Can not find proper documentation any place.. Thanks if you have any info to share. I`m also new to PPP, so I would be happy if anyone know of a working chatscript for SIM7000 and using CAT-M1.

Thanks,
Best regards

Benjamin Kruger

botletics, 2018-01-23:
Hi there,

First of all, I am not a Linux guy. According to the [SIM7000A page](http://simcomm2m.com/En/module/detail.aspx?id=173) the modules does indeed support PPP and according to the AT command manual you can use AT+CGDCONT to achieve that. I personally have not tried it. As for QMI modem interface, I really don't know but I have asked a SIMCom engineer about it and will get back to you if I find an answer.

Best,
Tim

Ben1902, 2018-03-08:
Hello again,

Now I have had some time to test, after my cellular provider finally has fixed CAT-M1 access where I live. Now the QMI interface is working(!), but from what I tried of your AT commands I found elsewhere in an article, it now runs on GPRS. Do you have any idea on how to disable GPRS and force CAT-M1? 

BR
Ben

botletics, 2018-03-08:
Hi there,

Check out the "AT+CNMP" command. You can do "AT+CMNP=38" to use LTE only.

Hope that helps!

sabareesan007, 2018-03-08:
Hi all,

@Ben1902 further you can config the band for CAT-M or NB-IOT using "AT+CBANDCFG". In our country we use band 8 for NB-IOT. 
@botletics  thank you for sharing your valuable work.

Cheers from Sri Lanka

botletics, 2018-03-26:
Seems like this has been resolved.

closed 2018-03-26 by botletics