botletics/SIM7000-LTE-Shield #5: Library compatible with SIM5320A?

dahdo1m, 2018-02-09:
Hi all,

I am trying to make a diy 3G cellular data logger for my school project, I'm using a TinySine 3G/GSM/GPS module with an Arduino UNO and an AT&T sim card. I keep getting:

AT+HTTPTERM
ERROR
AT+HTTPINIT
ERROR

I used the Adafruit library and this one uploaded here, but keep getting the same results. Could you please help?
 Thank you,

botletics, 2018-02-09:
It should be compatible, although I know HTTPS uses different commands. Did you first enable GPRS before posting? That's critical for the HTTP to work properly.

As a side note, I actually wrote a function for the SIM5320 HTTPS in my fork of the Adafruit FONA library in a different repo. The function is called "postData3G" in the .cpp file if you want to take a look.

botletics, 2018-02-09:
Sorry, actually now I recall that the SIM5320 doesn't use the same HTTP commands as the other modules. It uses the "AT+CHTTP____" commands instead. I would have to update my library to get it working.

dahdo1m, 2018-02-09:
I have been trying to send AT+CHTTPS___ using the pass through tunnel, this is what I'm getting:

AT+CHTTPSSTART
OK
AT+CHTTPSOPSE="https://dweet.io/dweet/for/dahdo1m?temp=24.6",443

"no response"!

dahdo1m, 2018-02-09:
I have been reading the SIM5320A AT Commands List, the HTTPS commands are quite different, and I feel that I don't fully understand what they are supposed to do.

botletics, 2018-02-09:
I haven't tried pass through tunnel. I'll update my library shortly with an HTTP/HTTPS function for you to try shortly. Will keep you updated. However, I can't actually test it out since I don't have a SIM5320A anymore.

botletics, 2018-02-09:
OK, I just updated the library and the "LTE_Demo" sketch. Just follow these steps:
- Delete the existing Adafruit FONA library you have in your libraries folder (path is Documents/Arduino/libraries by default)
- Follow [this wiki](https://github.com/botletics/SIM7000-LTE-Shield/wiki/Arduino-IDE-Setup) to get my version of the library installed
- Open the "LTE_Demo" example sketch but uncomment the "SIMCOM_3G" line instead of the default "SIMCOM_LTE" line at the very top of the code
- Upload and run the code. You should see the menu appear.
- Enter the "G" command in your serial monitor to enable 3G
- Enter the "3" command to send data to dweet.io via HTTP or HTTPS. Check out the function definition in the sketch on how to use the function. By default it sends via HTTPS.

Let me know if it works!
Thanks

dahdo1m, 2018-02-11:
Hi Tim,

I tried it today, the code you wrote does send the following AT Command CHTTPSSTART and CHTTPSOPSE. The response for the first is OK, but the response for the second is ERROR, just similar to what I described earlier when I used the pass through tunnel. Are we using proper AT Commands? Do you think that SIM5320A is not suitable for connecting to the internet and sending data? The reason I'm asking is because the AT Commands list for SIM5320A does not show HTTP commands, and the HTTPS commands feel insufficient to send data through the 3G network.

botletics, 2018-02-11:
The SIM5320A definitely works with HTTP and HTTPS, I tried it before. It just uses different commands than the other 2G modules.
Let me ask again:
- Did you first enable GPRS with the "G" command in the LTE_Demo sketch?
- Are you even connected to the network? Use "n" to check network registration

dahdo1m, 2018-02-11:
I enabled it by pressing G, or by pressing g then G, the network was available, signal appeared to be OK, and network was registered.

botletics, 2018-02-11:
Hmmmm that's strange. I'll investigate this when I get a chance.

botletics, 2018-02-15:
By the way, did you set the right APN for your SIM card? By default I use "hologram" but you need to change it for your specific SIM card if you're not using a Hologram.io SIM card.

dahdo1m, 2018-02-15:
I used the APN for AT&T "m2m.com.attz"

dahdo1m, 2018-02-15:
I have an AT&T trial sim card

botletics, 2018-02-16:
Please verify that it's sending the following AT command exactly as the line below:

AT+CHTTPSOPSE="www.dweet.io",443,2

botletics, 2018-02-16:
If it is and it's failing, then go to the LTE_Demo sketch and search for this line:

if (!fona.postData3G("www.dweet.io", 443, "HTTPS", URL)) // Server, port, connection type, URL

and use good 'ole HTTP instead and see what happens:

if (!fona.postData3G("www.dweet.io", 80, "HTTP", URL)) // Server, port, connection type, URL

Also try getting rid of "www" and see if that works. I'm pretty darn sure this worked with "www" when I tested it on SIM5320A.

dahdo1m, 2018-02-20:
Hi Timothy,

I tried doing this today I kept getting the response: ERROR, whenever it sends the "AT+CHTTPSOPSE" command

botletics, 2018-02-20:
Could you please take a screenshot of the serial monitor output? Thanks!

dahdo1m, 2018-02-20:
[image]

dahdo1m, 2018-02-20:
I tried changing it to "HTTP" and 80,2 , and removing "www." also, all gave the same response

botletics, 2018-02-20:
Looks like you have some weird characters showing up at the end of the "CHTTPSOPSE" command. Does it do that every time?

dahdo1m, 2018-02-20:
Only after I changed it

botletics, 2018-02-20:
Doesn't look like you were entering the port 80 correctly. You should be copying/pasting this line instead:

if (!fona.postData3G("www.dweet.io", 80, "HTTP", URL)) // Server, port, connection type, URL

the "2" means HTTPS and is interpreted inside the function, so you shouldn't input that. In this case it will interpret the "HTTP" as "1" and not a "2". Try that and take another screenshot please!

dahdo1m, 2018-02-20:
[image]

botletics, 2018-02-20:
Ahh OK, must be a library thing then. Will look into that.

botletics, 2018-02-20:
OK, just uploaded a new Adafruit_FONA.cpp file in the repo. Download that and replace the existing one in your Arduino libraries folder, restart the IDE, and re-upload and see if that fixed the weird characters.

dahdo1m, 2018-02-21:
It worked :)

botletics, 2018-02-21:
Awesome! You were able to post data via HTTPS as well?

closed 2018-02-21 by botletics

almagroos, 2018-07-12:
Hi,
I need your help,
I am trying to sen data my web page using Adafruit 3G FONA SIM5320A

I got the same error. Please see the attached picture.
[image]

botletics, 2018-07-12:
Sorry, currently I don't have a SIM5320A to test personally, but people say the library works with the same AT command sequence.

almagroos, 2018-07-12:
Thank you for your replay.

I downloaded the folder "code" and I followed all the instructions that you posted in your page.
I am not sure if I used the right Adafruit_FONA.CPP because the code cannot recognize the fona.post3G function.

any suggestions my project has been suspended.

botletics, 2018-07-12:
You might have an old version of the library. When did you download the code folder?

almagroos, 2018-07-12:
yesterday.

botletics, 2018-07-12:
Can you please copy the error message here? It compiles just fine for me.

botletics, 2018-07-12:
And did you set the line

`#define SIMCOM_3G // SIM5320A/E`

near the top of the code?

almagroos, 2018-07-12:
I did that

botletics, 2018-07-12:
Can you please copy the error message? You also need to delete any existing Adafruit FONA library inside your Arduino libraries folder (default for Windows is Documents/Arduino/libraries)

almagroos, 2018-07-12:
if (!fona.postData3G("www.dweet.io", 80, "HTTP", URL)) // Server, port, connection type, URL

C:\Program Files (x86)\Arduino\libraries\Adafruit_FONA\Adafruit_FONA.h:151:101: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]

   boolean postData(const char *server, uint16_t port, const char *connType, char *URL, char *body = "");

                                                                                                     ^

C:\Program Files (x86)\Arduino\libraries\Adafruit_FONA\Adafruit_FONA.cpp: In member function 'boolean Adafruit_FONA::postData(const char*, const char*, char*, const char*)':

C:\Program Files (x86)\Arduino\libraries\Adafruit_FONA\Adafruit_FONA.cpp:1552:51: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]

   if (! sendCheckReply(dataBuff, "DOWNLOAD", 10000))

                                                   ^

C:\Program Files (x86)\Arduino\libraries\Adafruit_FONA\Adafruit_FONA.cpp: In member function 'boolean Adafruit_FONA::postData(const char*, uint16_t, const char*, char*, char*)':

C:\Program Files (x86)\Arduino\libraries\Adafruit_FONA\Adafruit_FONA.cpp:1659:42: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]

   if (! sendCheckReply(auxStr, ">", 10000))

                                          ^

C:\Users\magro\AppData\Local\Temp\ccFiS6UQ.ltrans0.ltrans.o: In function `loop':

ccFiS6UQ.ltrans0.o:(.text+0x1e66): undefined reference to `Adafruit_FONA_3G::postData3G(char const*, unsigned int, char const*, char*)'

collect2.exe: error: ld returned 1 exit status

exit status 1
Error compiling for board Arduino/Genuino Mega or Mega 2560.

botletics, 2018-07-12:
Please see my previous comment and delete any old Adafruit FONA libraries that you may have installed before installing mine.

almagroos, 2018-07-12:
I did and I will do it again now

almagroos, 2018-07-12:
I got the same error message

botletics, 2018-07-12:
Also, if it doesn't even compile, how did you get it to run the code? You showed it running the LTE_Demo already...

almagroos, 2018-07-12:
When I use  if (!fona.postData("www.dweet.io", 80, "HTTP", URL)), it is working but the as in message below

FONA> G
	---> AT+CGATT=1
	<--- OK
	---> AT+CGSOCKCONT=1,"IP","m2m.com.attz"
	<--- OK
	---> AT+CIPMODE=1
	<--- OK
	---> AT+NETOPEN=,,1
	<--- ERROR
Failed to turn on
FONA>

almagroos, 2018-07-12:
it works when I use fona.postData() instead of ona.postData3G()

botletics, 2018-07-12:
Please use the updated version of the examples as well, you have an old version. The new LTE_Demo does not have the "postData3G()" function.

https://github.com/botletics/SIM7000-LTE-Shield/blob/master/Code/examples/LTE_Demo/LTE_Demo.ino

almagroos, 2018-07-12:
Should I change the port to 80 or not

almagroos, 2018-07-12:
Same error

FONA> n
	---> AT+CREG?
	<--- +CREG: 0,1
Network status 1: Registered (home)
FONA> 2
	---> AT+HTTPTERM
	<--- ERROR
	---> AT+HTTPINIT
	<--- ERROR
Failed to complete HTTP GET...
FONA>

botletics, 2018-07-12:
For the SIM5320 you need to use the "3" command, not "2". You can try port 80 and "HTTP" but I doubt that would make any difference. Sorry, I don't know what's going on with your setup.

almagroos, 2018-07-12:
Do you mean this line
#define SIMCOM_3G // SIM5320A/E

botletics, 2018-07-12:
In the post above you used the "2" command in the LTE_Demo. You should use "3" instead. But yes, you should also use the SIMCOM_3G define line.

almagroos, 2018-07-12:
is there any error in this line of code 

 if (!fona.postData("www.aelmagrous.com", 80, "HTTP", "GET /SensorDemo/add_data2.php?dataRow=50 HTTP/1.1")) // Server, port, connection type, URL
          Serial.println(F("Failed to complete HTTP/HTTPS request..."));

botletics, 2018-07-12:
You should include at least the host name, otherwise how does it know where to go?

` HTTP/1.1\r\nHost: dweet.io\r\n\r\n`

Sorry, Github is not the place to debug other people's code. If it doesn't work with the code I provide I will look into it, but once you change it I really don't have the time.

rene6262, 2018-07-24:
Hi, ive been following your steps but no matter what .cpp or .h i use i get lots of compilation errors...
I am using ESP32 and Sim5320A, this is what i get when trying to compile from your files.  My files work fine for HTTP posting but now i want to use HTTPS.

-------------------------------------------------------------------------------------------------------------------

Archiving built core (caching) in: C:\Users\rene6\AppData\Local\Temp\arduino_cache_949788\core\core_espressif_esp32_featheresp32_FlashFreq_80,UploadSpeed_921600,DebugLevel_none_5cb0df2a6ba85453e296ce37c363884e.a
sketch\F7A8X6BI6WOTYGJ.ino.cpp.o:(.literal._Z5setupv+0x48): undefined reference to `Adafruit_FONA::setNetworkSettings(__FlashStringHelper const*, __FlashStringHelper const*, __FlashStringHelper const*)'

sketch\F7A8X6BI6WOTYGJ.ino.cpp.o:(.literal._Z9setupGPRSv+0x1c): undefined reference to `Adafruit_FONA::postData(char const*, unsigned short, char const*, char*, char*)'

sketch\F7A8X6BI6WOTYGJ.ino.cpp.o: In function `setup()':

C:\Users\rene6\Desktop\Rene\TAB\F7A8X6BI6WOTYGJ/F7A8X6BI6WOTYGJ.ino:617: undefined reference to `Adafruit_FONA::setNetworkSettings(__FlashStringHelper const*, __FlashStringHelper const*, __FlashStringHelper const*)'

-------------------------------------------------------------------------------------------------------------------
Thanks

botletics, 2018-07-24:
Please use the .cpp and .h files in this repo as well as the example sketches and you shouldn't have any problems compiling for SIM5320 (3G).

botletics, 2018-07-26:
Oh, you're using ESP32? Sorry I haven't explored that much. I'm currently really busy but will try this when I get a chance.

rene6262, 2018-07-27:
Im not sure if the issue is from ESP32 or not, this is what i get in my output logs:

C:\Users\rene6\Documents\Arduino\libraries\Adafruit_FONA\Adafruit_FONA.cpp:1585:9: error: prototype for 'boolean Adafruit_FONA::postData(const char*, uint16_t, const char*, char*)' does not match any in class 'Adafruit_FONA'

 boolean Adafruit_FONA::postData(const char *server, uint16_t port, const char *connType, char *URL) {

         ^

In file included from C:\Users\rene6\Documents\Arduino\libraries\Adafruit_FONA\Adafruit_FONA.cpp:19:0:

C:\Users\rene6\Documents\Arduino\libraries\Adafruit_FONA\Adafruit_FONA.h:151:11: error: candidates are: boolean Adafruit_FONA::postData(const char*, uint16_t, const char*, char*, char*)

   boolean postData(const char *server, uint16_t port, const char *connType, char *URL, char *body = "");

           ^

C:\Users\rene6\Documents\Arduino\libraries\Adafruit_FONA\Adafruit_FONA.cpp:1507:9: error:                 boolean Adafruit_FONA::postData(const char*, const char*, char*, const char*)

 boolean Adafruit_FONA::postData(const char *request_type, const char *URL, char *body, const char *token) {

         ^

botletics, 2018-07-27:
First of all, could you please provide your exact code so I can debug? If you compile for Arduino Uno, Mega, Leonardo, etc. it should work just fine.

rene6262, 2018-07-27:
Sure, i am trying to use your LTE demo code, the only changes i made are software serial to hardware serial.  Here is the changes ive done, the rest is your code:

#include <HardwareSerial.h>
HardwareSerial Serial2(2);
HardwareSerial *fonaSS = &Serial2;

while (!Serial);
//  ss.begin(GPSBaud); // begin the GPS serial connection
Serial.begin(115200);
Serial.println(F("Initializing....(May take 3 seconds)"));
//fonaSerial->begin(115200);
Serial2.write("AT+CDIPR=115200"); // Set Baud to 115200
delay(500);
Serial2.begin(115200);
delay(500);
if (! fona.begin(*fonaSS)) {
  Serial.println(F("Couldn't find FONA"));
  pinMode(FONA_RST, OUTPUT);    //set a pin mode on the FONA_RST
  int attempt = 0;                              // a counter to check the number of times the fona has been reset
  while (! fona.begin(*fonaSS) && attempt != 5) {   //try to reset the FONA 5 times before giving up
    Serial.print("Resetting FONA...Attempt");
    Serial.println(++attempt);
    digitalWrite(FONA_RST, LOW);                  //set the FONA_RST pin low (Reset the fona)
  }
}
digitalWrite(FONA_RST , HIGH);       //pull FONA_RST  back to high

rene6262, 2018-07-27:
and with the Uno i get this library error:

In file included from C:\Users\rene6\Documents\Arduino\libraries\Adafruit_FONA\includes/platform/FONAPlatStd.h:33:0,

                 from C:\Users\rene6\Documents\Arduino\libraries\Adafruit_FONA\includes/platform/FONAPlatform.h:40,

                 from C:\Users\rene6\Documents\Arduino\libraries\Adafruit_FONA\Adafruit_FONA.h:22,

                 from C:\Users\rene6\Documents\Arduino\libraries\Adafruit_FONA\Adafruit_FONA.cpp:19:

C:\Users\rene6\Documents\Arduino\libraries\Adafruit_FONA\includes/platform/Arduino.h:34:31: fatal error: freertos/FreeRTOS.h: No such file or directory

 #include "freertos/FreeRTOS.h"

                               ^

botletics, 2018-07-27:
Using the LTE_Demo code I have on Github compiles with "ESP32 Dev Board" with the hardware serial change:

`// We default to using software serial. If you want to use hardware serial
// (because softserial isnt supported) comment out the following three lines 
// and uncomment the HardwareSerial line
//#include <SoftwareSerial.h>
//SoftwareSerial fonaSS = SoftwareSerial(FONA_TX, FONA_RX);
//// Use the following line for ESP8266
////SoftwareSerial fonaSS = SoftwareSerial(FONA_TX, FONA_RX, false, 256); // TX, RX, inverted logic, buffer size
//
//SoftwareSerial *fonaSerial = &fonaSS;

// Hardware serial is also possible!
//HardwareSerial *fonaSerial = &Serial1;

// For ESP32 hardware serial use these lines instead
#include <HardwareSerial.h>
HardwareSerial fonaSS(1);`

rene6262, 2018-07-28:
The issues im getting is calling the PostData command, it says const  *char to char is -fpermissive or something in that manor.

[29 more replies]