botletics/SIM7000-LTE-Shield #6: Program memory requires more please reduce

muzmmilkhan, 2018-02-12:
Sir good evening 
I run the IoT example everything is great but the Arduino Uno r3 has 32kb of ROM and 2kb of RAM and the IoT example requires 80% of ROM and 81% of RAM so that I get a problem because I want to embade the impact sensor to detect the accident but because of memory I can't embade
Please reduce down the memory requirement by that I can connect extra sensor and improve my project
Sir if it possible than try please thanks

botletics, 2018-02-12:
Hi, I am currently very busy with other things but will try look into this when I get a chance.

muzmmilkhan, 2018-02-14:
Ok thanks sir I will wait.

botletics, 2018-02-14:
Sorry, I have just been extremely busy trying to get my own work done. I will try to get to this when I have a chance but please first try to resolve the issue. Maybe you can solve the issue by removing unneeded code, etc.

botletics, 2018-02-15:
Hi, I just compiled the IoT_Example sketch for the various methods and only the PROTOCOL_MQTT_AIO causes such a large sketch (around 80% ROM/RAM. It's due to the Adafruit MQTT libraries and I don't think there's any way around that. However, the other options compile at around 65% ROM/RAM.

muzmmilkhan, 2018-02-20:
Ok thanks sir
You can close this issue

closed 2018-02-20 by botletics