botletics/SIM7000-LTE-Shield #8: GPS/GNS for SIM7000

toddsandercock, 2018-02-16:
Hi,

SIM7000 units use GNS AT commands as per the manual. I know AT+CGPSPWR? mysteriously works on the 7000 at the moment but it might be worth changing enableGPS to this:

`boolean Adafruit_FONA::enableGPS(boolean onoff) {
  uint16_t state;

  // first check if its already on or off

  if (_type == FONA808_V2 || _type == FONA_LTE_A || _type == FONA_LTE_C || _type == FONA_LTE_E) {
    if (! sendParseReply(F("AT+CGNSPWR?"), F("+CGNSPWR: "), &state) )
      return false;
  } else {
    if (! sendParseReply(F("AT+CGPSPWR?"), F("+CGPSPWR: "), &state))
      return false;
  }

  if (onoff && !state) {
    if (_type == FONA808_V2 || _type == FONA_LTE_A || _type == FONA_LTE_C || _type == FONA_LTE_E) {
      if (! sendCheckReply(F("AT+CGNSPWR=1"), ok_reply))  // try GNS command
	return false;
    } else {
      if (! sendCheckReply(F("AT+CGPSPWR=1"), ok_reply))
	return false;
    }
  } else if (!onoff && state) {
    if (_type == FONA808_V2 || _type == FONA_LTE_A || _type == FONA_LTE_C || _type == FONA_LTE_E) {
      if (! sendCheckReply(F("AT+CGNSPWR=0"), ok_reply)) // try GNS command
	return false;
    } else {
      if (! sendCheckReply(F("AT+CGPSPWR=0"), ok_reply))
	return false;
    }
  }
  return true;
}`

botletics, 2018-02-16:
Thanks for the comment! Yea, I noticed that too. Actually "AT+CREG" mysteriously works as well but "technically" it's supposed to be "AT+CGREG" instead. I think this was SIMCom's attempt to make it easy to migrate from the other 2G and 3G modules but I'm not sure. I'll go ahead and change it for good measure though.
Thanks!

toddsandercock, 2018-02-16:
AT+CGPSOUT I don't think works on the sim7000 for NMEA enable. AT+CGNSTST is the one.

`boolean Adafruit_FONA::enableGPSNMEA(uint8_t i) {
  if (_type == FONA808_V2 || _type == FONA_LTE_A || _type == FONA_LTE_C || _type == FONA_LTE_E) {
    if (i)
      return sendCheckReply(F("AT+CGNSTST=1"), ok_reply);
    else
      return sendCheckReply(F("AT+CGNSTST=0"), ok_reply);
  } else {
	char sendbuff[15] = "AT+CGPSOUT=000";
	sendbuff[11] = (i / 100) + '0';
	i %= 100;
	sendbuff[12] = (i / 10) + '0';
	i %= 10;
	sendbuff[13] = i + '0';
    return sendCheckReply(sendbuff, ok_reply, 2000);
  }
}`

botletics, 2018-02-16:
Sorry, that line of code is literally one line  :)

toddsandercock, 2018-02-16:
Yeah just edited it

botletics, 2018-02-16:
But I see what you mean, I changed that as well, testing now!

botletics, 2018-02-16:
And yup, "AT+CGNSPWR" works, so we're good there. Testing the NMEA command.

botletics, 2018-02-16:
The "AT+CGNSTST" errors out if I try to set it either way but maybe that was fixed in a later firmware version. I'm currently testing on 1351B01. When I get my next production batch of boards I'll grab one off the line and test it :)

Thanks again!

toddsandercock, 2018-02-16:
Hmmmm interesting. I too get the same error

+CGNSCFG=1 works but it only puts the NMEA to the NMEA port via USB.

botletics, 2018-02-16:
Sorry, I can't tell, but are you testing this with my shield or some other SIM7000 device? Just curious so I know if we're on the same page.

toddsandercock, 2018-02-16:
At the moment I just have a SIM7000E just plugged in via USB. Have been
reading your code to understand how the board works. Makes it much easier
to understand the manual.

Want to connect the module to an ESP32 when it finally rocks up. I think
Chinese new year might push out a few things.

toddsandercock, 2018-02-16:
Will your new board have the SIM7000G?

botletics, 2018-02-16:
Once the SIM7000G comes out and I get my hands on it yes, I'll have that version available for sale as well.

toddsandercock, 2018-02-16:
Sweet I will buy a couple of them when you get them. Any code I come up
with to optimise the library for CAT-M1 and NB-IOT I will send you. The
module is really cool and has some great features.

botletics, 2018-02-16:
Thanks! Check out my Indiegogo campaign or Twitter for the latest updates. I'll definitely post about the SIM7000G when I get a hold of it.

botletics, 2018-09-20:
I can confirm that AT+CGNSTST now works with firmware version 1351B02SIM7000G. Use AT+CGNSTST=1 to enable NMEA to AT port, and AT+CGNSTST=0 to disable.

closed 2018-09-20 by botletics