botletics/SIM7000-LTE-Shield #9: How to connecting to IBM Bluemix mqtt Broker

muzmmilkhan, 2018-02-21:
Hello Sir
i hope you Know about IBM Bluemix and also The node red sir i am trying to connect to the IBM Bluemix through MQTT and i have sing in to bluemix and create a new device successfully and i get some device parameter like divice id ... etc and i google about it and i get a programe to connect ibm through a esp8266 and eathernet .but i want to connect   GSM so that i wand your guidance for connect my GSM Model to ibm through your iot_example .
my Quation is that how to define below in iot_example
#define ORG "ic01na"
#define DEVICE_TYPE "ArduinoGsm"
#define DEVICE_ID "demoDevice"
#define TOKEN "JZp5ytS*_)H!tZf*EM"
char server[] = ORG ".messaging.internetofthings.ibmcloud.com";
char authMethod[] = "use-token-auth";
char token[] = TOKEN;
char clientId[] = "d:" ORG ":" DEVICE_TYPE ":" DEVICE_ID;
const char eventTopic[] = "iot-2/evt/status/fmt/json";
const char cmdTopic[] = "iot-2/cmd/led/fmt/json";
and connect to ibm bluemix mqtt broker.i hope you have time if you are busy thank ok give replay after you free please 
thanks

botletics, 2018-03-04:
Sorry, this isn't a place to ask questions about how to do things. I provide these resources in my spare time and cannot possibly answer everyone's particular requirements.

closed 2018-03-04 by botletics

muzmmilkhan, 2018-03-05:
its ok sir i got the answer and sorry for this isssue 
thanks