botletics/SIM7000-LTE-Shield #11: MQTT AIO for 3G?

jhonoryza, 2018-03-13:
i have test this library with sim5320 
network connection is works, but mqtt aio is not work

Retrying MQTT connection in 5 seconds...
	---> AT+CIPSTATUS
	<--- ERROR
	---> AT+CIPCLOSE
	<--- ERROR
	---> AT+CIPSHUT
	<--- ERROR

is there something iam missing?

botletics, 2018-03-13:
Actually I will need to completely change MQTT commands in the library because the SIM5320 has its own MQTT-specific commands ([see this MQTT app note](http://www.microchip.ua/simcom/WCDMA/APPNOTES/SIMCom_3G_MQTT_Application%20Note_V1.00.pdf)) and actually SIMCom told me they just updated the firmware to support MQTT commands for the SIM7000 as well so I don't have to do all the manual stuff. However, they haven't posted an MQTT application note yet. I'll keep you posted.

botletics, 2018-04-13:
Yes it would be best to use the specialized MQTT commands.

NamQB, 2019-04-15:
@botletics  @jhonoryza 
Can I get a sample code on MQTT data subcrice from a topic?
Thank you!

botletics, 2019-04-15:
You just have to check the SIM5320 MQTT app note.