botletics/SIM7000-LTE-Shield #12: Updated the Iot_Example please

muzmmilkhan, 2018-03-16:
Sir I want to subscribe to cloudmqtt because I am using 5 sensor in which two sensor need to subscribe I also know that the AIO support subscribe but it also need to much memory I know you are busy but as I am your follower and your my ideal person sir please update the ilt example I am trying to subscribe to cloudmqtt from last one month but I am not get success please allow the subscribe function in mqtt I bag you because I really need it please please please please please please update the subscribe function in cloudmqtt please please please please please 
I also respect you if there is something I say you don't like than sorry but please please please please update the IoT example for your follower or your brother thanks I love your project and also many people love your project I also purchasing the sim7000 
I hope you understand my feelings please I am doing my final year project after it I am going to open my business in bus tracking as a side business I am from India and I am doing my engineering in electronics and telecommunication so I am week in programming 
Sir please please please update the IoT example I very appreciate you
Thanks

botletics, 2018-03-16:
SIMCom is working on MQTT command so I will be waiting for that to be published. After those are published it will have everything you need.

muzmmilkhan, 2018-03-17:
Is SIMCom working on MQTT commands for all GSM module or specific module ?
I have Sim808 is the the MQTT command will Work for my sim808 sir!!!!
Thanks

botletics, 2018-03-17:
I am mainly referring to SIM7000 specifically.

muzmmilkhan, 2018-03-21:
Ok thanks

muzmmilkhan, 2018-03-22:
Please update the IoT example sir

muzmmilkhan, 2018-03-26:
Ha ha. Ha sir you have a time for making video but not for update.
That means you don't no how to subscribe to cloudmqtt in program .are you waiting for simcom AT command ha . they will update it after one year.
Than we have to wait upto one year.
If you have some AT command already than why not try it before and don't waste one year.
So I request you please try to subscribe to cloudmqtt it not only helps us also you 
Thanks

botletics, 2018-03-26:
You are not only begging for someone else to help you with your project but you're also now trying to criticize me?

closed 2018-03-26 by botletics

muzmmilkhan, 2018-03-27:
Thanks for compliment that i am begging and also try to criticize you.