botletics/SIM7000-LTE-Shield #14: Multiband LTE Module Support

atlury, 2018-03-31:
Any chance you could put a quick design for SIM7600E or similar modules like SIM7100? Would be very grateful. I guess the current pin-out may not support direct replacement.

Thanks

botletics, 2018-04-01:
That's not such an easy thing considering the pinouts are completely different. However, I do have other modules in mind, you'll have to see for yourself later :)

botletics, 2018-04-01:
By the way, I'm also taking pre-orders for the SIM7000G which is a global version, basically the A, C, and E versions combined. Maybe that's what you're looking for?

atlury, 2018-04-01:
Thanks for the quick response. There are still lots of places where LTE CAT-M1 and NB-IoT are not supported and only traditional LTE is supported. It would be great if you can provide a traditional LTE only module as well with GPS. 

The reason I ask you this is because I have seen you design and your antenna routing is very well done, which I dont observe even in some designs of adafruit.

SIM7000G still may not support tradition LTE.

botletics, 2018-04-01:
Thanks! And yes, as a spoiler alert the next board I'm releasing soon is traditional LTE and it can be LTE CAT-1 or CAT-4 simply by swapping out pin-compatible versions. It will support 3G as well and have GPS. Oh, and audio support. Stay tuned!

atlury, 2018-04-01:
ooh thats lovely. I am closing this with your answer. Hope to see it very soon.

closed 2018-04-01 by atlury