botletics/SIM7000-LTE-Shield #15: Linux USB Driver

rmarlan, 2018-04-09:
I am trying to support Raspberry Pi 3.  The linux driver code has the 7100 but not the 7000 listed.
When I try to compile on raspberry - jesse it gives error:
make[1]: *** /lib/modules/4.9.59-v7+/build: No such file or directory. Stop.

The Simcom site says there are both linux and android drivers for the 7000, but I can not find them.

The module works perfectly with Arduino!   Thanks.
Bob

botletics, 2018-04-09:
SIMCom told me the 7000 uses the same driver as the 7100 so that's what they gave me. I'll ask them about it and see what they say.

rmarlan, 2018-04-09:
Thanks Tim. Can you also ask if they can answer why it won’t compile
properly? Thanks again. Bob

botletics, 2018-04-09:
Sure, will do.

GiacomoGenovese, 2018-07-03:
Hi,
I have the same error, did you find a solution?

Thanks

botletics, 2018-07-03:
Sorry, they never answered this...

GiacomoGenovese, 2018-07-03:
I solved this problem by "There is a kernel-headers Raspbian package now: https://www.raspberrypi.org/documentation/linux/kernel/headers.md " from https://github.com/notro/rpi-source/wiki .
Now I have a new one:

_/home/pi/gobiserial/GobiSerial.c:55:28: fatal error: linux/usb-wwan.h: File o directory non esistente
 #include <linux/usb-wwan.h>_

botletics, 2018-07-03:
Great! Glad you figured it out, this will be very useful for others with the same issue.

botletics, 2018-07-03:
Oh wait, just realized you had another issue haha.

GiacomoGenovese, 2018-07-05:
I think that the make file should be adjusted to the actual firmware.
I don't know how...any help is welcome.

Thanks

TuanPi, 2018-09-12:
**I have a similar problem with RaspberryPi3. Have you solved this problem yet?**

botletics, 2018-09-12:
I am asking SIMCom again and will let you know.

botletics, 2018-09-20:
Any updates on this? Did you get it working? It would be useful to know if it worked for you, thanks!

GiacomoGenovese, 2018-09-21:
Hi,
I was not able to install the drivers and I moved to a different device.
Anyway, it works device works using AT commands.

Bests,
Giacomo.

diakbar, 2019-04-03:
> Hello. I seem to have a hack that gets the driver installation working without requiring the whole kernel to be compiled.
> 
> 1. Install the kernel headers using `sudo apt-get install raspberrypi-kernel-headers`
> 2. Identify the kernel version using `uname -r`
> 3. Peruse the available kernel branches using `git ls-remote --heads https://github.com/raspberrypi/linux`
> 4. Clone the appropriate branch somewhere (`~/Downloads/linux` in this example) using `git clone --depth=1 --branch YOUR_BRANCH https://github.com/raspberrypi/linux ~/Downloads/linux`, where THE_RIGHT_BRANCH could for example be `rpi-4.12.y`
> 5. Edit the GobiSerial.c file and replace `#include <linux/usb-wwan.h>` with `#include "/home/pi/Downloads/linux/drivers/usb/serial/usb-wwan.h"`
> 6. Lastly, edit `/lib/modules/$(uname -r)/build/Makefile` and replace `CFLAGS_MODULE  =` with `CFLAGS_MODULE   = -Wno-incompatible-pointer-types` to get past a compile warning that will otherwise fail your build.
> 
> After the above steps, I was able to run `make` from the gobiserial directory successfully and install `GobiSerial.ko` per the instructions in https://www.pianywhere.com/wp-content/uploads/2015/11/how_to_use_the_sim7100_module_in_linux-driver.pdf - despite the "7100" in the name of that document. The installation succeeded. Well... the installation commands exited without any errors. I don't know if they _succeeded_. Taking such a hacky approach might bite me in the ass, but YOLO.
 i got this error
```
rm -rf *.o *~ core .depend .*.cmd *.ko *.mod.c .tmp_versions Module.* modules.order
make -C /lib/modules/4.14.98-v7+/build M=/root/brofist/driver/sim7600-driver-linux modules
make[1]: Entering directory '/usr/src/linux-headers-4.14.98-v7+'
  CC [M]  /root/brofist/driver/sim7600-driver-linux/GobiSerial.o
/root/brofist/driver/sim7600-driver-linux/GobiSerial.c:120:17: warning: initialization discards ‘const’ qualifier from pointer target type [-Wdiscarded-qualifiers]
   .iface_info = sim9046_iface_info_array,
                 ^~~~~~~~~~~~~~~~~~~~~~~~
/root/brofist/driver/sim7600-driver-linux/GobiSerial.c:126:17: warning: initialization discards ‘const’ qualifier from pointer target type [-Wdiscarded-qualifiers]
   .iface_info = sim5360_iface_info_array,
                 ^~~~~~~~~~~~~~~~~~~~~~~~
/root/brofist/driver/sim7600-driver-linux/GobiSerial.c:132:17: warning: initialization discards ‘const’ qualifier from pointer target type [-Wdiscarded-qualifiers]
   .iface_info = sim7100_iface_info_array,
                 ^~~~~~~~~~~~~~~~~~~~~~~~
/root/brofist/driver/sim7600-driver-linux/GobiSerial.c: In function ‘GobiProbe’:
/root/brofist/driver/sim7600-driver-linux/GobiSerial.c:347:23: warning: format ‘%d’ expects argument of type ‘int’, but argument 2 has type ‘long unsigned int’ [-Wformat=]
          printk("num=%d, inface_num=%p", num, iface_info);
                       ^
/root/brofist/driver/sim7600-driver-linux/GobiSerial.c:348:23: warning: format ‘%d’ expects argument of type ‘int’, but argument 2 has type ‘long unsigned int’ [-Wformat=]
          printk("num=%d, inface_num=%d", num, iface_info[num]);
                       ^
In file included from ./include/linux/printk.h:7:0,
                 from ./include/linux/kernel.h:14,
                 from ./include/linux/list.h:9,
                 from ./include/linux/wait.h:7,
                 from ./include/linux/wait_bit.h:8,
                 from ./include/linux/fs.h:6,
                 from ./include/linux/tty.h:5,
                 from /root/brofist/driver/sim7600-driver-linux/GobiSerial.c:47:
/root/brofist/driver/sim7600-driver-linux/GobiSerial.c: In function ‘GobiOpen’:
./include/linux/kern_levels.h:5:18: warning: format ‘%ld’ expects argument of type ‘long int’, but argument 3 has type ‘unsigned int’ [-Wformat=]
 #define KERN_SOH "\001"  /* ASCII Start Of Header */
                  ^
./include/linux/kern_levels.h:14:19: note: in expansion of macro ‘KERN_SOH’
 #define KERN_INFO KERN_SOH "6" /* informational */
                   ^~~~~~~~
/root/brofist/driver/sim7600-driver-linux/GobiSerial.c:105:10: note: in expansion of macro ‘KERN_INFO’
   printk(KERN_INFO format, ##arg);
          ^~~~~~~~~
/root/brofist/driver/sim7600-driver-linux/GobiSerial.c:474:10: note: in expansion of macro ‘DBG’
          DBG( "invalid write size %d, %ld\n",
          ^~~
/root/brofist/driver/sim7600-driver-linux/GobiSerial.c: In function ‘GobiClose’:
./include/linux/kern_levels.h:5:18: warning: format ‘%ld’ expects argument of type ‘long int’, but argument 3 has type ‘unsigned int’ [-Wformat=]
 #define KERN_SOH "\001"  /* ASCII Start Of Header */
                  ^
./include/linux/kern_levels.h:14:19: note: in expansion of macro ‘KERN_SOH’
 #define KERN_INFO KERN_SOH "6" /* informational */
                   ^~~~~~~~
/root/brofist/driver/sim7600-driver-linux/GobiSerial.c:105:10: note: in expansion of macro ‘KERN_INFO’
   printk(KERN_INFO format, ##arg);
          ^~~~~~~~~
/root/brofist/driver/sim7600-driver-linux/GobiSerial.c:550:10: note: in expansion of macro ‘DBG’
          DBG( "invalid write size %d, %ld\n",
          ^~~
At top level:
/root/brofist/driver/sim7600-driver-linux/GobiSerial.c:661:12: warning: ‘GobiSuspend’ defined but not used [-Wunused-function]
 static int GobiSuspend(
            ^~~~~~~~~~~
  Building modules, stage 2.
  MODPOST 1 modules
  CC      /root/brofist/driver/sim7600-driver-linux/GobiSerial.mod.o

```
my RPI kernel: 4.14.98-v7+

bliteknight, 2019-05-04:
I'm at a loss, I followed @MikeBosw steps and got it to work...but after some reboots I'm now stuck with just ttyUSB0 and don't get 0-5 :(

artnum, 2019-05-08:
Hello. I have been able to get it working. If you look at the source code provided, you see « Linux Qualcomm Serial USB ». So by going to current Qualcomm USB Serial driver in the current linux kernel https://github.com/torvalds/linux/blob/master/drivers/usb/serial/qcserial.c, you can add your device in the list

    {USB_DEVICE(0x1E0E, 0x9001)},   /* SimCom SIM7000 */

And it did work quite nicely for me :

```
[   60.147826] usb 1-2: new high-speed USB device number 12 using xhci_hcd
[   60.288132] usb 1-2: New USB device found, idVendor=1e0e, idProduct=9001, bcdDevice= 3.18
[   60.288140] usb 1-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[   60.288144] usb 1-2: Product: SimTech, Incorporated
[   60.288149] usb 1-2: Manufacturer: SimTech, Incorporated
[   60.288152] usb 1-2: SerialNumber: 0123456789ABCDEF
[   60.846808] usbcore: registered new interface driver usbserial_generic
[   60.846822] usbserial: USB Serial support registered for generic
[   60.850656] usbcore: registered new interface driver option
[   60.850686] usbserial: USB Serial support registered for GSM modem (1-port)
[   60.850878] option 1-2:1.0: GSM modem (1-port) converter detected
[   60.851112] usb 1-2: GSM modem (1-port) converter now attached to ttyUSB0
[   60.852366] option 1-2:1.1: GSM modem (1-port) converter detected
[   60.852699] usbcore: registered new interface driver cdc_wdm
[   60.852995] usb 1-2: GSM modem (1-port) converter now attached to ttyUSB1
[   60.853056] option 1-2:1.2: GSM modem (1-port) converter detected
[   60.853162] usb 1-2: GSM modem (1-port) converter now attached to ttyUSB2
[   60.853258] option 1-2:1.3: GSM modem (1-port) converter detected
[   60.853823] usb 1-2: GSM modem (1-port) converter now attached to ttyUSB3
[   60.853879] option 1-2:1.4: GSM modem (1-port) converter detected
[   60.853971] usb 1-2: GSM modem (1-port) converter now attached to ttyUSB4
[   60.854752] usbcore: registered new interface driver qcserial
[   60.854767] usbserial: USB Serial support registered for Qualcomm USB modem
[   60.856943] qmi_wwan 1-2:1.5: cdc-wdm1: USB WDM device
[   60.857604] qmi_wwan 1-2:1.5 wwan0: register 'qmi_wwan' at usb-0000:00:14.0-2, WWAN/QMI device, 16:3a:e3:1d:f4:41
[   60.857706] usbcore: registered new interface driver qmi_wwan
[   60.862408] qmi_wwan 1-2:1.5 wwp0s20f0u2i5: renamed from wwan0
```

bliteknight, 2019-05-25:
Just an update, and possible tutorial to get this board working successfully with the Raspberry PI:
@botletics - This is a lengthy post and would be suited for a wiki page, just don't know where to post one at.

Before we beign, you will need to have these two things to make this work
- a lipo battery as stated - I used a 3.7 1800 mah lipo
- a micro usb male to male connector - I spliced two cables to make one, power is not needed, just the data cables (white & green)

The reason for the lipo battery is to supply enough current so the board can function without turning off the pi.
The reason for the micro usb cable is because we are going to use this board as a QMI modem and not mess with any of the AT commands stuff. It is still supported but I'd rather just have a internet connection and my choice of communication protocol to use.

**Step 1**: Install the latest stretch image
i'm using 2019-04-08-raspbian-stretch-lite and once you've gotten your configuration setup the way you want via raspi-config (ssh, timezone, locale etc) you are ready to begin the setup.

**Step 2**: Power the device
The pi zero has a 5v out which can be connected to the 5v in on the board (the one close to the two ground points); Just wire / solder / the 5v to 5V and the pi ground to ground. The battery full light should come on and say on when it has power. If the lte (blue led) is not blinking, hold down the power on switch for a couple seconds to turn it on. Once both lights are active plug in the device via the micro usb male to male cable.

**Step 3**: Check for device detection
Run this command
`ls -al /dev/ttyUSB* `
you should see ttyUSB0 ttyUSB1 ttyUSB2 ttyUSB3 ttyUSB04

Also, if you run 
`ifconfig`
You should see a **wwan0** device listed
If you don't see this then you might need to install the drivers like mentioned above, but going off a fresh image for the pi zero I did not encounter this issue.
To test AT commands, I would install gammu
`sudo apt install gammu `
then create a config file at this location ~/.gammurc
```
[gammu]
device = /dev/ttyUSB2
connection = at
```
The AT port will be **ttyUSB2** and to validate your device run this command
`sudo gammu identify`
This should spit out the device information.

**Step 4**: QMI Modem Setup
Alot of the information I got here is based on the documentation here:
http://www.embeddedpi.com/documentation/3g-4g-modems/raspberry-pi-sierra-wireless-mc7455-modem-raw-ip-qmi-interface-setup
But there are some bits missing that I had to search countless forums to find a solution - 
install the latest libqmi-utils
```
apt remove libqmi-utils
wget http://www.freedesktop.org/software/libqmi/libqmi-1.22.4.tar.xz
tar -vxf libqmi-1.22.4.tar.xz
cd libqmi-1.22.4
./configure --prefix=/usr --disable-static
make && sudo make install
```
This will take a long time on the pi zero w, about 30 mins to complete. If you have a pi3 I would do it on there and run 
`make -j 4 && sudo make install`
to speed things up

Once that is done, it's time to fix a bug in the qmi code:
https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=845979
Edit this file: **/usr/bin/qmi-network**
find this string: START_NETWORK_ARGS="apn='$APN'"
and change it to: START_NETWORK_ARGS="'apn=$APN'"

Or if you would rather do it from the terminal, run this command
`sudo sed -i "s/apn='/'apn=/g" /usr/bin/qmi-network`

Once that's done, install udhcpc
`sudo apt install -y udhcpc`

As this modem only works in raw-ip mode (- not fully clear what that is); even if you connect to a carrier you will not be assigned an IP until you request one, which is what udhcpc does.

Next, create a file to store your APN information
`sudo nano /etc/qmi-network.conf`
I'm using a thingsmobile SIM card so my info looks like this
`APN=TM,ip-type=4`
save the file, and now it's time to test the connection.

**Step 5**: Internet Please?
First, try to ping google.com via the wwan0 interface for 4 tries:
`ping -c 4 -I wwan0 www.google.com `
 you should see 100 packet loss, now it's time to tell the modem to connect to the carrier via:

```
qmicli -d /dev/cdc-wdm0 --device-open-"net=net-raw-ip|net-no-qos-header" --wds-start-network="apn=TM,ip-type=4" --client-no-release-cid
  
ip link set dev wwan0 down

echo Y > /sys/class/net/wwan0/qmi/raw_ip

ip link set dev wwan0 up
```
The first command is the connect command, you might not need the **--wds-start-network** parameter, but I put it in there to make sure it used the correct APN values.
The next three commands put the modem in raw-ip mode (more info about that from the URL in Step 4)
Once that command completes, it's time to get an IP address
`udhcpc -i wwan0`
You should see a **sending discover** output in the terminal a handful of times, before getting an IP with a lease time. Once you have that and can verify your IP is that same via
`ifconfig` 
you can test pinging google again
`ping -c 4 -I wwan0 www.google.com `
This time you should have a 0% packet loss.

Hopefully this will help anyone who wants to use this board with a raspberry pi and also doesn't want to be limited to using AT commands. 

I was able to have it up and running using nodejs + socket.io to do bi-directional communication from the pi and my server, sending data from an attached sensor every second in real time for 3+ days with no issue.

Thanks to @botletics for creating such a great board.

botletics, 2019-05-25:
Awesome, thanks for the update! Maybe I could start a wiki page and paste what you've written in there.

botletics, 2019-05-25:
I just created a wiki page with what you've posted which you can find [here](https://github.com/botletics/SIM7000-LTE-Shield/wiki/Raspberry-Pi)

bliteknight, 2019-05-26:
Awesome, thanks glad I could help.

cellphonedude, 2019-06-29:
@bliteknight @botletics please update the wiki to include prerequisites for Raspbian buster:

Before compiling run
sudo apt-get install libperl-dev libgtk2.0-dev libgudev-1.0-dev

frankurcrazy, 2019-07-11:
@bliteknight 
I followed the steps you provide; unfortunately, I wasn't able to get IP address with `udhcpc`.
When using AT commands, I can successfully get IP address after invoking:
1. AT+CSTT="internet.iot" (the apn of my carrier)
2. AT+CIICR
3. AT+CIFSR
and is able to ping internet host with AT+CIPPING.

Do you have any suggestions?

frankurcrazy, 2019-07-15:
It turns out that the version of libqmi matters. Using the libqmi in ubuntu 16.04 official repository doesn't work. After building libqmi-1.22.4 manually, I can now get IP address correctly with udhcpc.

jogendarkumar, 2019-07-22:
> [gammu]
> device = /dev/ttyUSB2
> connection = at

I' m getting this error. 
sudo gammu identify
Warning: No configuration file found!

jogendar, 2019-07-22:
> To test AT commands, I would install gammu
> `sudo apt install gammu `
> then create a config file at this location ~/.gammurc
> 
> ```
> [gammu]
> device = /dev/ttyUSB2
> connection = at
> ```

I' m getting this error.
sudo gammu identify
Warning: No configuration file found!

jogendarkumar, 2019-07-23:
FInally 
`sudo apt-get install libgudev-1.0-dev`
solved my problem

adrianwiejak, 2019-09-10:
@bliteknight  i try connect:

`qmicli -d /dev/cdc-wdm0 --device-open-"net=net-raw-ip|net-no-qos-header" --wds-start-network="apn=internet,ip-type=4" --client-no-release-cid`

and i have some errors :( :

```
error: operation failed: Transaction timed out
[/dev/cdc-wdm0] Client ID not released:
	Service: 'wds'
	    CID: '3'
qmicli: symbol lookup error: qmicli: undefined symbol: qmi_device_close_async
```

Do you have any suggestions?

bliteknight, 2019-09-11:
@adrianwiejak

I've run into that issue when I was getting a weak signal and I had to test it outside.

Another thing that you can try if going outside doesn't work is to reflash the firmware of the modem.

I've gotten the SIM7000G variant and another SIM7000A and both would not connect till I reflashed the firmware, not sure why but I'd try it outside first before reflashing.

jupiter8888, 2020-02-22:
Taught i would give you some advice to ease your life a bit. I had my share of troubles making these modem work. Days of pulling my hair out.

I'm using a sim7000g pcie with a usb adapter wich is more or less the same, and using the QMI WAN means running some script to connect or doing it manually. Not the best for me as i have a few dozen of them in the field left unatended and it need to connect back itself after a power loss or a reset.

I was using Huawei E-303 in Hilink mode before. Great modem, with usb-modeswitch being part of raspbian, you plug the thing and it work. Period. I needed something like this in Lte Cat1.

So the trick to make this modem being recognised as an ethernet adapter is to change it's mode with the AT+CECMMODE=1 command. This obscure command is not even published by Simcom themselfes, ask me why.

Starting with a fresh install of raspbian, plug your modem in a USB port with a valid sim card. enter lsusb -t, it will show you this:

[codepi@raspberrypi:~ $ lsusb -t
/: Bus 01.Port 1: Dev 1, Class=root_hub, Driver=dwc_otg/1p, 480M
|__ Port 1: Dev 3, If 0, Class=Vendor Specific Class, Driver=option, 480M
|__ Port 1: Dev 3, If 1, Class=Vendor Specific Class, Driver=option, 480M
|__ Port 1: Dev 3, If 2, Class=Vendor Specific Class, Driver=option, 480M
|__ Port 1: Dev 3, If 3, Class=Vendor Specific Class, Driver=option, 480M
|__ Port 1: Dev 3, If 4, Class=Vendor Specific Class, Driver=option, 480M
|__ Port 1: Dev 3, If 5, Class=Vendor Specific Class, Driver=qmi_wwan, 480M][/code]

Now with minicom installed start it with this: sudo minicom -D /dev/ttyUSB3

enter your apn if not set: AT+CGDCONT=1,"IP","yourapn"

Now with a valid sim card in the modem (important or it will not work) enter this command: AT+CECMMODE?

It will tell you +CECMMODE=0

Enter: AT+CECMMODE=1

Now check again with AT+CECMMODE?

It will be: +CECMMODE=1

Now check with lsusb -t, it should be now a cdc_ether device
Code: Select all

pi@raspberrypi:~ $ lsusb -t
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=dwc_otg/1p, 480M
    |__ Port 1: Dev 4, If 0, Class=Vendor Specific Class, Driver=option, 480M
    |__ Port 1: Dev 4, If 1, Class=Vendor Specific Class, Driver=option, 480M
    |__ Port 1: Dev 4, If 2, Class=Vendor Specific Class, Driver=option, 480M
    |__ Port 1: Dev 4, If 3, Class=Vendor Specific Class, Driver=option, 480M
    |__ Port 1: Dev 4, If 4, Class=Vendor Specific Class, Driver=option, 480M
    |__ Port 1: Dev 4, If 5, Class=Communications, Driver=cdc_ether, 480M
    |__ Port 1: Dev 4, If 6, Class=CDC Data, Driver=cdc_ether, 480M
And you should have interwebs right away. Type ifconfig and your modem should be listed as eth0 (if you have a pi zero) or eth1.

Ping google: ping -I eth0 8.8.8.8

That's all, nothing else to do.

crazydiam0nd, 2020-05-07:
I am really struggling to get this Waveshare 700E Pi zero hat to work.
Have spend hours typing different commands, it's almost impossible and cannot even find the drivers online.
Why is this so difficult? It's supposed to be made for the Pi Zero and plugs stright on using the TX and RX which I have had working fine with Minicom connected to an AtMega chip etc.

Might as well put it in the bin.
Any detailed how to guides?

imrj, 2020-05-08:
am trying so same thing and always something breaks....i thought i was right there with the guide from @jupiter8888 but not so fast, mine throws me 
AT+CECMMODE=1
ERROR

crazydiam0nd, 2020-05-08:
> am trying so same thing and always something breaks....i thought i was right there with the guide from @jupiter8888 but not so fast, mine throws me
> AT+CECMMODE=1
> ERROR

Same.
I've managed to get it connected and can send AT commands that will give me IP address etc. but this is no good as I need constant connection.
I'm not using USB, I'm using the hat on a Pi zero which will use try/AMA0 

seems like I have tried everything now.

imrj, 2020-05-08:
yea it seems the only way to get it working with a constant IP connection is to make sure use qmi with USB, the USB-TTL serial wont work.....mine its now seen as a wwan0 adapter on ifconfig when i switched to using USB port on SIM7000, but when following the instructions from @jupiter8888 i cant get to making it a CDC device so it wont get an IP address.....the command AT+CECMMODE? just errors out on me.....the interesting thing is that hologram dash does see my device connected to the internet....

crazydiam0nd, 2020-05-08:
I managed to get the commands working that @jupiter8888 advised on but I had to take a different approach.

AT+CSTT="iot.cslm2m.com" (apn of SIM card)
AT+CIICR
AT+CIFSR
Then followed Jupiter's guide and all commands worked.
apart from I'm not using USB as this is a hat with Sim7000e and uses UART as far as I can tell.    /tty/AMA0

once I type in ifconfig I cannot see any other adaptors.
loosing the will to live with this 😆

crazydiam0nd, 2020-05-08:
> yea it seems the only way to get it working with a constant IP connection is to make sure use qmi with USB, the USB-TTL serial wont work.....mine its now seen as a wwan0 adapter on ifconfig when i [...]

mine is headless
I'm wondering whether compiling the kernel again but somehow changing 
#include "/home/pi/Downloads/linux/drivers/usb/serial/usb-wwan.h"
changing it so it creates wwan on tty/AMA0 instead?
I'm not sure how to even go about this.
I can't use USB as that's being used for serial communication from an ATMega382 chip and I need this project as small footprint as possible.

imrj, 2020-05-08:
> mine is headless
> I'm wondering whether compiling the kernel again but somehow changing
> #include "/home/pi/Downloads/linux/drivers/usb/serial/usb-wwan.h"
> changing it so it creates wwan on tty/AMA0 [...]

mine is headless too, but i can use USB port, so am a little further than you in that my Pi sees a wwan0 adapter, it just wont get the IP address......do you show a wwan0 adapter? if not then you need to probably do the compile option.....honestly I dont know how you will get the serial port to be seen as a network adapter, you might want to try a USB-TTL adapter (CP2102), which is what I was using on mine before going to the native USB port on the SIM7000, but with the USB-TTL it coudlnt even get it to show as a network adapter.....

imrj, 2020-05-08:
another option you can try if you absolutely have to use the ttyAMA0 serial port is to install a serial-to-wifi transparent bridge option, you might want to look at  this --->  https://raspberrypi.stackexchange.com/questions/15382/raspberry-pi-used-as-a-cheap-serial-to-wlan-converter

crazydiam0nd, 2020-05-08:
> another option you can try if you absolutely have to use the ttyAMA0 serial port is to install a serial-to-wifi transparent bridge option, you might want to look at this ---> https://raspberrypi.stackexchange.com/questions/15382/raspberry-pi-used-as-a-cheap-serial-to-wlan-converter

Thanks, I'll look into that.
I've hooked it up now with USB but I don't see Wwan, I see Eth0 but the IP address is a local one. and it keeps crashing the Pi now in minicom.
AT commands show IP address but for the life of me I cannot get this unit to work.
might look at other chips like the Telit one as people don't seem to have this problem.
all I want is my pi to have internet access so I can subscribe and publish to a topic on AWS IOT. it needs constant internet access so I can send a command and receive GPS coordinates and maybe SSH into it.

jupiter8888, 2020-05-08:
@crazydiam0nd 

You won't see a wwan, with the at+cecmmode=1 command because it turns the modem into an ethernet device and will be named eth0 and ifconfig will give you an internal address of the 192.168.255 range.
if you type AT+CGPADDR in minicom you'll see your external ip. 

Start fresh with a new raspbian up to date, plug the usb cable,  install minicom and try agin. Don't try to access the modem with any other application in the same time,  it won't work. 
To ssh in your pi out of your lan i suggest [](https://remote.it/) this is what i use. Also suggest you to use node-red [](https://nodered.org/) It will do whatever you want. i use my at commands right into it with USB2 setup in a serial node. and there is nodes specifically for AWS. better use mqtt for what you want to do.

note that i have Revision:1529B02SIM7000G.
Also those AT+CSTT, AT+CIICR and AT+CIFSR will only work when a pdp context is activated and in multiple connexion mode, wich i'm not.
multiple connexion mode:  at+cipmux? gives me +cipmux: 0, means i'm not in multi mode
cipstatus gives me OK,  STATE: IP INITIAL

from SIM7000 Series_AT Command
Manual_V1.04 : 
<<Only after PDP context is activated, local IP address can be obtained by
AT+CIFSR, otherwise it will respond ERROR. To see the status use
AT+CIPSTATUS command. Status should be:
IP GPRSACT, TCP CONNECTING, UDP CONNECTING, SERVER
LISTENING, IP STATUS, CONNECT OK, TCP CLOSING, UDP
CLOSING, TCP CLOSED, UDP CLOSED in single-connection mode (see
<state> parameter);
IP STATUS, IP PROCESSING in multi-connection mode (see <state>
parameter).>>

maybe you should reset the modem to it'original state and reenter you apn. the try to ping google from a command prompt if you have eth0 (not from minicom) with ping -I eth0 8.8.8.8

good luck.

crazydiam0nd, 2020-05-08:
@jupiter8888 

Thanks for the information.
my Pi zero is currently running the latest 4.19.97 as I only installed it a few days ago. it's the small version for Pi Zero and I'm running headless.

I have 2 Pi Zero's, 1 with WiFi and 1 without.

I've only installed Node-Red on them and have successfully setup a serial monitor and AWS-IOT connection. it's working perfectly but now it's time to make my project mobile.

I'll try again but after reading countless forums it appears this HAT will only work in PPP mode if using USB and I need to use the onboard TX and TX pins as the USB is needed for something else.
I'll reset the unit and try again from scratch as suggested.
I've downloaded Linux headers, gobiserial drivers etc.
my Pi doesn't list this hat as a usb device when running lsusb or ls /tty/usb*

You'd of thought having a headless Pi with mobile internet (Without using AT commands) would be simple, mobile phones do it 🤷🏻‍♂️

absolutely bizarre. 
3 days of trying I'm almost ready to throw in the towel

imrj, 2020-05-08:
> @jupiter8888
>
> Thanks for the information.
> my Pi zero is currently running the latest 4.19.97 as I only installed it a few days ago. it's the small version for Pi Zero and I'm running headless.
>
> I [...]

@crazydiam0nd 
if you want to try you could get a huawei usb dongle, its cheap, and its working on a pi4 beautiful over 4G LTE for me with another drone setup. and none of the nonsense with this board. 
Just offering up ideas, i can relate to the frustration.....
and you know whats silly, is that I have Windows IoT 10 setup on the same exact hardware and it works perfect, but when I throw in the Raspbian SD card it all turns to sh1t...its a shame Windows is working and not Linux.....unreal

@jupiter8888 
why does mine error out with AT+CECMMODE? , i am connecting to ATT network and with an IP address....but i can only see that external IP thru AT commands, its not passing the IP to Linux, and am confused how you say it should show as ETH0, that would be the internal builtin ethernet adapter, at least for me with a pi4......the SIMCOM is showing as wwan0 but it wont get an IP at all....

imrj, 2020-05-08:
@jupiter8888  
so my wwan0 i know is the SIM modem because the mac address is that of the modem itself, and i do see an IP via AT commands only, so am mostly there, whatever it is, its just linux not routing the IP stack properly from the SIM device

crazydiam0nd, 2020-05-08:
I've literally just successfully flashed the firmware to B08, installed raspbian lite on a fresh SD and about to try again.
I didn't want to go the dongle route as needed something I could end up using on my own PCB for my project.
also the USB is being used to communicate with an ATMega with an FTL232.
although I could use the UART and logic level convertor.
it's incredibly frustrating and I may even contact Simcom and see if they have a solution. 
you'd think @waveshare wouldn't sell these as IOT hats for the PI if we can only use AT commands to send and receive data.
I need it listening constantly and subscribed to a topic.
I'll have a play as not easily defeated but getting there

jupiter8888, 2020-05-08:
@crazydiam0nd 
Don't install Linux headers, gobiserial drivers etc. Only a fresh stretch or buster nothing else for testing, also node-red serial node and minicom won't work together, use one or the other. shutdown node-red if using minicom, use ttyUSB2 as it is the AT port.

I work with ppp with other modems (sim5216 and sim 5320 shields) and it work without problems. And i'm using the uart with level shifters to talk with my arduinos. It's a proven solution for me.

@imgr 
I don't know why the at+cecmmode don't work for you, you have a valid sim card inserted? what firmware? you`re connected via usb? Did you tried resetting the hat to it's factory default?

cellphonedude, 2020-05-08:
So this LTE one will not work with ppp as the previous ones did. Qmi is
what is used to effectively get these things online. I believe
network-manager works with these zero config. You'll need to use the USB as
serial won't be fast enough to utilize LTE speed.

Try searching with qmi instead of ppp.

I have a silly script I use to get mine online but it's in no way stable
the correct way to implement these IMHO is to use USB with qmi and network
manger there's is a Linux package the supports these out of the box.

The latest Kali image works out of the box after your put in your apn, I've
got mine online and up for a few weeks at this point.

jupiter8888, 2020-05-08:
One thing worth mentionning is when using a pi zero with usb i had frequent disconnect of the serial port in node-red,  probably because EMI or if the antenna is too close from the pi. i diched the pi zero in flavor of pi 3b or orange pi pc plus with full size usb ports and no problems since then.

crazydiam0nd, 2020-05-09:
@cellphonedude Thanks for this information, which version of Kali?
I wonder if I can strip this version of Kali back as won't need all the pen testing tools.
Also I wonder which packages make it work out the box so to speak?

@jupiter8888 Again, thanks for your help.
I've run into problems with Rasbian lite not recognising the sim7000e when plugged into USB so I tried the full desktop version and I added some sheilding between the HAT and pi.
This worked and I can now see as USB device and eth0.
However, I still cannot ping out using ping -I eth0 8.8.8.8

I think I'll throw in the towel now and look at other options.
I could potentially be buying thousands of these if my project gets off the ground and if @waveshare or @SimCom are unable to help then I'll look into using Telit or Sierra.

Which dongle works? I'd be interested to know what chip it uses.
4G is not needed but at least 3G. Cost is a factor.
Surely there is a solution other than LoRaWan
I chose this unit as the specs seemed to match my project with GPS etc

crazydiam0nd, 2020-05-09:
Ok I've kind of got this working now.
When I say "kind of" it means it works for a ping request and then everything goes to $h!t

Maybe someone can look into the method I used and get it stable?

Wwan0 obtained IP address and all seemed good, but it doesn't stick for long.
I got 31 packets transmitted and 15 received, 51.6129% packet loss, time 716ms

I'll post up how as soon as I get 5 minutes to look through my terminal history.

jupiter8888, 2020-05-09:
@crazydiam0nd Do you have a pi 3 or 4 around? I would test with those as i think the shielding is very poor on a zero. Whith the shield on check the serial node status in node-red, chance you'll see it disapear a few times per minute like me. That's why i moved away from zeros. It seems that cat-M1 put more energy around the antenna hence the EMI.

The cheapest and easiest setup i have in my iot stuff is a few pi zeros with huwawei E-303 3g dongles (15$ each from china, there are variants for different regions,beware.)  in hilink mode. Plug and play, period. you could also switch it in ppp mode with a usb-modeswithch script and use at commands for signal etc... Of course you have to design a space for it in your pcb and use some kind of hub if you have other usb devices. but theses are rock solid, i have them in the field for 3 years and never failed. I use a usb 1m usb extension cable to raise the dongle out of the metal enclosure, up in the air in a plasic pipe.

cellphonedude, 2020-05-09:
Was the lastest for the pi 4 whatever networking stack they used by default
supports modems with qmi interfaces.

imrj, 2020-05-09:
@crazydiam0nd @jupiter8888 
ill be dam....all the wasted time and installing this got it working in < 5min.....unbelievable....and with a GUI and all......am all set....hope works out for you guys too, it should
oh its free too....

https://raspap.com/

crazydiam0nd, 2020-05-10:
@jupiter8888 
I've not got a Pi 3 or 4 as lent it to someone but I've moved the sheild away from the Pi and put the Pi in an anti static bag, and that's stopped it crashing.
Thanks for the heads up on the dongle, I may get one and reverse engineer it. 

@cellphonedude  it was your original qmi guide that I followed on github that led me in the right direction, so thanks for that.
I had to change a few commands and remove the need for APN user name and password. But this has got me online.
Just the IP doesn't stick for long before it changes back and on every reboot I have to start all over again.
But making progress. 

@imrj  I'm not really looking for a GUI solution but will look into this and see if it will run in CMI
GUI is so slow on a Pi zero

Actually looking at an orange pi 4G board, they look promising.

jupiter8888, 2020-05-10:
@crazydiam0nd  I have one of these sleeping on my shelve, usless for me as it doesn't work here in Canada, you want it? make me an offer i'll ship it to you.

crazydiam0nd, 2020-05-10:
@jupiter8888 why doesn't it work in Canada?
I'm in the UK. Surely you have world Sims?

jupiter8888, 2020-05-10:
https://www.aliexpress.com/item/32858836485.html

It's like your botletic shield, sim7000E is for Europe, sim7000A for america and sim7000G is global.

imrj, 2020-05-11:
@jupiter8888 @crazydiam0nd 
this is what it looks for me in a working state, using with zerotier now and its very reliable (ive scrambled my ether and IPs)

enxb827e67f694f: flags=4099<UP,BROADCAST,MULTICAST>  mtu 1500
        ether b8:27:eb:67:69:4f  txqueuelen 1000  (Ethernet)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

lo: flags=73<UP,LOOPBACK,RUNNING>  mtu 65536
        inet 127.0.0.1  netmask 255.0.0.0
        inet6 ::1  prefixlen 128  scopeid 0x10<host>
        loop  txqueuelen 1000  (Local Loopback)
        RX packets 487078  bytes 45531083 (43.4 MiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 487078  bytes 45531083 (43.4 MiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

wlan0: flags=4099<UP,BROADCAST,MULTICAST>  mtu 1500
        ether b8:27:eb:67:3c:2a  txqueuelen 1000  (Ethernet)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

wwan0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 22.187.177.175  netmask 255.255.255.252  broadcast 22.186.107.107
        ether 32:5d:3d:6a:55:1d  txqueuelen 1000  (Ethernet)
        RX packets 3407  bytes 775014 (756.8 KiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 6148  bytes 962442 (939.8 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

zt2lr3yo3a: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 2800
        inet 172.27.77.48  netmask 255.255.0.0  broadcast 172.27.255.255
        ether b2:f7:3a:87:4a:c1  txqueuelen 1000  (Ethernet)
        RX packets 1103  bytes 112966 (110.3 KiB)

imrj, 2020-05-11:
@jupiter8888 @crazydiam0nd 

notice my ethernet and wifi on the Pi are both unassigned, so as to they not be the default routes....zerotier interface (last one) is default route, havent tried enabling either one of them yet.....

I never  got the modem to show as any other adapter type except wwan0 (ppp was possible when i tried it as a dial connection, but was messy)

crazydiam0nd, 2020-05-11:
> @jupiter8888 @crazydiam0nd
>
> notice my ethernet and wifi on the Pi are both unassigned, so as to they not be the default routes....zerotier interface (last one) is default route, havent tried [...]

I've requested a refund on the HAT, Its been days now and have followed every guide, script and command going.
I had success but soon as a ping was requested the IP would disappear.
I even installed the Modem Supervisor Script for RAW-IP Modems 

Is it the SIM7000E NB-IoT HAT that you are using?
people are having success with the 7600 but I doubt I will ever touch another SimCom embedded device again after this and the lack of support.

Now I need to find one that will work and won't cost the earth

jupiter8888, 2020-05-12:
@crazydiam0nd  I'm using this:  https://www.aliexpress.com/item/32967753699.html
Set apn, cecmmode and away you go. Don't try it with a pi zero, EMI will kick out the device from the usb port. with full size usb like pi 3,4 or orange pi no problem.

 You have a CAT-M1 or NB-iot Sim card don't you? As you know that regular 3g, 4g, lte sims won't work with this modem.

crazydiam0nd, 2020-05-12:
> @crazydiam0nd I'm using this: https://www.aliexpress.com/item/32967753699.html
> Set apn, cecmmode and away you go. Don't try it with a pi zero, EMI will kick out the device from the usb port. with [...]

That's incredibly cheap. I may have to order one.
And this works PPP? 
Basically I need the internet connection up constantly.

Yes I'm using an IOT simcard made by tele2iot that's a world SIM and I purchased from a company that supplies all of my alarm monitoring diallers.

This unit kind of looks like the SIXFAB one that I was considering buying. Uses a mini PCI card.

I'm using with the Pi Zero but have managed to prevent it getting interference, if it wasn't for you mentioning it before then I'd probably have never worked it out, so Thanks for that
I used a longer quality usb cable and this stopped it crashing.

I might get an Esp32 instead or another embedded board. Needs to be small enough to fit inside my project. 
I'll order another pi 3 and test different boards

[9 more replies]