botletics/SIM7000-LTE-Shield #16: Add support to upload firmware properly to solve IMEI issue

bengchet, 2018-04-12:
Hi,

First of all I would like to say your project is great, I love it. Here is some experience I did some firmware upgrade and I would like to share it.

I remembered I bought SIM7000E module before June 2017 and the module firmware is not very updated. So many AT command stated in manual can't be used. Been followed your guide on firmware update tool, I found that IMEI on module was missing after firmware upgrade. 

To solve this, you will need restore QCN file attached follow the instruction under section **How to update xQCN file if necessary** in _How to update SIM7000 FW version with QPST.pdf_. The QCN file is standard and works for all SIM7000 series. Then run AT+SIMEI=<your IMEI number>. You can find IMEI number on top of SIM7000 module. Hope it helps.

[MDM9206_SIM7000E_P1.02_20180118.xqcn.zip](https://github.com/botletics/SIM7000-LTE-Shield/files/1902242/MDM9206_SIM7000E_P1.02_20180118.xqcn.zip)

** Better to confirm with SIMCOM engineer if the file works for all SIM7000.

botletics, 2018-04-12:
Thanks for your feedback! I will look into this.

closed 2018-06-04 by botletics

gmgaston, 2019-02-27:
While running the IOT_Example provided, it runs down to the point where it tries to get the IMEI and then freezes.   I looked at the Adafruit_FONA.cpp file that is supplied on this github site.  It looks like it is expecting a 15 character IMEI but the IMEI on my Hologram SIM is 19 characters long.  I tried to adjust both the .ino and the .cpp file but still doesn't work.  I might not have done the adjustments correctly. 
Any help to get my SIM7000-LTE-Shield working will be appreciated. 
Thanks
Greg

adam6806, 2019-02-28:
I just added a ticket before I saw this comment from @gmgaston I'm having the exact same issue. I'm going to see if I can make the adjustment to the code as well but I would definitely appreciate a fix here. It is strange that the LTE demo worked perfectly but the iot example isn't. Wouldn't the both use the same library?

gmgaston, 2019-02-28:
I may have been mistaken about the length of the IMEI. I at first thought the IMEI was the 19 digit number on the hologram SIM card but that is not correct.

I do stand by my comment that I still can't get past where the IOT _Example tries to fetch the IMEI. I just hangs there.
The LTE_Demo does work OK.

Thanks
Greg

adam6806, 2019-02-28:
@gmgaston why don't we move this over to my issue then instead of this closed one. I did come to the same conclusion about the imei length. One other thing I noticed is that the lte demo has a reply buffer array and the iot example does not. reply buffer is used in that method. maybe there's something there.

gmgaston, 2019-02-28:
Adam:
OK. I don't know much about the issues blog. I just found it when I did a Google search for botletics/SIM7000-LTE-Shield and IMEI.

I will be spending some time on this in next few days. If I find something I'll let you know ..

Thanks
Greg

adam6806, 2019-02-28:
@gmgaston Link to the issue is here. If you find anything please post there. https://github.com/botletics/SIM7000-LTE-Shield/issues/75

botletics, 2019-03-04:
Make sure your board has a LiPo battery connected to it, and with proper polarity (red wire on the left side of the connector when you plug it in). Usually weird characters in serial monitor means there's a power issue.

hasikp1, 2020-06-30:
@botletics do you have the SIM7000C firmware with you? I used the waveshare SIM7000C firmware but it shows as B08SIM7000.
@bengchet I used the method you mentioned and restored a IMEI of SIM7000E it was successfull, but when I used it in the above mentioned SIM7000C firmware shows as B08SIM7000E. Do you have a qcn for SIM7000C