botletics/SIM7000-LTE-Shield #19: No External MCU Needed?

TangoJLabs, 2018-06-01:
SIMcom released a media statement [1] earlier this year about the SIM7000G, and it suggests that no external MCU is needed:

>4.   Supports secondary development and saves costs: Without the need for an external MCU and by fully using the resources of SIM800/SIM900 wireless modules, it enables function to port and embeds programs that otherwise would have been implemented by the external MCU into the wireless module, achieving savings in costs. Moreover, RTOS multithreading, messages and other technologies are used for more efficient development. API interfaces of hardware are used for reduced difficulties in development.

It looks like you use an Arduino in all of your examples, but have you been able to use the shield standalone, or is the shield configuration limiting that feature of the SIM7000?

[1] http://www.simcomm2m.com/phone/En/media/detail.aspx?id=206

botletics, 2018-06-01:
Hmmm, interesting. I haven't seen that, but it would be super cool to figure out how to develop custom firmware for it! I'll ask SIMCom about that.

Thanks for the insight!

TangoJLabs, 2018-06-01:
Great, if you are able to make progress, please let us know!

closed 2018-06-01 by TangoJLabs