botletics/SIM7000-LTE-Shield #20: Generic MQTT

ram1505, 2018-06-02:
Hi,
This might be pretty lame to ask, but how do I use your sample code with my existing own MQTT server and not with Adafruit or CloudMQTT as described in the examples?

botletics, 2018-06-02:
You can try setting up the code for CloudMQTT and change the credentials to match your MQTT server. It should work for most cases. However, you should also check the line

`if (!fona.MQTTconnect("MQIsdp", MQTT_CLIENT, MQTT_USERNAME, MQTT_KEY)) Serial.println(F("Failed to connect to MQTT broker!"));`

since some MQTT servers just use "MQTT" instead of "MQIsdp".

closed 2018-06-02 by botletics

ram1505, 2018-06-04:
Hi,
I tried using both CloudMQTT as well as MQTT AIO. I'm getting the following error:

```

	---> AT+CIPSHUT
	<--- SHUT OK
	---> AT+CIPMUX=0
	<--- OK
	---> AT+CIPRXGET=1
	<--- OK
AT+CIPSTART="TCP","x.x.x.x","1883"
	<--- OK
	<--- CONNECT OK
	<--- > 
	---> 16 27 0 6 77 81 84 84 3 130 0 15 0 15 56 54 53 50 51 52 48 51 48 48 54 51 53 50 50 48 44 32 48 46 48 26
	<--- OK
Failed to connect to MQTT broker!
Publishing GPS data: 0.000000, 0.000000
	<--- > 
	---> 48 38 0 18 100 101 118 105 99 101 47 99 111 111 114 100 105 110 97 116 101 115 48 46 48 48 48 48 48 48 44 32 48 46 48 48 48 48 48 48 26
	<--- SEND OK
	---> AT+CIPCLOSE
	<--- ERROR
Failed to close connection!
Waiting for 10 seconds
```

botletics, 2018-06-04:
I just tried it on mine and it works great. Exactly what hardware are you using?

ram1505, 2018-06-04:
I've a starter-kit with SIM7000E, B05 with an Arduino Leonardo on it. My MQTT server is based on mosquitto, no username and password for testing purpose. MQTT is hosted on port 1883.

botletics, 2018-06-04:
Did you change "MQIsdp" to "MQTT" in the IoT_Example sketch?

ram1505, 2018-06-04:
Yes, I did. When the script sends this 
`16 27 0 6 77 81 84 84 3 130 0 15 0 15 56 54 53 50 51 52 48 51 48 48 54 51 53 50 50 221 61 93 31 173 117 26`
as connect, I don't see an acknowledgement packet. I tried the above with PacketSender client too and I don't see an acknowledgement packet in response.

botletics, 2018-06-04:
You said it doesn't even work with CloudMQTT?

botletics, 2018-06-04:
Since you don't have any username and password, are you using double quotes like

MQTT_USERNAME = "";
MQTT_KEY = "";

ram1505, 2018-06-04:
Mine, I didn't try with other service providers as I want this to work with my own MQTT server. And yes, I use double quotes for 
MQTT_USERNAME = "";
MQTT_KEY = "";

ram1505, 2018-06-04:
Should I populate this with my IMEI number? 

char MQTT_CLIENT[16] = " ";  // We'll change this to the IMEI

And, the packet conversion for CONNECT doesn't seem to show the protocol MQTT correctly:

```
6 27 0 6 77 81 84 84 3 130 0 15 0 15 56 54 53 50 51 52 48 51 48 48 54 51 53 50 50 48 48 44 48 46 48 26

I don't see a 4D 51 54 54 = M Q T T 
```

ram1505, 2018-06-04:
I just tried with CloudMQTT and I get the following error:

```
2018-06-04 14:35:27: New client connected from 111.xx.xx.xx as 865xxxxxxxxxxxx (c1, k15, u'wrwoumgj').
2018-06-04 14:35:29: Socket error on client 865xxxxxxxxxxxx, disconnecting.
```

botletics, 2018-06-04:
Sorry, been really busy and trying to keep up with Github. So have you tried CloudMQTT with "MQIsdp" (the default configuration of the IoT_Example sketch)? That should work, provided that you supply the right credentials and port in the #define section of the code.

ram1505, 2018-06-04:
Yes, thats correct. I used MQIsdp and thats when I got the above "Socket Error". I gave up on MQTT and was trying to do a HTTP Post and then realised that the code is meant for HTTP and not for HTTPS. My server is HTTPS.

botletics, 2018-06-04:
And for CloudMQTT you used the server port, not the SSL port right?

ram1505, 2018-06-04:
That's right too.

botletics, 2018-06-04:
Hmm, very strange that the default code won't work for you on CloudMQTT because it works for me and others have stated that it works for them as well.  Not sure what's going on to be honest.

coddingtonbear, 2018-10-21:
I'm sure @ram1505 has figured this out by now, but in case somebody else comes across this same problem, it might be that you're using the wrong value for `MQTT_KEY`.  What should be used for `MQTT_KEY` is shown in the CloudMQTT dashboard as "password", *not* "API key".

hjames9, 2019-09-09:
@ram1505 @botletics Did you ever figure out a solution? I'm running into the exact same problem making a generic MQTT connection.

I've tried several of the SIM7000 MQTT examples (seems like the FONA library has several MQTT apis) with the same results.  Can someone point me to a short code snippet that's known to work?

hjames9, 2019-09-09:
@ram1505 @botletics Actually I was just able to get it working with a generic MQTT server directly using the Adafruit MQTT Fona library.

i.e.
Adafruit_MQTT_FONA and Adafruit_MQTT_Publish
`
Adafruit_MQTT_FONA mqtt(fona, mqttServer, mqttPort);
Adafruit_MQTT_Publish feed(&mqtt, "yourMQTTTopic");
if(mqtt.connected() || ((ret = mqtt.connect()) == 0) {
  feed.publish("your message");
} else {
  Serial.print(F("Unsuccessful MQTT connection to server: ")); 
  Serial.println(mqtt.connectErrorString(ret));
}
`