botletics/SIM7000-LTE-Shield #21: B05 Firmware

johanso-au, 2018-06-03:
Has there been a formal release of B05 as yet?

I noticed that Telstra (Australia) have issued an approval for the 7000E module for firmware rev. B05.

Reference: [http://simcomm2m.com/UploadFile/TechnicalFile/SIM7000E_Telstra_Certificate_2018.PDF](http://simcomm2m.com/UploadFile/TechnicalFile/SIM7000E_Telstra_Certificate_2018.PDF) (may require a login?)

Also worth noting that a Telstra approval is good news for shields sold in Australia.

APN Settings for Telstra:

`fona.setGPRSNetworkSettings(F("telstra.internet")); // For Telstra (AU) SIM card - CAT-M1 (Band 28)`

botletics, 2018-06-03:
I believe B05 firmware is still under development but I'll ask. And yes, good news for Australia, although the module already works there.

johanso-au, 2018-06-03:
Great. It would be good to leave this issue open to track B05 impacts once released.

Approval is very good news, as Telstra is known to block the Type Allocation Code of non-approved devices to protect the integrity of the network.

botletics, 2018-06-03:
Yep that's what I had in mind :)

botletics, 2018-06-04:
Just uploaded B05 firmware for SIM7000E which you can [find here](https://github.com/botletics/SIM7000-LTE-Shield/tree/master/SIM7000%20Documentation/Firmware%20Updater%20Tool).

closed 2018-06-04 by botletics