botletics/SIM7000-LTE-Shield #23: invalid conversion from 'const char*' to 'char*' [-fpermissive]

qingfeng-zhang, 2018-06-04:
There got error " invalid conversion from 'const char*' to 'char*' [-fpermissive] " when I compiled your examples by  using your latest version 

In member function 'boolean Adafruit_FONA::postData(const char*, const char*, const char*, const char*)'

Adafruit_FONA.cpp error: invalid conversion from 'const char*' to 'char*' [-fpermissive]

   if (! sendCheckReply(body, ok_reply, 10000)) ,
anyone met the problem?

botletics, 2018-06-04:
What version of the Arduino IDE do you have?

per1234, 2018-06-04:
Which board do you have selected from the **Tools > Board** menu?

qingfeng-zhang, 2018-06-04:
ESP32  DEV,   we have used your previous version  without problem  for a few months

botletics, 2018-06-04:
OK, I just uploaded new files, please try the library again. It compiles fine on ESP32 now.

closed 2018-06-04 by botletics