botletics/SIM7000-LTE-Shield #26: How point out problem by at commands?

makotaka, 2018-06-17:
I am testing "SIM7XXX_LTE_Demo", but  unfortunately, In my environment, "_[2] Post to dweet.io via LTE CAT-M / NB-IoT (SIM70000)_" returns error code 601.
It seems network error, so I used AT commands to find out the reason, but I couldn't find out yet. 
It seems on roaming network, but IP function couldn't work. What I should try next ?

```
FONA> 2
	---> AT+HTTPTERM
	<--- ERROR
	---> AT+HTTPINIT
	<--- OK
	---> AT+HTTPPARA="CID",1
	<--- OK
	---> AT+HTTPPARA="URL","dweet.io/dweet/for/865233030122643?temp=600.24&batt=3600"
	<--- OK
	---> AT+HTTPACTION=0
	<--- OK
HTTP status: 601
Data length: 0
Failed to complete HTTP GET...

```

```
at+cgnapn
+CGNAPN: 1,"apn.konekt.io"
OK
at+cops?
+COPS: 0,0,"440 20 Hologram",8
OK
at+creg?
+CREG: 0,5
OK
AT+CIPPING="8.8.8.8"
ERROR
```

botletics, 2018-06-17:
Hi, most likely you didn't use the"G" command first.

botletics, 2018-06-18:
No, if you run it you will also see that it's not just enabling GPRS but setting the APN and bearer config. Check out [the demo wiki](https://github.com/botletics/SIM7000-LTE-Shield/wiki/Shield-Demo) which shows the 2G version and a slightly simplified version for the SIM7000 without the unnecessary GPRS commands.

closed 2018-06-18 by botletics

makotaka, 2018-06-20:
Thanks, I could do it.

MKorostoff, 2019-02-06:
https://github.com/botletics/SIM7000-LTE-Shield/issues/26#issuecomment-398079402 solved the issue I was having

JGLJGL, 2020-01-23:
Hi, I do try G comand first, so it enables data correctly, but then, it's not able to do neither the GET nor the POST. 
Any idea which the problem could be?

`09:32:51.192 -> FONA> G
09:32:53.534 -> 	---> AT+CIPSHUT
09:32:53.568 -> 	<--- SHUT OK
09:32:53.601 -> 	---> AT+CGATT=1
09:32:57.293 -> 	<--- OK
09:32:57.326 -> 	---> AT+SAPBR=3,1,"CONTYPE","GPRS"
09:32:57.359 -> 	<--- OK
09:32:57.627 -> 	---> AT+SAPBR=3,1,"APN","ep.inetd.gdsp"
09:32:57.660 -> 	<--- OK
09:32:57.693 -> 	---> AT+CSTT="ep.inetd.gdsp","",""
09:32:57.759 -> 	<--- OK
09:32:57.792 -> 	---> AT+SAPBR=3,1,"USER",""
09:32:57.826 -> 	<--- OK
09:32:57.859 -> 	---> AT+SAPBR=3,1,"PWD",""
09:32:57.892 -> 	<--- OK
09:32:57.959 -> 	---> AT+SAPBR=1,1
09:32:57.959 -> 	<--- OK
09:32:58.026 -> 	---> AT+CIICR
09:32:58.026 -> 	<--- OK
09:32:58.026 -> FONA> i
09:33:00.869 -> 	---> AT+CSQ
09:33:00.902 -> 	<--- +CSQ: 7,99
09:33:00.902 -> RSSI = 7: -100 dBm
09:33:00.902 -> FONA> i
09:33:04.622 -> 	---> AT+CSQ
09:33:04.655 -> 	<--- +CSQ: 7,99
09:33:04.655 -> RSSI = 7: -100 dBm
09:33:04.655 -> FONA> n
09:33:05.932 -> 	---> AT+CGREG?
09:33:05.999 -> 	<--- +CGREG: 0,5
09:33:05.999 -> Network status 5: Registered roaming
09:33:05.999 -> FONA> 2
09:33:13.127 -> 	---> AT+HTTPTERM
09:33:13.160 -> 	<--- OK
09:33:13.194 -> 	---> AT+HTTPINIT
09:33:13.227 -> 	<--- OK
09:33:13.261 -> 	---> AT+HTTPPARA="CID",1
09:33:13.294 -> 	<--- OK
09:33:13.328 -> 	---> AT+HTTPPARA="URL","http://dweet.io/dweet/for/865234030907363?temp=28.70&batt=3600"
09:33:13.429 -> 	<--- OK
09:33:13.464 -> 	---> AT+HTTPACTION=0
09:33:13.497 -> 	<--- OK
09:33:23.493 -> Failed to complete HTTP GET...
09:33:23.526 -> 	---> AT+HTTPTERM
09:33:23.560 -> 	<--- ERROR
09:33:23.595 -> 	---> AT+HTTPINIT
09:33:23.628 -> 	<--- ERROR
09:33:23.628 -> Failed to complete HTTP GET...
09:33:23.695 -> 	---> AT+HTTPTERM
09:33:23.695 -> 	<--- ERROR
09:33:23.762 -> 	---> AT+HTTPINIT
09:33:23.795 -> 	<--- ERROR
09:33:23.795 -> Failed to complete HTTP POST...
09:33:23.795 -> FONA> `

botletics, 2020-01-29:
When you go to dweet.io/get/latest/dweet/for/865234030907363 after you try to do a GET or POST, does the data show up?