botletics/SIM7000-LTE-Shield #27: Couldn't find FONA

AustinGregory223, 2018-06-24:
Working with SIm 7000c LTE DFRobot sheild and am getting this response "Couldn't find FONA" in the serial moniter

AustinGregory223, 2018-06-24:
/***************************************************
  This is an example for our Adafruit FONA Cellular Module

  Designed specifically to work with the Adafruit FONA
  ----> http://www.adafruit.com/products/1946
  ----> http://www.adafruit.com/products/1963
  ----> http://www.adafruit.com/products/2468
  ----> http://www.adafruit.com/products/2542

  These cellular modules use TTL Serial to communicate, 2 pins are
  required to interface
  Adafruit invests time and resources providing this open source code,
  please support Adafruit and open-source hardware by purchasing
  products from Adafruit!

  Written by Limor Fried/Ladyada for Adafruit Industries.
  BSD license, all text above must be included in any redistribution
 ****************************************************/

/*
THIS CODE IS STILL IN PROGRESS!

Open up the serial console on the Arduino at 115200 baud to interact with FONA

Note that if you need to set a GPRS APN, username, and password scroll down to
the commented section below at the end of the setup() function.
*/
#include "Adafruit_FONA.h"

#define FONA_RX 2
#define FONA_TX 3
#define FONA_RST 4

// this is a large buffer for replies
char replybuffer[255];

// We default to using software serial. If you want to use hardware serial
// (because softserial isnt supported) comment out the following three lines 
// and uncomment the HardwareSerial line
#include <SoftwareSerial.h>
SoftwareSerial fonaSS = SoftwareSerial(FONA_TX, FONA_RX);
SoftwareSerial *fonaSerial = &fonaSS;

// Hardware serial is also possible!
//  HardwareSerial *fonaSerial = &Serial1;

// Use this for FONA 800 and 808s
Adafruit_FONA fona = Adafruit_FONA(FONA_RST);
// Use this one for FONA 3G
//Adafruit_FONA_3G fona = Adafruit_FONA_3G(FONA_RST);

uint8_t readline(char *buff, uint8_t maxbuff, uint16_t timeout = 0);

uint8_t type;

void setup() {
  while (!Serial);

  Serial.begin(115200);
  Serial.println(F("FONA basic test"));
  Serial.println(F("Initializing....(May take 3 seconds)"));

  fonaSerial->begin(4800);
  if (! fona.begin(*fonaSerial)) {
    Serial.println(F("Couldn't find FONA"));
    while (1);
  }
  type = fona.type();
  Serial.println(F("FONA is OK"));
  Serial.print(F("Found "));
[... 788 lines cut ...]

      //Serial.print(c, HEX); Serial.print("#"); Serial.println(c);

      if (c == '\r') continue;
      if (c == 0xA) {
        if (buffidx == 0)   // the first 0x0A is ignored
          continue;

        timeout = 0;         // the second 0x0A is the end of the line
        timeoutvalid = true;
        break;
      }
      buff[buffidx] = c;
      buffidx++;
    }

    if (timeoutvalid && timeout == 0) {
      //Serial.println(F("TIMEOUT"));
      break;
    }
    delay(1);
  }
  buff[buffidx] = 0;  // null term
  return buffidx;
}

botletics, 2018-06-24:
You're not using the library here (see the wiki), so of course it won't work. Beyond that I simply don't have time to give technical assistance for products that aren't my own.

closed 2018-06-24 by botletics

AustinGregory223, 2018-06-24:
Thankyou that’s fine

AustinGregory223, 2018-06-24:
Hello Timothy, I am 12 years old and I am entering a ICT comp using Arduino. If I was to buy your shield would it work in Australia. Would that than help the code to upload from your library because using the DFRobot shield I get the same response in my serial monitor every time “Coudnt find FONA”
Thanks
Austin

Closed #27<https://github.com/botletics/SIM7000-LTE-Shield/issues/27>.

AustinGregory223, 2018-06-24:
Hello Timothy, I am 12 years old and I am entering a ICT comp using Arduino. If I was to buy your shield would it work in Australia. Would that than help the code to upload from your library because using the DFRobot shield I get the same response in my serial monitor every time “Coudnt find FONA”
Thanks
Austin

botletics, 2018-06-24:
You need to define the right pins for your shield before it can find it. You can order my shield but lead times are longer because it's out of stock and I'm planning production for it.

Jen-n14, 2023-05-10:
Hi!
I would like to ask for help about my project. I am making a flood monitoring system with sms notification usinh arduino uno r3, gsm 900a module and water level sensor. Below is the wiring of my system. 
I tried uploading the code and it was successful yet my serial monitor says couldn't find fona. 
May I know what can I do to solve the issue?

Your help will be greatly appreciated💗

Code:
//Tech Trends Shameer
//Flood Alert Call and SMS

#include <SoftwareSerial.h>
#include "Adafruit_FONA.h"
 
 
#define FONA_RX            2
#define FONA_TX            3
#define FONA_RST           4
    
 

int resval = 0;  // holds the value
int respin = A3; // Water Level Sensor Pin 

#define FONA_RI_INTERRUPT  0
SoftwareSerial fonaSS = SoftwareSerial(FONA_TX, FONA_RX);
 
Adafruit_FONA fona = Adafruit_FONA(FONA_RST);

char PHONE_1[21] = ""; // Enter your Mobile Number 

char tempalert[141]= "Flood Alert"  ; 
void setup()
{
//  pinMode(gas_sensor_pin,INPUT);
  Serial.begin(115200);
//  dht.begin();
  Serial.println(F("Initializing....(May take 3 seconds)"));
  delay(5000);
  fonaSS.begin(9600); // if you're using software serial
  if (! fona.begin(fonaSS)) {           // can also try fona.begin(Serial1) 
    Serial.println(F("Couldn't find FONA"));
    while (1);
  }

   fona.print ("AT+CSMP=17,167,0,0\r");
    Serial.println(F("FONA is OK"));
   //fona.sendSMS(PHONE_1, welcomemessage);
//    pinMode(gas_sensor_pin, INPUT);
 
}

void loop(){
  
 
  resval = analogRead(respin); 
  
  
  Serial.print("Water Level:");
  Serial.println(resval);

  
  if(resval>200 )
  {    
     Serial.println("Flood Alert");  
     make_multi_call();
     send_multi_sms();
   }
  
  }

void send_multi_sms()
{
  if(PHONE_1 != ""){
    Serial.print("Phone 1: ");
    fona.sendSMS(PHONE_1,tempalert);
    delay(20000);
  }
   
   
}
void make_multi_call()
{
  if(PHONE_1 != ""){
    Serial.print("Phone 1: ");
    make_call(PHONE_1);
    delay(20000);
  }
   
}

void make_call(String phone)
{
    Serial.println("calling....");
    fona.println("ATD"+phone+";");
    delay(20000); //20 sec delay
    fona.println("ATH");
    delay(1000); //1 sec delay
}

Wiring :
S -> A3
+ -> 5v
- -> GND
Rx -> tx of arduino 
Tx -> rx of arduino
VCC -> 5v
GND -> GND

Jen-n14, 2023-05-10:
/