botletics/SIM7000-LTE-Shield #28: 3g/4g link  with ESP32 for quad-copter

mkeyno, 2018-06-24:
hi , sorry to open ticket for my question but actually I've intend to use 3g/4g link  along side with my Ardupilot compatible hardware which is based on ESP32, but I don't know which SIM series suitable for this matter whereas  any drone can fly above the telephone towers  and hens communication quality may drop down  , also co-working different radio frequencies (3g/4g, wifi, telemetry)  would be an issue for itself , so I really appreciated to have your advises for this matter

mkeyno, 2018-07-11:
thanks for reply , but may I ask for your possible experience about SIM7000 SPI interface?is it possible to connect camera with SPI interface directly to module and launch video stream while control other feature module with AT command?

botletics, 2018-07-11:
SIM7000 doesn't have SPI interface as far as I know.