botletics/SIM7000-LTE-Shield #29: LTE_Demo not compiling

elaghouh, 2018-06-29:
Hello,

I'm trying to use your library with a SIM5320A shield, but when I go to compile the LTE_Demo example, I get the following: 

>/var/folders/pq/c3qw6h7x0zd76hnxmt42d56m0000gp/T//ccMbhD8M.ltrans0.ltrans.o: In function `loop':
>/var/folders/pq/c3qw6h7x0zd76hnxmt42d56m0000gp/T/arduino_modified_sketch_679301/LTE_Demo.ino:799: undefined reference to `Adafruit_FONA_3G::enableGPRS(bool)'
>/var/folders/pq/c3qw6h7x0zd76hnxmt42d56m0000gp/T/arduino_modified_sketch_679301/LTE_Demo.ino:810: undefined reference to `Adafruit_FONA_3G::enableGPRS(bool)'

I don't understand why it is doing that. Any help would be appreciated!

botletics, 2018-06-29:
You need to uncomment

//#define SIMCOM_3G // SIM5320A/E

and comment out the default 

#define SIMCOM_7000  // SIM7000A/C/E/G

Let me know if that works!

botletics, 2018-06-29:
Note that you will also need to set the pin definitions appropriately based on the specific hardware you're using.

botletics, 2018-06-29:
Oh wait you're right, there is an error. I must've changed the library structure. Let me fix that real quick...

botletics, 2018-06-29:
OK fixed it, please change out your [Adafruit_FONA.h file](https://github.com/botletics/SIM7000-LTE-Shield/blob/master/Code/Adafruit_FONA.h) with the one I just uploaded and it should compile. Was a stupid error I made, sorry.

closed 2018-06-29 by botletics