botletics/SIM7000-LTE-Shield #32: Close (but no cigar!)

seangavin, 2018-08-25:
Hi!
Im using the TinySine SIM5320A shield on a Uno board.
I removed old versions of the Fona Library and downloaded your latest one from github yesterday(8/24/18).
I changed the #define to SIMCOM_3G.
I updated the APN info since I'm using an AT&T sim card that belongs to a spare (working)phone I have.
It compiled and uploaded fine and ran, can turn on GPRS, can check the network etc, but I just can't get the dweet to work. I think I'm really close. 

See the Serial output below.

Any words of wisdom you can share?

FONA> 
OPL DONE

PNN DONE

SMS DONE

+VOICEMAIL: INIT_STATE, 1, 3

PB DONE

+STIN: 25
n
	---> AT+CREG?
	<--- +CREG: 0,1
Network status 1: Registered (home)
FONA> i
	---> AT+CSQ
	<--- +CSQ: 13,99
RSSI = 13: -88 dBm
FONA> G
	---> AT+CGATT=1
	<--- OK
	---> AT+CGSOCKCONT=1,"IP","wap.cingular"
	<--- OK
	---> AT+CGAUTH=1,1,"wap@cingulargprs.com","cingular1"
	<--- OK
	---> AT+CIPMODE=1
	<--- OK
	---> AT+NETOPEN=,,1
	<--- Network opened
FONA> 3
	---> AT+CHTTPSSTART
	<--- OK
Waiting 1s to ensure connection...
	---> AT+CHTTPSOPSE="www.dweet.io",443,2
	<--- OK
Waiting 1s to make sure it works...
	---> AT+CHTTPSSEND=81
	<--- >
	---> GET /dweet/for/014682003952479?temp=678.96&batt=3700 HTTP/1.1
Host: dweet.io

	<--- OK
	---> AT+CHTTPSSEND
	<--- OK
	---> AT+CHTTPSRECV?
	<--- +CHTTPSRECV: LEN,0
	---> AT+CHTTPSRECV=0
	<--- ERROR
	---> AT+CHTTPSCLSE
	<--- +CHTTPSSEND: 8
Failed to complete HTTP/HTTPS request...
FONA> 
OK

botletics, 2018-08-25:
Hi there, from [this dweet link](http://dweet.io/get/latest/dweet/for/014682003952479) it looks like it actually did post to dweet successfully. However, it looks like it's just not reading back the server response properly. How many times have you tried the command?

seangavin, 2018-08-26:
Thanks for the really quick reply. You appear to be correct(as always!). When I open a browser and type: http://dweet.io/follow/014682003952479, you can see the dweet there. So it worked, it's the server response, or receiving the server response or the interpretation of the server response that is the problem.
Am I reading it correctly, that it is receiving nothing back?

To answer your question- I've run the program/command dozens of times and am consistently getting the same result.

botletics, 2018-08-26:
Hmmm I'm almost thinking it might be the AT+CHTTPSCLSE command being run too soon before the CHTTPSSEND command finishes completely. I tried editing the .cpp file so that it should wait for the +CHTTPSSEND: 0 response before proceeding, so hopefully that works. However, unfortunately I don't have a SIM5320 board to test with so if you don't mind please let me know if it works for you. Just replace the [Adafruit_FONA.cpp file](https://github.com/botletics/SIM7000-LTE-Shield/blob/master/Code/Adafruit_FONA.cpp) with the one you currently have and re-upload.

seangavin, 2018-08-27:
I'm afraid that didn't work. Same as before. I think the initial problem is the "receive" before the "close".
So I added a 2 second delay before the "---> AT+CHTTPSRECV?" and that seemed to really help. It tells me that the replybuffer is 379 in length, which is the correct length for the dweet response.  And it prints out the first 255 characters of it.So that's great progress.
Then  I decided to print out the contents of the replybuffer to see everything being received and even though it says it is 379 in length, after 254 bytes, it's all garbage. I noticed in your example, you specify replybuffer[255], so you're stomping over memory you don't own. 
I bumped that up to 400, but now I'm hitting up against memory resource problems. More to follow- just too late tonight to work on it.

botletics, 2018-08-27:
The problem is trying to use the CHTTPSRECV before the response before the +CHTTPSSEND: 0 reply. So yes, a delay could work there. However, I will obviously want to streamline that logic so it won't be depending on the delay alone.

And yes, I am aware of the problem where it only prints out part of the server response, but I don't think I want to fix it because on Arduino Uno the IoT_Example sketch barely runs for doing POST requests, etc. If the buffer sizes are too large the Arduino can crash. Also, no server response should be more than 255 long anyway. If it is, it's like reading a book every time. For any good product deployment it should only respond with a success flag and maybe some other info but the dweet thing is just for testing.

botletics, 2018-08-29:
Now that I have a SIM5320A board I can see if I can fix the issue.

botletics, 2018-08-29:
I just updated the library and the [LTE_Demo sketch](https://github.com/botletics/SIM7000-LTE-Shield/blob/master/Code/examples/LTE_Demo/LTE_Demo.ino) to include the changes. All I had to do was add a 1s delay after the "+CHTTPSSEND: 0" response and it works great!

closed 2018-08-29 by botletics