botletics/SIM7000-LTE-Shield #36: Adafruit_Fona_LTE does not name a type

aggounix, 2018-09-16:
I am trying the tutorial on https://www.hackster.io/botletics/real-time-2g-3g-lte-arduino-gps-tracker-iot-dashboard-01d471
when I try to upload the Iot_Example to the arduino I get this error:
Adafruit_FONA_LTE does not name a type
how can I fix this probleme
I am new to arduino

botletics, 2018-09-16:
That error means you didn't install the right library. It needs to be the one from here. Please [follow these instructions to install](https://github.com/botletics/SIM7000-LTE-Shield/wiki/Arduino-IDE-Setup)

closed 2018-09-16 by botletics

keirthsiva, 2018-12-21:
@botletics Hi, I'm working with the SIM7000A to send texts to my phone. I have a hologram sim. I have followed the instructions and am using the example code for LTE_Demo, and downloaded the files/libraries from your link above, however I am still getting the error message of: "  'Adafruit_FONA_LTE' does not name a type ". Can you please let me know how I can go about solving this issue. Thanks.

botletics, 2018-12-21:
Please go to your Arduino libraries folder and delete any library called Adafruit FONA. Then [follow these instructions](https://github.com/botletics/SIM7000-LTE-Shield/wiki/Arduino-IDE-Setup) to install the library on this GitHub repo. Again, don't go to Adafruit's GitHub repo to get the library! It won't compile. You should see the LTE_Demo appear in the examples within Arduino IDE if you installed the right library.