botletics/SIM7000-LTE-Shield #37: Couldn't find FONA

aggounix, 2018-09-16:
I added the library and I uploaded the library but now I have another issue in the port monitor : Couldn't find FONA
I am using Hologram Sim card

botletics, 2018-09-16:
It's probably because you're using different hardware. Please keep your posts under the same issue and unfortunately I can't help everyone trying to use this library with their own hardware. [Look here if you want the official Botletics SIM7000 shield kit](https://www.botletics.com/products/sim7000-shield)

closed 2018-09-16 by botletics

aggounix, 2018-09-17:
I bought the the same 7000E shield in the tutorial from botletics and using it with your tutorial here:
https://www.hackster.io/botletics/real-time-2g-3g-lte-arduino-gps-tracker-iot-dashboard-01d471
With this library and no other hardware is used here

botletics, 2018-09-17:
The default initialization code doesn't wait for the module to turn on, so you have to press the Arduino's reset button after you see the green PWR LED for the SIM7000 the first time you start it up.

aggounix, 2018-09-22:
I think there is a probleme with the SIM7000 because the PWR LED is never green (the status LED in [Shield pinout.](https://cdn.instructables.com/F4D/3G87/JFIIS97S/F4D3G87JFIIS97S.LARGE.jpg))
May be it is a problem with the SIM7000 shield

botletics, 2018-09-22:
That won't cause a compile error though.

aggounix, 2018-09-22:
this is not a compile error, the code is televersed to arduino and during execution it couldn't find the fona and I think because the shield is not powered up 
The Staatus LEd is not working at all

botletics, 2018-09-22:
Please email me at botletics@gmail.com so we can share images more easily.

aggounix, 2018-09-22:
Thanks  a lot, I just sent you the images to your mail

botletics, 2018-09-22:
Thanks. The issue is most likely because you didn't solder the headers onto the board at all, so they're not making connection.

aggounix, 2018-09-22:
How  can I solder the headers to the headers, it seems to me fragile I am afraid to break it, do you have images on how must be done

botletics, 2018-09-22:
Yes, please see [this wiki page](https://github.com/botletics/SIM7000-LTE-Shield/wiki/Assembly)

aggounix, 2018-09-22:
Should I do it with a soldering tool kit like this: https://learn.adafruit.com/adafruit-guide-excellent-soldering/tools

botletics, 2018-09-22:
You just need a soldering iron and solder. There are many options available online, like on Amazon.

aggounix, 2018-10-01:
Great News, I soldered the pins to the shield and I get the status LED turned ON, I tried the IOT_Example and after pressing arduino reset button, I get it work. Iget these message:

FONA is OK
Found SIM7000E (European)
	---> AT+GSN
	<--- xxxxxxxxxxxxxxxxxxx
Module IMEI: xxxxxxxxxxxxxxxxxxx
	---> AT+CGNSPWR?
	<--- +CGNSPWR: 0
	---> AT+CGNSPWR=1
	<--- OK
Turned on GPS!
	---> AT+CIPSHUT
	<--- SHUT OK
	---> AT+SAPBR=0,1
	**<--- ERROR**
Failed to disable GPRS!
	---> AT+CIPSHUT
	<--- SHUT OK
	---> AT+CGATT=1
	<--- 
Failed to enable GPRS, retrying...
	---> AT+CIPSHUT
	<--- 
	---> AT+CGATT=1
	<--- 

Is it normal to get error for enabling GPS and failing enabling GPRS

botletics, 2018-10-01:
Are you using a LiPo battery? Does the green "PWR" LED next to the module ever turn off when you try to enable data? Disabling data and getting an error is OK, but enabling afterwards should work as long as you're connected to the network.

aggounix, 2018-10-01:
I am using the usb connection to power the shield.  The "PWR" LED is always on. In the tutorial it is said the iot code must be able to send gps position to dweet normally. What am I doing wrong to not have the position on dweet

botletics, 2018-10-01:
The USB does not power the shield, as mentioned in the wiki. Please run the LTE_Demo sketch first and make sure you're connected to the network (use "n" command) and that everything else works (GPS, etc). Then you can move on to the IoT_Example sketch.

aggounix, 2018-10-01:
When I use th en command in LTE_Example I get this message:
Network status 2: Not registered (searching)
I am using a hologram sim card

botletics, 2018-10-01:
That would explain why you can't enable data. Please see [this wiki page](https://github.com/botletics/SIM7000-LTE-Shield/wiki/Connecting-to-Network) on getting connected.

mduiop, 2021-03-07:
I am using the SIM7000A module purchased from Amazon in the U.S..  I am attempting to run LTE_Demo on my Arduino Mega 2560.  I have a 3.7v battery connected to the battery terminal.  I am using the provided headers soldered onto the module.  Activated Hologram SIM.  I set my vars as such:

// For SIM7000 shield
#define FONA_PWRKEY 6
#define FONA_RST 7
//#define FONA_DTR 8 // Connect with solder jumper
//#define FONA_RI 9 // Need to enable via AT commands
#define FONA_TX 1 // Microcontroller RX
#define FONA_RX 0 // Microcontroller TX
//#define T_ALERT 12 // Connect with solder jumper

When I run this code on the Arduino with the SIM7000A I get:

. . .
Timeout: No response to AT... last ditch attempt.
	---> AT
	<--- 
. . .
	---> ATE0
	<--- 
Couldn't find FONA

botletics, 2021-03-07:
The issue looks like it's because you changed the pin definitions. Please load the unaltered LTE_Demo sketch and see if it works.

mduiop, 2021-03-07:
> The issue looks like it's because you changed the pin definitions. Please load the unaltered LTE_Demo sketch and see if it works. By default the code uses software serial with pins 10 and 11 for UART.

OK, I could have sworn I did, but to "humor you" I decided to do that yet again.  Well, it works now, so thank you very much for the quick reply.  (If you detect sarcasm it is self-deprecating . . . not directed at you . . . mocking myself).

sagarchowdhury, 2021-04-24:
i am also getting the same error couldnt find fona

mduiop, 2021-04-24:
> i am also getting the same error couldnt find fona

Make sure you have reception, and don't change the pin definitions. That worked for me. I used the scripts unaltered and they worked.

sagarchowdhury, 2021-04-24:
> > i am also getting the same error couldnt find fona
> 
> Make sure you have reception, and don't change the pin definitions. That worked for me. I used the scripts unaltered and they worked.

yes the the network led light is blinking after every 3 seconds, what i did differently is i used breadboard to connect the header with the module ,and connected each header pin with a jumper wire to the Arduino to simulate it like a shield ,do you think there may be loose connection because i didnt solder the header pins with the module

mduiop, 2021-04-24:
I'm no expert, but yes, that is a big possibility. Solder the header pins
and use it as a shield if you can. Requires a good connection.

sagarchowdhury, 2021-04-24:
> I'm no expert, but yes, that is a big possibility. Solder the header pins and use it as a shield if you can. Requires a good connection.
> […](#)

one strange thing i noticed is i am getting replies at 9600 baud rate in com port but it should ideally be 115200 ,when i am changing to the 115200 i am not finding anything ,the serial monitor is blank

mduiop, 2021-04-24:
Yeah, I'm pretty sure the serial monitor has to be set to the same baud
rate set in the script.