botletics/SIM7000-LTE-Shield #41: Firmware update failed

robertgrochowski, 2018-10-22:
Hello,

I have installed all the drivers, no unknown devices appear in Windows Device Manager.
I also followed upgrade instructions step by step, checked everything few times but I still can't manage to update because of "ConnectServer failed1, Please repower 1 DUT"

[image]

Thanks in advance

botletics, 2018-10-22:
Hi there,

To aid in assisting you, what hardware are you using? Do you have a 3.7V LiPo battery connected? It may be that your device is shutting off when trying to update. Updating firmware requires a lot of power.

robertgrochowski, 2018-10-22:
Hi,

Thank you for fast reply. To be honest, I use another SIM700E shield (exactly DFRobot Shield NB) but the producer doesn't include information how to upgrade it. Only Your tutorial seems to be suitable. 

I power it from power supply (12v through VIN) and the board works like a charm. 

I have tried also your 2nd tutorial to upgrade it through QPST Tool but unfornatelly I run into difficulties because I couln't select port from the list (the list was empty). I tried to run as administartor, restart my PC and board as well but that didn't help. Do you have any suggestions? 

Thanks in advance.

botletics, 2018-10-22:
Do you have any sort of indication that the module is turning off during the update process, especially right before it says to repower DUT?

botletics, 2018-10-22:
Could you show me a screenshot of what device manager looks like before you do the update? Go through literally each entry in device manager that's related to the SIM7000, right click, update driver, and point it to the drivers folder.

robertgrochowski, 2018-10-22:
Of course, I have done it for every SIM device.

[image]

botletics, 2018-10-22:
Hmm not sure, sorry.

closed 2018-10-22 by botletics

lutfisan, 2018-11-16:
> 
> 
> No I don't see any, but I managed to go a little bit further in the firmware update process.
> 
> Now it stucks at "waiting switch to download mode..."
> 
> I've noticed that after starting upgrade process, in device manager appear new device called "QHSUSB_BULK"
> [image]

is it solve? i have same problem

robertgrochowski, 2018-11-16:
Unfortunately, no.

royallyovershoot, 2019-11-02:
Hey, @robertgrochowski I know it's more than a year late but I encountered the same exact problem with DFRobot and figured out how to fix it. Feel free to PM me since I this is a Botletics chain :)

robertgrochowski, 2019-11-02:
@royallyovershoot Thanks for letting know. Can you post your solution here? It will help others as well.

royallyovershoot, 2019-11-02:
Ok, there are two test points on the board (TP1 and TP2) shown on the attached snippet of the schematic. Those need to be shorted together during update, so using a soldering iron you can solder two pin headers and use a shunt to short them when you are updating the firmware and then remove the shunts when you're done with the update. 

It's important to keep in mind that those two pins must be shunted before powering up the board.
I got to 100% but I got an error that it failed to back-up. But everything is working fine. This was included from the hardware design guide: [here](https://simcom.ee/documents/SIM7000x/SIM7000%20Hardware%20Design_V1.05.pdf)
On Page 20  under "Other interface" -> Boot_CFG it explains it. 

[image]

madatr, 2019-11-19:
> Ok, there are two test points on the board (TP1 and TP2) shown on the attached snippet of the schematic. Those need to be shorted together during update, so using a soldering iron you can solder two pin headers and use a shunt to short them when you are updating the firmware and then remove the shunts when you're done with the update.
> 
> It's important to keep in mind that those two pins must be shunted before powering up the board.
> I got to 100% but I got an error that it failed to back-up. But everything is working fine. This was included from the hardware design guide: [here](https://simcom.ee/documents/SIM7000x/SIM7000%20Hardware%20Design_V1.05.pdf)
> On Page 20 under "Other interface" -> Boot_CFG it explains it.
> [image]
> 
> [image]
> 
> [I removed the SIM7000 top metal cap]

MAN, YOU ARE A GOD SEND!!!!!
I've been stuck on this for two weeks and was at giving up point.
Thank you for taking up time to share this. 
and shame on manufacturers who don't share crucial information like this.

royallyovershoot, 2019-11-19:
@madatr Hah, I'm glad that it helped! It was a dick move from the manufacturers but hope you complete your project with ease now :)

hasikp1, 2019-11-19:
Hi, I have a SIM7000E module bought from aliexpress. It only have 7 pins as output, G-GND,R-RXD,T-TXD,K-PWRKEY,V-VCC,G-GND and S-SLEEP. I tried to update the firmware following the guide provided by @botletics. I also got the "QHSUSB_BULK" in device manager. Does this mean somehow I have to get boot_cfg and VDD_EXT shorted based on steps provided by @royallyovershoot. Is there any other alternatives?

madatr, 2019-11-20:
well... I can't tell you what to do exactly. The board i'm using has a USB connector provided with necessary hardware to facilitate that, the error/issue I had is that the update software wouldn't be able to switch the board to download mode so I had to force it using @royallyovershoot instructions.
I would recommend that you make sure that your board can communicate with the PC correctly first as the issue might be from that end.
btw, my board does not have the boot_cfg and vdd_ext as well, i just shorted the pins on the chip using a wire and had someone use the software to update it:
[image]

KerryRJ, 2020-06-11:
I have a waveshare NBT-IoT hat. I also got stuck at "waiting switch to download mode" and the QHSUSB_BULK device appeared in device manager. I just updated the driver in device manager from https://www.waveshare.com/wiki/File:SIM7X00-Driver.7z and the update process immediately continued.

hasikp1, 2020-06-15:
@madatr Are you using this module
[image]

semorem, 2020-09-16:
> I have a waveshare NBT-IoT hat. I also got stuck at "waiting switch to download mode" and the QHSUSB_BULK device appeared in device manager. I just updated the driver in device manager from https://www.waveshare.com/wiki/File:SIM7X00-Driver.7z and the update process immediately continued.

@KerryRJ hello there, i have the same issue that you did with the same hardware, but i already have the lastest dirvers installed. Did you just update the drivers? Did you do it during the firmware's update?

KerryRJ, 2020-09-16:
I updated the driver during the firmware update.

Kind Regards

Kerry Jones
Owning Member

Info-Overdrive cc
[IO Drive square logo - blue - small]

m. +27(0)612403717
e. kerry@iodrive.co.za<mailto:kerry@iodrive.co.za>
a. F3 Chief Mhlabunzima Rd., Pietermaritzburg, 3201

Website<http://www.iodrive.co.za/>        Email<mailto:kerry@iodrive.co.za>            IM<sip:kerry@iodrive.co.za>

semorem, 2020-09-16:
Thanks dude! It seems to be working!!!

BuiHao1996, 2020-10-30:
> Hello,
> 
> I have installed all the drivers, no unknown devices appear in Windows Device Manager.
> I also followed upgrade instructions step by step, checked everything few times but I still can't manage to update due to "ConnectServer failed1, Please repower 1 DUT" error.
> 
> [image]
> 
> Thanks in advance

Could you send me your software to upload SIM 7600CE ?

MaffooClock, 2021-01-27:
> No I don't see any, but I managed to go a little bit further in the firmware update process.
> 
> Now it stucks at "waiting switch to download mode..."
> 
> I've noticed that after starting upgrade process, in device manager appear new device called "QHSUSB_BULK"
> [image]

When I got to this, I simply installed the driver (from the same folder as all the other drivers), and as soon as it finished, QDL resumed automatically.

VeselaHouba, 2021-04-16:
I know this thread is closed, but just for reference:
Using waveshare sim7000E

1. Blew off dust of my old Windows Laptop
1. Plugged the module to USB, didn't see any issues with power consumption
1. Installed drivers as described in https://github.com/botletics/SIM7000-LTE-Shield/wiki/USB-Drivers
1. Following upgrade guide https://github.com/botletics/SIM7000-LTE-Shield/wiki/Updating-Firmware
1. Upgrade hanged on something like "Waiting to switch to download mode"
1. Opened device manager, installed latest driver from https://www.waveshare.com/wiki/File:SIM7X00-Driver.7z
1. Update tool picked-up by itself and continued the installation until finish

kjm1102, 2023-12-29:
More than 2 years after the last post, I might get lucky?
I've got a 7080G with 1951B05sim7080 firmware that doesn't work too well but the downloader wants to load 1951B04sim7080 which is even older.
I'd like to upgrade it to 1951B07sim7080 that runs great in another 7080G but I can't figure out how to copy it from the modem it's in.
So I'm looking for a link to where I might get some 1951B07sim7080 firmware

eisaev, 2024-01-02:
@kjm1102 Have you tried this?
https://github.com/Xinyuan-LilyGO/LilyGo-T-PCIE/tree/master/update_simxxxx_firmware/1951B08SIM7080

kjm1102, 2024-01-02:
tnx for the links. I grabbed the 3 files that this wretched flasher I got from waveshare says it needs & copied them to a folder on an E drive to match the byzantine path it needs (there seems to be no way to edit the path) but the usual story, "computer says no". Can you see what I've missed from the attached? I'm new to modem flashing.
[image]

eisaev, 2024-01-03:
> tnx for the links. I grabbed the 3 files that this wretched flasher I got from waveshare says it needs & copied them to a folder on an E drive to match the byzantine path it needs (there seems to be no way to edit the path) but the usual story, "computer says no". Can you see what I've missed from the attached? I'm new to modem flashing. ![noexist](https://private-user-images.githubusercontent.com/92007731/293807827-1b56f12e-0dc4-461c-9699-e51dee20c9a8.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDQyNjUxNzcsIm5iZiI6MTcwNDI2NDg3NywicGF0aCI6Ii85MjAwNzczMS8yOTM4MDc4MjctMWI1NmYxMmUtMGRjNC00NjFjLTk2OTktZTUxZGVlMjBjOWE4LmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAxMDMlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMTAzVDA2NTQzN1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTFhNzBmYjBjNjQ1OTE4Y2JlYWJmMDEyOGM4ZGRjNDQ0MGEzNmZiNTlmZDk3YjkxZDYzODE0ZGYzODYyOGNkNDcmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.E3oVFp0LmiDc1eoG_Inqj80HvkyyrKUKpkAZovtmJmI)

The flasher expects that there will be more subfolders in the path. Increase the size of the flasher window to see the full path and try to create the required folder structure.

kjm1102, 2024-01-03:
tnx Evgeniy.  I finally got the firmware folders where the updater tool expects to find them. 
1) At the beginning device manager shows 3 comports:
    SimTech HS-USB AT Port 9205 (COM27)
    SimTech HS-USB Diagnostics 9205 (COM29)
    Simtech HS-USB NMEA 9205 (COM28)
2) The tool starts on com 29, reads the imei & does a few other things then swaps to COM27 
3) There is a USB disconnect & the 3 comports ABOVE vanish to be replaced Qualcom HS-USBQDLoader 9008 (com30)
4) The tool reconnects on COM30 but timesout waiting for prog_firehose_nand.elf which is definitely present in the folder
Any thoughts on what I could try next?