botletics/SIM7000-LTE-Shield #42: Factory Image / Erase EFS

Thalhammer, 2018-10-26:
Hello,
It seems like I programmed a bug in my DAM App which causes the device to reset and then boot into Download/Sahara mode. In this mode I can download a firmware using qpst but it seems like the efs is untouched in this process. Is there a way to erase efs / recovery a factory image ?

Thalhammer, 2018-10-26:
Didn't work, the Uart AT port pins never really worked for me anyway and the Software AT COM is not available because the modem is in Download mode. I have contacted Simcom on this, but I'm not sure if I will get a response.

Thalhammer, 2018-10-26:
Maybe I didn't express myself good enough:
I'm not stuck in Download mode,
1. The module boots, tries to start my DAM App.
2. The DAM App crashes and the module resets into download mode.
3. I open QPST and can either do a Firmware Download or get a memory dump
4. After this the module reboots normally and tries to start my App and we are ack to step 1.

Initially I thought I had a bad bootloader, because I sometimes used the reset pin to reboot it, but I redownloaded the firmware and the problem persists. So I need to somehow delete the "custapp/cust_app.bin" file. However I have no Idea how to do this since EFS Explorer does not work in download mode and the time between boot and reset is to short to remove the file from efs.

botletics, 2018-10-26:
Sorry, unfortunately I wouldn't know. I haven't had any time to look into the DAM App.

Thalhammer, 2018-10-26:
As said I have contacted Simcom via E-mail but I'm not sure if they will reply. I got the impression that you have quite a good contact to them, maybe you can ask them about this issue ?

botletics, 2018-10-26:
They hardly answer my questions either, I have about 5 questions in the queue lol.

botletics, 2018-10-26:
No problem, it's nice to see people trying to contribute as well. So far I've been the only one slaving away at documenting this thing. That being said, I haven't gotten even the remotest chance to look at the DAM app feature.

Thalhammer, 2018-10-26:
Once you have the correct tools it actually works quite well, I'm currently working on reimplementing the functions in simcom's provided libraries to allow compilers other than armcc to be used (thats how I bricked it). And I already did some experiments with different Hardware components (Uart,VisualAT, GNSS, MQTT, Dataconnections, PSM, GPIO) and they all seem to work. I also mapped API GPIO Numbers to Pinnumbers on the Module and wrote myself a program to upload my dam to the module (updating is now as simple as "make upload"). Currently I'm testing LTE Catm and NbIOT (I only had a regular 2G sim until today, because those are really hard to get here in germany and I had to order a hologram sim).
You will get a E-mail for your Success story page once I tried.

botletics, 2018-10-27:
You mentioned PSM. Did you get that working apart from the DAM app? It didn't seem to work for me. Here's what I did:

* Set AT+CPSMS=1 which only takes effect after restarting the module
* Set AT+CSCLK=1
* Disconnected USB interface
* Restarted module with AT+CPOWD=1 then used PWRKEY to turn it back on

Another note I read is: "If the network supports PSM and accepts that the module uses PSM, the network confirms usage of PSM by allocating an Active Time value to the module. Module will be into PSM according to the command from network."

Thalhammer, 2018-10-27:
To be honest I did not do a lot using AT commands, however, I did some tests on my own and here are the results:
Note:
* I only have a cheap multimeter to measure current, so those numbers might not be 100% accurate
* I did not yet get any of the two LTE modes working (don't know the reason for this, because it finds one using `AT+COPS=?` but fails to connect)
* I did all my test with USB connected because the AT Port only produces garbage (I don't yet know why and to be honest don't care a lot because I don't need that port anyway)

State: Hologram SIM, not connected to Network (configured to use LTE, but it does not work for some reason)
Both commands execute with ok and after a short delay (I guess about 20s) the modules leds turn off and current drops to ~7.8µA
I sent a SMS using Holograms console but nothing happened on the module (which is expected since it is not connected to a network).
Pressing the PWRKEY wakes the module up but after a while it returnes back to PSM mode.
Executing `AT+CPSMS=0` and `AT+CSCLK=0` after pressing PWRKEY makes it stay on again.

State: Hologram SIM, connected to 2G network (Vodafone)
Both commands execute with ok, however even after waiting for what feels like an eternaty nothing happened (No current change, not lost functions, module still receives SMS). Current was around ~30mA.

Regarding the Note about network support: I don't know if any of the 3 available Network providers support it and I will probably not get any information from them since germanys providers are quite restrictive about what information they provide. Since Hologram supports all 3 I will however retry the test with the other two providers (telekom and O2).

However I feel like this PSM is different from the PSM inside DAM since inside my Module I had to specify a time in seconds (greater than 10 Minutes), which the module went to sleep and after this duration the module rebootet as if started using PWRKEY. (This also worked when connected to 2G, but at least my Dataconnection got closed before powerdown, I didn't test if a sms would wake it)
I will do more testing later.

Thalhammer, 2018-10-27:
State: Hologram SIM, connected to 2G network
Init PSM inside DAM.
* Module deactivates Network and switches into PSM mode.
* After pretty much exactly 10 Minutes the module awakes and starts DAM Execution from the beginning (Not sure if memory contents are preserved, but my wild guess would probably be no, but I'll test).
* Sending a SMS right after entering PSM did not wake the module

And to be honest I doubt that SMS receiving will ever work in PSM since 7µA is damn low current (even for todays hardware) I don't think that keeping a receiver circuit alive with such little power is possible at all. This feels more like a rtc and maybe some small in sock memory area.
What will work is switching to PSM mode for a specified time and than awaking and checking for sms. This would mean the response time to a sms would be 10 minutes in the worst case but in many applications this might not matter. It will also be useful for tracker devices for example:
* Store current location to EFS
* Sleep for x Minutes
* Awake, check if location changed and send sms/publish MQTT if it did
* Repeat

Thalhammer, 2018-10-27:
There are however some modes that do allow preserving power while still being able to receive:
Sleep mode and eDRX. The module requires around 2mA in this mode (depends on network) but it seems like it is otherwise completely active.
According to the document the following must happen:
* USB disconnected
* `AT+CSCLK=1` needs to be executed
After this the module should stop responding to UART and switch to a low power state with only RX active. As far as I can tell this should not require any network support. You can wake up the module via paging data (TCP/IP), SMS or by pulling DTR low.
I will try this and respond when I did.

Thalhammer, 2018-10-27:
I just did the same and I'm also down to ~1mA with periodic spikes. Note that I had to delete my dam module first since the running module apparently kept it from sleeping. I did, however, notice some APIs related to sleep, which I do not yet use, so this might just need tweaking. I don't think we can get a lot lower than this without losing connectivity. Maybe eDRX can do a little bit but don't expect wonders.
Looking at the pdf the lowest current without total connection loss seems to be around 0.7mA with a 40s eDRX window. (This would mean a worst case response time of 40s which might or might not be enough).
Sending an SMS to the module in this state resulted in a short burst to around 60-80mA for about 2-3s after which it went down to 1mA almost instantly. Delay between sending and receive was about the same as without sleep mode.

And 1mA is not really that much considering we are connected to a regular 2G network (at least I am) with no added delay. If you really need less power draw one will probably need to go for alternatives like Sigfox or LoraWAN (which have there own problems like worse coverage, higher cost, etc and tbh I don't think current draw will be that much lower).

botletics, 2018-10-27:
I found out how to set the frequency of the pulses. It's just by the eDRX command. I've updated the [current consumption wiki page](https://github.com/botletics/SIM7000-LTE-Shield/wiki/Current-Consumption) to reflect the different power modes and tests I ran.

Thalhammer, 2018-10-27:
I did some experimenting with the dams and it seems like it works just fine from inside dams if you power off the uart. Yes but note that support for those pulse frequencies depends on the network, not all providers might support it, so mileage may vary. Also note that increasing eDRX values should increase sms/data receive delays (note 100% sure, but thats the way I understand the datasheet).

botletics, 2018-10-27:
Yes, that certainly increases delay for SMS. Basically if you set the cycle time to 40.96 seconds and send an SMS to it, you have to wait up to 40.96 seconds before you get the text message.

Thalhammer, 2018-10-27:
Well it does work doesn't it ?

botletics, 2018-10-27:
No, PSM (9uA) doesn't work but sleep mode (~1mA) does.

Thalhammer, 2018-10-27:
Well I will retry tomorrow but I'm quite sure it did in my case. You just need to disconnect from the network or the module might not enter PSM mode.

botletics, 2018-10-28:
You actually need to disconnect from the network? So a procedure like this:
* Set AT+CPSMS=1 to enable PSM mode
* Do something like AT+CFUN=0 to disconnect from network
* Restart module

botletics, 2018-10-28:
And by the way, you mentioned that you can't get connected to a network? Try setting AT+CNMP and AT+CMNB then use AT+COPS=? to list out the networks and connect to one with AT+COPS. More info [here](https://github.com/botletics/SIM7000-LTE-Shield/wiki/Connecting-to-Network)

Thalhammer, 2018-10-28:
Well I didn't try it explicitly but when i was not connected to a network my module changed to a state where it only drew around 7uA and I could wake it by pulling DTR low as described in the datasheet.

Well I can connect to a network, just not to an LTE one. When I do a search with at+cops the module finds 4 networks (3×2G(type 0) and 1 NB Iot (type 9)). When I use automatic mode the module chooses a 2G one and if I try to select the LTE network manually the module takes a while and the responds with ERROR. I already tried to disable Only LTE Mode and also only NBIoT mode the ERROR keeps the same.

Thalhammer, 2018-10-28:
No, I don't have a CAT-M network.
We do have a test network somewhere near the next large City but it's too far to just drive there to test my software.
Where did you read that the Hologram SIM does not support NB-IoT, I could not find anything about this on their website.
Anyway: At the moment I don't really care since LTE Cat-M and NB-IoT is still far worse than traditional 2G and Sleep current is almost the same it does make a difference for me right now. It would have been cool to test my software with 4G but as long as there is no coverage it's not a top priority. Today I finished reverse engineering the first of that two simcom libs (txm_dam.lib) and will upload source code and build scripts by the end of this week once I did enough testing to be more or less sure I did not miss a crucial bug. This will then allow building DAM modules with GCC arm toolchains and remove the burden of having to use ARMCC which is not free and supports neither modern C++ nor anywhere near the features of GCC.

botletics, 2018-10-28:
Hologram told me that. LTE CAT-M and NB-IoT are new technologies and they haven't partnered up with all the carriers for those networks yet. However, in the USA there's nation-wide LTE CAT-M support for AT&T and Verizon on the Hologram SIM.

I'm not sure why you would say that 2G is far better than CAT-M or NB-IoT since those technologies require far less power consumption. GSM requires maybe 330mA of current (SIM800) whereas LTE CAT-M requires only roughly 100mA (SIM7000)

Thalhammer, 2018-10-28:
Because the current while sleeping is almost the same (1.4 vs 1.7ma) and thats whats it mostly does. 
2G isn't better than 4G, but the 4G coverage is far worse here in germany than 2G, so in most regions there simply is only 2G available. Thats why it doesn't matter that much in my case since most of the time I wouldn't have an LTE connection anyway.

botletics, 2018-10-28:
Oh I see.

closed 2019-06-10 by botletics

chris-makaio, 2019-07-24:
I‘m not sure if this is stale, but for reference: eDRX won‘t work on 2G

admiralnelson, 2019-11-15:
I tried to enable sleep mode, but never worked. I use ESP32 and my SIM7000 modem DTR pin is connected to GPIO 2, which is also wired to ESP32 LED light, so I know it glows when high. AT+CSCLK=1 returned OK, but modem current consumption still hovering around 50mA. Isn't it should drop to ~1.5mA when it detects high input in DTR pin? 
When I pull DTR high, the modem does not transmit any serial data (example: simple command such as AT+COPS?), so I assume sleep mode is working, but why is the power consumption does not go down?