botletics/SIM7000-LTE-Shield #44: Firmware Versions

lidraikh, 2018-11-01:
The firmware folder doesn't have a version for the SIM7000A, do you have any?
I tried to reach out to Simcom without luck.

The modem I have is facing issues sending SMS when connected to Verison network directly, but it can establish a connection over Http.

appreciate your help

botletics, 2018-11-01:
Yes I have it, but please email me privately. I haven't gotten explicit permission to share it publicly, but they did say I could share it with my "customers"

closed 2018-11-01 by botletics

lidraikh, 2018-11-05:
thanks for that,
on the same day, I got it from Simcom team.

but unfortunately, the modem still can't send SMS when connected to Verizon network using Verizon SIM card,  which is provisioned for SMS and DATA.