botletics/SIM7000-LTE-Shield #47: at command can't work

lutfisan, 2018-11-16:
at command can't work while using uart but it works if use usb
[image]

botletics, 2018-11-16:
Please be more specific:
* What hardware?
* What connections?
* LiPo battery?
* Etc.

There's no way I can help you without this.

lutfisan, 2018-11-16:
SIM7000E
uart (not work)
usb (work)
supply 5v usb
[image]

botletics, 2018-11-16:
I know what you are referring to, but that doesn't help me a bit unfortunately. I need to know your exact hardware setup otherwise I'll have to close this issue.

lutfisan, 2018-11-16:
[image]
I use this module. when I connect this module via usb at command work perfect, but if I connect via uart using usb to uart at command not work