botletics/SIM7000-LTE-Shield #49: No PPP on UART with B05

Redferne, 2018-11-22:
Hi,

Just tried the B05 firmware but couldn't setup PPP over UART anymore:
```
-> AT+CGREG?
<- OK
-> AT+CGATT?
<- ..+CGATT: 1....OK..
-> AT+CPSI?.
<- +CPSI: LTE CAT-M1,Online,240-XX,0x0086,7681025,0,EUTRAN-BAND1,1925,2,2,0,0,125,0 OK
-> AT+CGPADDR.
<-  +CGPADDR: 1,10.41.136.51 OK
-> AT+CGNAPN.
<- +CGNAPN: 1,"xxx.com.xx" OK
-> ATD*99***1#.
<- ERROR
```

I had to downgrade to B03 again:
```
-> ATD*99***1#.
<- CONNECT 150000000
```
And PPP over UART works fine again. Any ideas?

botletics, 2018-11-22:
I'm assuming you're using SIM7000E? Sorry, I really have no idea because I only use SIM7000A and SIM7000G here in the US.

supersjimmie, 2018-12-11:
I  have heard there is a B06 firmware coming (for my sim7000e).
They gave an early version to me because an issue with the NMEA port, but that caused other issues. So you might wait or even ask Waveshare for the B06 firmware release.

korseby, 2019-02-18:
Have you found a solution yet?

I have the same problem using the Waveshare HAT (https://www.waveshare.com/wiki/SIM7000E_NB-IoT_HAT).

I've updated to the B06 Firmware from their website but getting the same error when using ATD.

supersjimmie, 2019-02-18:
This was the reply then:
If you send AT command to get GPS data via USB interface, USB interface will be hang-up as soon as GPS is enabled, because GPS data is outputed from USB port by default. We doubt that the problem you face is caused by this reason. Did you try to change the NMEA output?

xmichaelx, 2019-03-17:
@korseby @supersjimmie Same issue here. Any good news on this? I've managed to get GPS working and establish TCP connection using sample programs  but with ppp no luck so far.

I'm using these [instructions](https://www.waveshare.com/wiki/SIM7000E_NB-IoT_HAT). Do you have any other recommendation for similar board that works better with ppp?

korseby, 2019-03-18:
As far as I understand it, linux handles the IoT devices using a different protocol. I tried to play around with `usb_modeswitch` and `modprobe usbserial` and tried to bring linux to recognize the Waveshare board as `wwan` interface (using the IoT protocols), but without luck so far. In theory, the `ModemManager` is supposed to do all that, but it seems to be broken in the current Raspbian release.

Below is what I have tried as far as I remember it. As many of the config option have changed because Raspbian switched to `systemd` I do not know anymore how to properly configure network interfaces with terminal commands... That s****s and there is no proper documentation around how the commands are with IoT devices ...

Editing /etc/network/interface.d/wwan0 (this has become obsolete) ...
```
iface wwan0 inet dhcp
   wwan_apn  "APN"
```

Set the APN and bring up the IoT connection manually (you can also do this using minicom -D /dev/ttyS0) ...
```
echo -ne 'AT^NDISDUP=1,1,"APN"\n\r' > /dev/ttyS0
echo -ne 'AT+CIPCSGP=1,"APN","",""' > /dev/ttyS0
echo -ne 'AT+CGDCONT=1,"IP","APN"' > /dev/ttyS0
echo -ne 'AT+CNACT=1' > /dev/ttyS0
echo -ne 'AT+CSTT="APN"\n\r' > /dev/ttyS0
echo -ne 'AT+CIICR\n\r' > /dev/ttyS0
```

Bringing up wwan0 manually ...
```
ifconfig wwan0 up
dhclient wwan0
netstat -rn
ip route add default via x.x.x.x dev wwan0
```

Playing around with ModemManager and/or the CLI ...
```
mmcli -L
nmcli c add type gsm wwan0 /dev/usbS0 con-name APN apn APN
nmcli r wwan0 on
```

Maybe you can figure it out better than I do :)

Some other people had similar problems ...
https://blog.soracom.io/beginners-guide-to-iot-cellular-connectivity-on-raspberry-pi-and-linux-devices-55d4f7489adf
https://docs.ubuntu.com/core/en/stacks/network/network-manager/docs/configure-cellular-connections
https://www.linuxquestions.org/questions/linux-wireless-networking-41/help-using-3g-usb-dongle-through-wwan-interface-rather-than-ppp-4175537653/

Good luck.

xmichaelx, 2019-03-18:
@korseby Thank you, at the very least you've given me hope :)

xmichaelx, 2019-03-18:
I've  tried using *wvdial* with following config (SIM7000 plugged in via USB):
```
[Dialer Defaults]
Init1 = ATZ
Init2 = ATQ0 V1 E1 S0=0 &C1 &D2
Init3 = AT+CGDCONT=1,"IP","plus"
Modem Type = Analog Modem
ISDN = 0
Phone = *99#
Modem = /dev/ttyUSB2
Username = blank
Password = blank
Baud = 9600
```

and the output from wvdial:

```
--> WvDial: Internet dialer version 1.61
--> Cannot set information for serial port.
--> Initializing modem.
--> Sending: ATZ
OK
--> Sending: ATQ0 V1 E1 S0=0 &C1 &D2
ATQ0 V1 E1 S0=0 &C1 &D2
OK
--> Sending: AT+CGDCONT=1,"IP","plus"
AT+CGDCONT=1,"IP","plus"
OK
--> Modem initialized.
--> Sending: ATDT*99#
--> Waiting for carrier.
ATDT*99#
CONNECT 150000000
--> Carrier detected.  Waiting for prompt.

--> Connected, but carrier signal lost!  Retrying...
--> Sending: ATDT*99#
--> Waiting for carrier.
--> Timed out while dialing.  Trying again.
--> Sending: ATDT*99#
--> Waiting for carrier.

```

and LED on my SIM7000E-NB-IoT-HAT starts starts repeatedly flashing as if the device stoppped responding and indeed that's what it looks like. 

Reissuing wvdial gives:

```
--> WvDial: Internet dialer version 1.61
--> Cannot set information for serial port.
--> Initializing modem.
--> Sending: ATZ
--> Sending: ATQ0
--> Re-Sending: ATZ
--> Modem not responding.
```

so it looks like the serial port stopped responding altogether.

I'm kinda disappointed - I've sort of managed to get GPS signal when hat is connected to GPIO pins and also managed to get the samples working, but getting working ppp/wwan0 seems to be impossible.

Anyways @korseby thanks for help. Do you have any recommendations for alternative that you know works well on raspberry?

qsun, 2019-07-22:
It looks like I've same issue with B05 firmware on SIM7000E, but I could not confirm it's working with B03.

korseby, 2019-07-22:
Latest waveshare firmware includes some additions, but could not read about fixes... It could be worth a try though.

https://www.waveshare.com/w/upload/6/60/NDA_SIM7000E_Firmware_Release_Note.pdf

qsun, 2019-07-23:
@korseby 

Thank you very much for your reply, and I can confirm B08 firmware is working fine.

```
AT+GMR
Revision:1351B08SIM7000E

OK
AT+CGREG?
+CGREG: 0,2

OK
AT+CGREG?
+CGREG: 0,1

OK
AT+CPSI?
+CPSI: LTE CAT-M1,Online

OK
AT+CGPADDR
+CGPADDR: 1,10.118.51.20
+CGPADDR: 13,0.0.0.0

OK
AT+CGNAPN
+CGNAPN: 1,"telstra.m2m"

OK
ATD*99***1#
CONNECT 150000000
```

qsun, 2019-07-23:
@korseby 

Thank you very much for your reply, and I can confirm B08 firmware is working fine.

```
AT+GMR
Revision:1351B08SIM7000E

OK
AT+CGREG?
+CGREG: 0,2

OK
AT+CGREG?
+CGREG: 0,1

OK
AT+CPSI?
+CPSI: LTE CAT-M1,Online

OK
AT+CGPADDR
+CGPADDR: ~~~
+CGPADDR: ~~~
OK
AT+CGNAPN
+CGNAPN: 1,"telstra.m2m"

OK
ATD*99***1#
CONNECT 150000000
```

botletics, 2019-07-23:
I guess I can close this now?

closed 2019-07-23 by botletics