botletics/SIM7000-LTE-Shield #50: getNetworkStatus uses +CREG, should be +CGREG

AstarothPoD, 2018-11-26:
When using the SIM7000E with an NB-IoT-only SIM, CREG never returns the correct status, only 0,0 or 0,2.
CGREG however works as intended. Could this be changed?

botletics, 2018-11-26:
Hmmmm I'm not sure why that would be the case because CGREG is actually for GPRS (2G) so CREG should be more appropriate for NB-IoT. I'll ask SIMCom why that might be the case. It's an easy fix but I'd rather know what's going on first.

AstarothPoD, 2018-11-26:
CEREG also works.

botletics, 2018-11-26:
May I ask what country and SIM card you're using?

AstarothPoD, 2018-11-26:
AT
OK
AT+CREG?
+CREG: 0,0
OK
AT+CGREG?
+CGREG: 0,1
OK
AT+CEREG?
+CEREG: 0,1
OK
AT+CPSI?
+CPSI: LTE NB-IOT,Online,242-01,0xA411,18460008,457,EUTRAN-BAND20,6352,0,0,-7,-77,-70,14
AT+COPS?
+COPS: 0,0,"242 01 Telenor",9

botletics, 2018-12-10:
I had another user report that CREG would say "0,3" (denied) but CGREG would say "0,5" (registered, roaming) so I went ahead and changed the library file to use CGREG for SIM7000 and SIM7500.

closed 2018-12-10 by botletics

OctaviusK, 2019-03-11:
> 
> 
> I had another user report that CREG would say "0,3" (denied) but CGREG would say "0,5" (registered, roaming) so I went ahead and changed the library file to use CGREG for SIM7000 and SIM7500.

Dear botletics,  We also has similar issue. May we know did SIMcom reply you on the above issues? Thks.

botletics, 2019-03-11:
No, I don't remember getting a response for that but u may not have asked. Either way, it's fixed in the library already.

OctaviusK, 2019-03-12:
Thank you for your reply. Don't know why I still get similar issue. Does SIM7000E already support IPv6?
I am using version 1351B07SIM7000E.  

Please see below SIM7000E response:

Revision:1351B07SIM7000E
CSUB:V02
APRev:1351B07SIM7000,V02
at+cgatt?
+CGATT: 1
OK
at+cgreg?
+CGREG: 0,1
OK
at+cgdcont?
+CGDCONT: 1,"IPV4V6","b2b-nbiot","0.0.0.0.0.0.0.0.0.0.0.0.0.0.0.0",0,0,0,0
+CGDCONT: 13,"IP","CMNET","0.0.0.0",0,0,0,0
at+cops?
+COPS: 0,0,"SGP-M1",9
OK
at+cgpaddr
+CGPADDR: 1,36.1.116.0.72.1.0.0.0.1.0.0.37.178.61.42
+CGPADDR: 13,0.0.0.0
OK
at+creg?
+CREG: 0,0
OK