botletics/SIM7000-LTE-Shield #55: Sim7000E manufacturer-assigned name

supersjimmie, 2018-11-28:
In the readme, you say:
> Included FONA_LTE_A and FONA_LTE_E types in .h file but not in .cpp declaration (I don't have a SIM7000E module with me so I don't know what its manufacturer-assigned name is)

I have a SIM7000E, so how can I help you?

botletics, 2018-11-28:
You must be looking at a different product's documentation. I never say that anywhere.

closed 2018-11-28 by botletics

supersjimmie, 2018-11-28:
On this page https://github.com/botletics/SIM7000-LTE-Shield
Almost at the bottom (before the last bullet), that is where you say it.

Copy paste from that page:
"Included FONA_LTE_A and FONA_LTE_E types in .h file but not in .cpp declaration (I don't have a SIM7000E module with me so I don't know what its manufacturer-assigned name is)"

botletics, 2018-11-28:
Oh I see what you're saying now, thanks. I thought for some reason you were asking for help about something regarding the SIM7000E. I have since used the SIM7000E so that's not relevant anymore.

supersjimmie, 2018-11-29:
No problem, here to help each-other, right?  :)
But then back, I have a sim7000e, is there anything I can do to help you add the missing part? (what manufacturer-assigned name do you mean exactly?)

botletics, 2018-11-29:
Haha thanks, but I think I'm good for now. I have SIM7000E boards here. What I was missing at first was the model number from the comma d ATI, like SIM7000E 1351B04 or something.