botletics/SIM7000-LTE-Shield #56: UDP functions beside the TCP functions

supersjimmie, 2018-12-07:
Currently there are only TCP functions like ```UDPconnect``` and ```TCPsend```, not for UDP.
My carrier, T-Mobile in The Netherlands, uses UDP.

There's actually only 1 slight difference between UDP and TCP functions: ```AT+CIPSTART=\"UDP\"```.
What would be the nicest way to make this library also support UDP? Clone all TCP* functions to UDP* functions with only a few lines changed, or some more robust way to avoid all identical functions?

botletics, 2018-12-07:
You could just add a flag as a function input that defines whether it's TCP or UDP and then the function will run either command based on that. If I had time I'd look into it more but I'm crazy busy at the moment.

supersjimmie, 2018-12-07:
Ok, I'll try with something like a ```boolean useUDPinstead``` flag.