botletics/SIM7000-LTE-Shield #57: Commands echo back instead of OK

supersjimmie, 2018-12-12:
At several different point, sometimes the program goes into a loop, because the sim7000 starts to repeat the commands instead of replying with OK.

This happens at more places, during GPS, during Connect to cell, and more.
Looks like:
```
Failed to get GPS location, retrying...
	---> AT+CGNSINF
	<--- AT+CGNSINF
Failed to get GPS location, retrying...
	---> AT+CGNSINF
	<--- AT+CGNSINF
```
I tried to flush the fonaSS, send an extra AT, wait a while, I cannot find why and I cannot find how to get out of this strange state...

botletics, 2018-12-12:
Did you set **ATE1** which enables echo? For example, if you were trying to send AT commands via USB and set ATE1 so that you can see the stuff you're typing.

botletics, 2018-12-12:
To disable, set **ATE0** which should also be set during module initialization in the LTE_Demo code.

supersjimmie, 2018-12-12:
Yes it does ATE0 at the beginning.
Here's some more:

```
Attempting to open comm with ATs
	---> AT
	<--- OK
	---> ATE0
	<--- OK
	---> ATE0
	<--- OK
	---> ATI
	<--- SIM7000E R1351

OK

	---> AT+CPMS="SM","SM","SM"
	<--- +CPMS: 0,0,0,0,0,0
FONA is OK
Found SIM7000E (European)
	---> AT+GSN
	<--- 8652340xxxx
Module IMEI: 8652340xxxx
	---> AT+CFUN=1
	<--- OK
	---> AT+CGDCONT=1,"IP","cdp.iot.t-mobile.nl"
	<--- OK
	---> AT+CGREG?
	<--- +CGREG: 0,5
Network status 5: Registered roaming
Connected to cell network!
	---> AT+CGNSPWR=1
	<--- OK
Turned on GPS!
	---> AT+CGNSINF
	<--- 
Failed to get GPS location, retrying...
	---> AT+CGNSINF
	<--- +CFUN: 1
Failed to get GPS location, retrying...
	---> AT+CGNSINF
	<--- AT+CGNSINF
Failed to get GPS location, retrying...
	---> AT+CGNSINF
	<--- AT+CGNSINF
```

As you can see, there is also a strange thing happening here:
```
---> AT+CGNSINF
	<--- +CFUN: 1
```
Like it comes back with something from an earlier command.
Then it starts to malfunction...

botletics, 2018-12-12:
Looks like your shield is powering off. Messages like "+CFUN: 1" are usually displayed shortly after it boots up. Please make sure you are using a 3.7V LiPo battery with the shield.

supersjimmie, 2018-12-12:
I have a newer version of the shield, it lacks things like the PWR push button and also the possibility to add the LiPo. When I asked them about it, Waveshare told me that I don't need that.... (looking at the PCB, there is no regulator for the LiPo too)

It works fine about 9 out of 10 times.

botletics, 2018-12-12:
Newer version? The newest version of the Botletics SIM7000 shield has a LiPo battery connector. If you're using something else that sorry, I don't know.

botletics, 2018-12-12:
Sounds like Waveshare didn't fully test their product then. From what you're experiencing, the module is definitely shutting down. If there's a power indicator LED you should be able to see it shut down. Also, try enabling GPS. I think the module will most certainly shut off because GPS draws another 30mA of current.

closed 2018-12-12 by botletics

supersjimmie, 2018-12-12:
LOL, they've also changed the Power-LED. It's always on, regardless of the actual state of the module...
(they also changed the PWR-pin behavior, but I am able to do the power down, while the LED stays on)
I guess the only way to save some extra current during power down, is to kick that LED off the PCB...
They only way so actually see the power down is via the current draw and the activity LED that really stops blinking. (and the module stops accepting AT commands when I test it)

There is a large 1000uF capacitor on the board, which they say should fix it.
I will try to add an additional tantalum capacitor, which is also very helpful with power-peaking modules likes esp8266's. (tantal's are better for this kind of stabilization)

But just to clarify things a little bit, this is actually not exactly the arduino shield that you use, it's really a module made for the raspberry pi. It's had the same features (like the powerbutton) in the recent past, but because some people had trouble powering it on, they decided to change it.  I had very recent contact with them by e-mail about several of those changes and some issues with the B05 firmware.
They just keep saying that "The SIM7000E is updated", every time I complain.

botletics, 2018-12-12:
Yea I understand you're using the Waveshare Raspberry Pi HAT instead of the shield I created (the Botletics one). Sorry, not much I can do to help there lol. I'm a firm believer that product support should come from the people selling it  :)

supersjimmie, 2018-12-13:
Sure, now we know that it's caused by the lack of lipo support, it's up to the sellers.
But you helped me a lot by finding out that the device was restarting, before that I had no clue...

Fredovsky, 2019-02-13:
@supersjimmie did you manage to solve/identify the issue ? I got the exact same problem, AT response is either : empty; echoes the command; or rarely I have the odd +CFUN: 1. I'm powering up through a separate supply able to handle high load so I don't think power is an issue here.

I know this is not related to Botletics as I'm using the Waveshare piHat board, but curious if others have resolved this problem.

Thanks

supersjimmie, 2019-02-13:
> @supersjimmie did you manage to solve/identify the issue ? I got the exact same problem, AT response is either : empty; echoes the command; or rarely I have the odd +CFUN: 1. I'm powering up through [...]

@Fredovsky 
After a lot of mails, I got the message that "the engineer went on holiday", then nothing... That was about a month ago. I think I will send them a mail to ask if the engineer ever came back.

supersjimmie, 2019-02-14:
@Fredovsky  they now replied that it is solved in firmware: https://www.waveshare.com/wiki/File:1351B06SIM7000E.rar
But I told them last december that I already use that. Perhaps you can check to be sure?

Fredovsky, 2019-02-16:
Thanks I'll try again next week if I can flash the new firmware but last I tried the QPST tool was returning a "DUT 1" (not sure) error. I'll let you know ! Thanks.

Fredovsky, 2019-02-18:
Ok so I updated to the B06 firmware. Problem remains the same. I noticed I don't have any problems when I type the AT commands manually through serial on the Sim7000 USB port (using Putty for exemple).

When using the Arduino and the Fona library, it starts echoing or giving blank replies after 5 or 6 commands...

botletics, 2019-02-24:
Mhmm sorry I'm not sure what is happening. It should find it and communicate right away.