botletics/SIM7000-LTE-Shield #58: Do you have Amazon AWS Tutorial..?

kittinook, 2018-12-13:
Hi, I've interested this board. Do you have the example for communication between the board and AWS IoT cloud?

kittinook, 2018-12-18:
Thanks for your fast response, you can share example AT command logs for Microsoft Azure. I can't find the example AT command in this web http://www.simcomm2m.com/En/module/detail.aspx?id=175.

botletics, 2018-12-18:
I can't share it publicly so you would have to order a Botletics shield before I could share it.

bradleytompkins, 2019-01-09:
Can you share these with me? We have purchased several botletics shields to experiment with, and are trying to get them talking to Azure IoT Hub.

slipiduche, 2019-03-29:
> Not currently, no. However, there are example AT command logs I have from SIMCom for Microsoft Azure which theoretically should work for other platforms like AWS IoT and anything that requires certificates. Basically you store the certificate in the SIM7000's EFS (electronic file system) and use the SSL commands to connect.

Hi, there are some example where shows how store and read data from SIM7000's EFS?

botletics, 2019-04-01:
[This should help](https://github.com/botletics/SIM7000-LTE-Shield/tree/master/SIM7000%20Documentation/AT%20Command%20Logs) but also check the related AT command manual.

slipiduche, 2019-04-01:
> [This should help](https://github.com/botletics/SIM7000-LTE-Shield/tree/master/SIM7000%20Documentation/AT%20Command%20Logs) but also check the related AT command manual.

oh thanks you. i'm realy confused, i don't know from where extract the .cer, i mean a SD? a web server? or the download from pc?. where i could put the .cer to apply these commands.

thanks so much.

botletics, 2019-04-07:
That depends on what platform you're using (Azure, AWS, etc.) and that file would be on your computer and sent to the SIM7000 via USB with AT commands.

slipiduche, 2019-04-16:
thank you was very usefull, this is my trouble now:

FONA> C
---> AT+CCID
<--- 89577321111786486512
SIM CCID = 89577321111786486512
FONA> n
---> AT+CGREG?
<--- +CGREG: 0,2
Network status 2: Not registered (searching)
FONA> n
---> AT+CGREG?
<--- +CGREG: 0,2
Network status 2: Not registered (searching)
FONA> 1
---> AT+CPSI?
<--- +CPSI: NO SERVICE,Online
---> AT+COPS?
<--- +COPS: 0

OK
FONA> i
---> AT+CSQ
<--- +CSQ: 99,99
RSSI = 99: -115 dBm
FONA>

im using the sim7000g in colombia but "Network status 2: Not registered
(searching)" im trying with CLARO and TIGO operators

slipiduche, 2019-04-24:
i have a doubt, the certificate must be in what format? i try this
 -----BEGIN CERTIFICATE-----<LF>
MIIDQTCCAimgAwIBAgITBmyfz5m/jAo54vB4ikPmljZbyjANBgkqhkiG9w0BAQsF<LF>
ADA5MQswCQYDVQQGEwJVUzEPMA0GA1UEChMGQW1hem9uMRkwFwYDVQQDExBBbWF6<LF>
b24gUm9vdCBDQSAxMB4XDTE1MDUyNjAwMDAwMFoXDTM4MDExNzAwMDAwMFowOTEL<LF>
MAkGA1UEBhMCVVMxDzANBgNVBAoTBkFtYXpvbjEZMBcGA1UEAxMQQW1hem9uIFJv<LF>
b3QgQ0EgMTCCASIwDQYJKoZIhvcNAQEBBQADggEPADCCAQoCggEBALJ4gHHKeNXj<LF>
ca9HgFB0fW7Y14h29Jlo91ghYPl0hAEvrAIthtOgQ3pOsqTQNroBvo3bSMgHFzZM<LF>
9O6II8c+6zf1tRn4SWiw3te5djgdYZ6k/oI2peVKVuRF4fn9tBb6dNqcmzU5L/qw<LF>
IFAGbHrQgLKm+a/sRxmPUDgH3KKHOVj4utWp+UhnMJbulHheb4mjUcAwhmahRWa6<LF>
VOujw5H5SNz/0egwLX0tdHA114gk957EWW67c4cX8jJGKLhD+rcdqsq08p8kDi1L<LF>
93FcXmn/6pUCyziKrlA4b9v7LWIbxcceVOF34GfID5yHI9Y/QCB/IIDEgEw+OyQm<LF>
jgSubJrIqg0CAwEAAaNCMEAwDwYDVR0TAQH/BAUwAwEB/zAOBgNVHQ8BAf8EBAMC<LF>
AYYwHQYDVR0OBBYEFIQYzIU07LwMlJQuCFmcx7IQTgoIMA0GCSqGSIb3DQEBCwUA<LF>
A4IBAQCY8jdaQZChGsV2USggNiMOruYou6r4lK5IpDB/G/wkjUu0yKGX9rbxenDI<LF>
U5PMCCjjmCXPI6T53iHTfIUJrU6adTrCC2qJeHZERxhlbI1Bjjt/msv0tadQ1wUs<LF>
N+gDS63pYaACbvXy8MWy7Vu33PqUXHeeE6V/Uq2V8viTO96LXFvKWlJbYK8U90vv<LF>
o/ufQJVtMVT8QtPHRh8jrdkPSHCa2XV4cdFyQzR1bldZwgJcJmApzyMZFo6IQ6XU<LF>
5MsI+yMRQ+hDKXJioaldXgjUkK642M4UwtBV8ob2xJNDd2ZhwLnoQdeXeGADbkpy<LF>
rqXRfboQnoZsG4q5WTP468SQvvG5<LF>
-----END CERTIFICATE-----<LF><CR>

and not connect

at log is that
<SUB><---<CR><LF>
OK<CR><LF>
--->AT+CFSgfis=3,"root_ca.pem"<CR><LF>
<SUB><---<CR><LF>
+CFSGFIS: 1189<CR><LF>
<CR><LF>
OK<CR><LF>

+CNACT: 1,"100.100.197.199"<CR><LF>
<CR><LF>
OK<CR><LF>
--->AT+SMCONF="URL",a5xpqsmvbu9sq-ats.iot.us-west-2.amazonaws.com,8883<CR>
<CR><LF>
<SUB><---<CR><LF>
OK<CR><LF>
--->AT+SMCONF="CLIENTID",device2<CR>
<CR><LF>
<SUB><---<CR><LF>
OK<CR><LF>
--->AT+SMCONF="KEEPTIME",60<CR>
<CR><LF>
<SUB><---<CR><LF>
OK<CR><LF>
--->AT+SMCONF="CLEANSS",0<CR>
<CR><LF>
<SUB><---<CR><LF>
OK<CR><LF>
--->AT+SMCONF="QOS",0<CR>
<CR><LF>
<SUB><---<CR><LF>
OK<CR><LF>
--->AT+CSSLCFG?<CR>
<CR><LF>
<SUB><---<CR><LF>
OK<CR><LF>
--->AT+CSSLCFG="sslversion",0,3<CR>
<CR><LF>
<SUB><---<CR><LF>
OK<CR><LF>
--->AT+CSSLCFG=0,1,0<CR>
<CR><LF>
<SUB><---<CR><LF>
ERROR<CR><LF>
<CR><LF>
--->AT+CSSLCFG=convert,2,root_ca.pem<CR>
<CR><LF>
<SUB><---<CR><LF>
OK<CR><LF>
--->AT+CSSLCFG=convert,1,my_client.pem,my_key.pem<CR>
<CR><LF>
<SUB><---<CR><LF>
OK<CR><LF>
--->AT+CSSLCFG?<CR>
<CR><LF>
<SUB><---<CR><LF>
OK<CR><LF>
--->AT+CIPSTATUS<CR><LF>
<SUB><---<CR><LF>
OK<CR><LF>
<CR><LF>
STATE: IP GPRSACT<CR><LF>
--->AT+CIFSR<CR><LF>
<SUB><---<CR><LF>
100.100.197.199<CR><LF>
<CR><LF>
--->AT+CIPSTATUS<CR><LF>
<SUB><---<CR><LF>
OK<CR><LF>
<CR><LF>
STATE: IP STATUS<CR><LF>
--->AT+SMSSL=1,root_ca.pem,my_client.pem<CR>
<CR><LF>
<SUB><---<CR><LF>
OK<CR><LF>
--->AT+SMSSL?<CR>
<CR><LF>
<SUB><---<CR><LF>
+SMSSL: 1,"root_ca.pem","my_client.pem"<CR><LF>
<CR><LF>
OK<CR><LF>
--->AT+CSSLCFG?<CR>
<CR><LF>
<SUB><---<CR><LF>
OK<CR><LF>
--->AT+CGATT?<CR>
<CR><LF>
<SUB><---<CR><LF>
+CGATT: 1<CR><LF>
<CR><LF>
OK<CR><LF>
--->AT+SMCONN<CR>
<CR><LF>
<SUB><---<CR><LF>
ERROR<CR><LF>

When i try not secure connection with cloudmqtt these commands works but not with AWS

brunokruse, 2019-05-10:
> Not currently, no. However, there are example AT command logs I have from SIMCom for Microsoft Azure which theoretically should work for other platforms like AWS IoT and anything that requires [...]

I saw the azure example in your AT Command Logs; thank you for that. However, it seems AWS only supports Https. The firmware on some of the shields support SSL only via TCP.  Is there info on specific firmware releases and features to confirm? I am using B017000G.

botletics, 2019-05-13:
Sorry, I'm not sure if there's anything on specific firmware versions.

jefflikesbagels, 2019-06-22:
I apologize in advance for my ignorance, as this is my first time programming a SIM7000. I am trying to perform the same task as above but using hologram.io. I created a new function in the Adafruit_FONA.cpp library and called it postDataHTTPS:

```
boolean Adafruit_FONA::postDataHTTPS(const char *request_type, const char *URL, const char *body, const char *token, uint32_t bodylen) {
  // NOTE: Need to open socket/enable GPRS before using this function
  // char auxStr[64];
  
    sendCheckReply(F("AT+GMR"), ok_reply, 10000);
	sendCheckReply(F("AT+CNACT=1,\"hologram\""), ok_reply, 10000);
	sendCheckReply(F("AT+CNACT?"), ok_reply, 10000);
	sendCheckReply(F("AT+CSSLCFG=\"convert\",2,\"hologram.cer\""), ok_reply, 10000);
	sendCheckReply(F("AT+SHSSL=1,\"hologram.cer\""), ok_reply, 10000);

	char urlBuff[strlen(URL) + 22];
	sprintf(urlBuff, "AT+SHCONF=\"URL\",\"%s\"", URL);
	if (! sendCheckReply(urlBuff, ok_reply, 10000))
		return false;
	
	sendCheckReply(F("AT+SHCONF=\"BODYLEN\",100"), ok_reply, 10000);
	sendCheckReply(F("AT+SHCONF=\"HEADERLEN\",100"), ok_reply, 10000);
	sendCheckReply(F("AT+SHCONN"), ok_reply, 10000);
	
	char dataBuff[strlen(body) + 22];
	sprintf(dataBuff, "AT+SHBOD=\"%s\",100", body);
	
	//if (! sendCheckReply(dataBuff, ok_reply, 10000))
	//	return false;
	sendCheckReply(dataBuff, ok_reply, 10000);
	//sendCheckReply(F("AT+SHBOD=\"TEST\",100"), ok_reply, 10000);
  
	sendCheckReply(F("AT+SHAHEAD=\"Content-Length\",\"120\""), ok_reply, 10000);
	sendCheckReply(F("AT+SHSTATE?"), ok_reply, 10000);
	sendCheckReply(F("AT+SHREQ=3"), ok_reply, 10000);
	sendCheckReply(F("AT+SHREAD=0,227"), ok_reply, 10000);
	sendCheckReply(F("AT+SHDISC"), ok_reply, 10000);
  
  return true;
}
```
I've been reading the SIM7000 documentation for the HTTPS commands, and I am struggling to get it working as I am sure I have mistakes somewhere. In my Arduino sketch I have the following:
```
        // Post data to website via 2G or LTE CAT-M/NB-IoT
        // Create char buffers for the floating point numbers for sprintf
        // Make sure these buffers are long enough for your request URL
        char URL[150];
        char body[100];
        char deviceID[] = "######";
        char tagID[] = "[\"_RESTAPI_\", \"WATER_LOW\"]";
        char message[] = "\"Water_Low\"";

        // POST request
        sprintf(URL, "https://dashboard.hologram.io/api/1/csr/rdm");
        sprintf(body, "{\"deviceid\": %s, \"tags\": %s, \"data\": %s}", deviceID, tagID, message);

        Serial.println(F("Attempting to perform HTTPS POST..."));
        Serial.print("URL: ");
        Serial.print(URL);
        Serial.println();
        Serial.print("Body: ");
        Serial.print(body);
        Serial.println();
        if (!fona.postDataHTTPS("POST", URL, body)){
          Serial.println(F("Failed to complete HTTPS POST!"));
        } else {
          Serial.println(F("Successfully performed HTTPS POST!"));
        }
```
When the sketch runs, I get the following on the serial monitor:
```
Attempting to perform HTTPS POST...
[... 40 lines cut ...]
U3RhcmZpZWxkIENsYXNzIDIgQ2VydGlmaWNhdGlvbiBBdXRob3JpdHkwHhcNMDQw
NjI5MTczOTE2WhcNMzQwNjI5MTczOTE2WjBoMQswCQYDVQQGEwJVUzElMCMGA1UE
ChMcU3RhcmZpZWxkIFRlY2hub2xvZ2llcywgSW5jLjEyMDAGA1UECxMpU3RhcmZp
ZWxkIENsYXNzIDIgQ2VydGlmaWNhdGlvbiBBdXRob3JpdHkwggEgMA0GCSqGSIb3
DQEBAQUAA4IBDQAwggEIAoIBAQC3Msj+6XGmBIWtDBFk385N78gDGIc/oav7PKaf
8MOh2tTYbitTkPskpD6E8J7oX+zlJ0T1KKY/e97gKvDIr1MvnsoFAZMej2YcOadN
+lq2cwQlZut3f+dZxkqZJRRU6ybH838Z1TBwj6+wRir/resp7defqgSHo9T5iaU0
X9tDkYI22WY8sbi5gv2cOj4QyDvvBmVmepsZGD3/cVE8MC5fvj13c7JdBmzDI1aa
K4UmkhynArPkPw2vCHmCuDY96pzTNbO8acr1zJ3o/WSNF4Azbl5KXZnJHoe0nRrA
1W4TNSNe35tfPe/W93bC6j67eA0cQmdrBNj41tpvi/JEoAGrAgEDo4HFMIHCMB0G
A1UdDgQWBBS/X7fRzt0fhvRbVazc1xDCDqmI5zCBkgYDVR0jBIGKMIGHgBS/X7fR
zt0fhvRbVazc1xDCDqmI56FspGowaDELMAkGA1UEBhMCVVMxJTAjBgNVBAoTHFN0
YXJmaWVsZCBUZWNobm9sb2dpZXMsIEluYy4xMjAwBgNVBAsTKVN0YXJmaWVsZCBD
bGFzcyAyIENlcnRpZmljYXRpb24gQXV0aG9yaXR5ggEAMAwGA1UdEwQFMAMBAf8w
DQYJKoZIhvcNAQEFBQADggEBAAWdP4id0ckaVaGsafPzWdqbAYcaT1epoXkJKtv3
L7IezMdeatiDh6GX70k1PncGQVhiv45YuApnP+yz3SFmH8lU+nLMPUxA2IGvd56D
eruix/U0F47ZEUD0/CwqTRV/p2JdLiXTAAsgGh1o+Re49L2L7ShZ3U0WixeDyLJl
xy16paq8U4Zt3VekyvggQQto8PT7dL5WXXp59fkdheMtlb71cZBDzI0fmgAKhynp
VSJYACPq4xJDKVtHCN2MQWplBqjlIapBtJUhlbl90TSrE9atvNziPTnNvT51cKEY
WQPJIrSPnNVeKtelttQKbfi3QBFGmh95DmK/D5fs4C8fF5Q=
-----END CERTIFICATE-----
```
However I am not sure how to include that on the AT+CSSHCFG command.

Thanks!

TimRoadley, 2019-06-23:
I'm having the same issues as @jefflikesbagels 

In addition, I'm concerned that if I release these IoT devices to our customers then we will have to recall them all to update the root CA when those certificates expire.

I wonder if there's a way to force the device to trust the root CA regardless of who it is and then just continue with the HTTPS POST. 

I think this would be ok, since the configuration on what we're sending to will be baked in to the firmware of our devices.

Thoughts?

TimRoadley, 2019-06-23:
In case it helps anyone, I managed to upload a cert with the following procedure (not using the EFS Explorer)

**Copy a CA Root Cert (for the site you're trying to connect to) to the SIM7000:**

1) Go to **ssltools.com**
2) Enter the website you want the root cert for **(e.g. https://putsreq.com)**
3) Download the root certificate PEM, which should give you a **.cer** file
4) Strip out all carriage returns and then use word or something similar to count the characters in the cert file

**Use these commands to load the cert onto the SIM7000:**

AT+CFSINIT
AT+CFSWFILE=3,"AddTrust.crt",0,1496,10000      **_<-- 1496 is the char length_**
_paste the cert data into the terminal within 10 seconds_
AT+CFSTERM

This puts the cert in the 'customer' directory on the SIM7000. Tomorrow I'll see if this lets me make that https post, try your luck with this approach in the mean time.

jefflikesbagels, 2019-06-23:
@TimRoadley Thanks for the info. I've been thinking about how I can integrate all of this into the Arduino so I can remotely push a new certificate to the EEPROM using some creativity with hologram.io's tools, then use the AT+CFSWFILE command to write the certificate, but unfortunately I'm using an UNO which does not have enough EEPROM space to store the entire certificate. For my use case, I'm just doing a simple DIY project for a friend, so if it lasts until 2034 when the CA expires that's good enough for me haha.

What's strange is the ssltools site is giving me a 502 bad gateway when I tried to download the root certificate, but all of the others work. I originally just used Chrome to export it anyways. I think I am past the certificate part, so now I have to figure out why the AT+SHCONN command is failing.

jefflikesbagels, 2019-06-25:
Well this is frustrating. I did some more digging, and found another issue that is preventing me from making progress. According to the SIMCOM technical documentation, the max string size for the URL on the AT+SHCONF and AT+SHREQ commands is 64 bytes. For sending a data route through Hologram, the Arduino needs to do an HTTPS POST to the following URL:
`https://dashboard.hologram.io/api/1/csr/rdm?apikey=##############################`
I was not adding the API key before (duh - hologram was rejecting the API call), but now that I am adding it, the length of the URL is 81 bytes. I tried setting up an HTTP redirect on my personal web server to shorten the length, but the redirect prevents the hologram REST API from parsing the data properly.

One thing that helped me immensely was using the [Restlet Client Chrome Extension](https://chrome.google.com/webstore/detail/restlet-client-rest-api-t/aejoelaoggembcahagimdiliamlcdmfm). Between that and sifting through the Hologram REST API documentation again helped me figure out what format it's actually expecting.

Getting back to the issue at hand, it could be possible that the AT+SHCONN command fails because the Hologram API is rejecting the connection due to the incorrect URL (without the API key). Surely I am missing something here? The 64 byte URL limit is going to completely break the SIM7000's ability to do HTTPS POST commands to activate Hologram data routes. The next option may be using a TCP socket connection to Hologram Cloud: [Socket API](https://hologram.io/docs/reference/cloud/embedded/), [Device Key](https://help.hologram.io/articles/2741968-what-is-a-device-key). It looks like that would be the better solution anyways.

Sorry to derail a bit from the original intention of working with AWS, but I believe the procedure will be very similar to Hologram, so this development will still be beneficial. It looks like AWS supports both HTTPS and MQTT calls, while Hologram supports HTTPS and TCP socket calls.

TimRoadley, 2019-06-25:
@jefflikesbagels out of interest what firmware version are you running (and what chip)?

My testing has paused since I blew up my SIM7000E with a firmware update. Be careful with firmware over the air (FOTA)!

jefflikesbagels, 2019-06-25:
@TimRoadley I have a SIM7000A running 1351B03SIM7000A firmware.

TimRoadley, 2019-06-25:
@jefflikesbagels  I wonder if https://github.com/botletics/SIM7000-LTE-Shield/blob/master/SIM7000%20Documentation/Firmware/1351B04SIM7000A.rar would help (I have no idea)

jefflikesbagels, 2019-06-26:
@TimRoadley Thanks, I went ahead and updated to B04 just for good measure.

I finally got the Arduino sending data to Hologram via the Socket API!!! The issue I found is very silly too. For the FONA library commands, a lot of them are used in the following (or a similar) fashion:
```
        // Connect to TCP server
        if (!fona.TCPconnect(host, port)) {
          Serial.println(F("Failed to connect to server!"));
          delay(5000);
          break;
        } else {
          Serial.println(F("Successfully connected to server!"));
        }
        delay(5000);
        // Send TCP payload
        if (!fona.TCPsend(TCPpayload,sizeof(TCPpayload))) {
          Serial.println(F("Failed to send TCP payload!"));
          delay(5000);
          break;
        }
```
Where there is an if statement checking whether the function returned false or true. With this code it was not working properly at all. However, on a whim I decided to try and simplify the code as much as possible, and removed all of these checks down to the following:
```
        fona.TCPconnect(host, port);
        fona.TCPsend(TCPpayload,sizeof(TCPpayload));
        fona.TCPclose();
```
And all of a sudden it started working! One thing I noticed before was that I would get the "failed to connect" message on the serial monitor, but would continue receiving responses from the SIM7000, almost like the code is getting ahead of itself. I know I've deviated really far from the original goal of using HTTPS POST, but give this a shot and see if it helps. It's possible that removing all of the extra if statements and logic will allow the SIM7000 to send data properly. Here's the final snippet of code for my TCP socket connection:
```
        // Send TCP payload to server via LTE CAT-M/NB-IoT
        char host[] = "cloudsocket.hologram.io";
        uint32_t port = 9999;
        char devicekey[] = "xxxxxxxx";
        char data[] = "Water_Low";
        char topics[] = "WATER_LOW";
        char TCPpayload[strlen(devicekey)+strlen(data)+strlen(topics)+24];
        sprintf(TCPpayload, "{\"k\":\"%s\",\"d\":\"%s\",\"t\":\"%s\"}", devicekey, data, topics);
        Serial.println(TCPpayload);

        // Connect to GPRS
        fona.enableGPRS(true);

        // Connect to TCP server
        fona.TCPconnect(host, port);

        // Send TCP payload
        fona.TCPsend(TCPpayload,sizeof(TCPpayload));

        // Disconnect from TCP server
        fona.TCPclose();

        // Disconnect from GPRS
        fona.enableGPRS(false);
```

botletics, 2019-08-01:
Hey guys, there is now a [Botletics community forum](https://community.botletics.com/) that makes it easier to post questions and things. Feel free to join!

sethivansh6, 2020-07-18:
> 1. ssltools.com

> In case it helps anyone, I managed to upload a cert with the following procedure (not using the EFS Explorer)
> 
> **Copy a CA Root Cert (for the site you're trying to connect to) to the SIM7000:**
> 
> 1. Go to **ssltools.com**
> 2. Enter the website you want the root cert for **(e.g. https://putsreq.com)**
> 3. Download the root certificate PEM, which should give you a **.cer** file
> 4. Strip out all carriage returns and then count the characters in the cert file (look at the bottom of VSCode for a character count)
> 
> **Use these commands to load the cert onto the SIM7000:**
> 
> AT+CFSINIT
> AT+CFSWFILE=3,"AddTrust.crt",0,1496,10000 **_<-- 1496 is the char length that noted before_**
> _paste the cert data into the terminal within 10 seconds_
> AT+CFSTERM
> 
> This puts the cert in the 'customer' directory on the SIM7000. Tomorrow I'll see if this lets me make that https post, try your luck with this approach in the meantime.

@TimRoadley Hey I used all the things you stated from downloading and remove carriage return to sending. But I GOT an error while writing the command

AT+CFSWFILE=3,"dweet.crt",0,1931,10000 
DOWNLOAD

ERROR

Do you know any reason why it happened?
do I have to place that file in a certain folder? or do I have to remove Begin certificate and end certificate line?

blazczak, 2020-07-30:
@sethivansh6 ERROR during AT+CFSWFILE points at the module not receiving the (correct) file contents within the self-imposed time (you specified 10000 = 10 seconds). Perhaps there is a mismatch on the number of bytes (you specified 1901). No other content validation is performed in this step, it's just a straight EFS put.

Btw, when working in *nix command line, one can just `wc -c` (or even `ls -l`) the local file to get the exact number of bytes when preparing the transfer.

botletics, 2020-12-09:
Hey guys, please see [this AWS AT command log](https://github.com/botletics/SIM7000-LTE-Shield/blob/master/SIM7000%20Documentation/AT%20Command%20Logs/SIM7000_AWS_Log.txt) from SIMCom. Hope it helps!

tomlankhorst, 2021-05-25:
@botletics Thanks, that helps a bit. I'm looking to configure the module to just validate the server cert. The client will be authenticated via username/password so there is no client cert. 

Did anyone achieve this?

botletics, 2021-05-26:
Please see the `addRootCA()` and `TCPconnect()` functions [here](https://github.com/botletics/SIM7000-LTE-Shield/wiki/Library-Functions#fonaaddrootca). Please also see [this set of AT commands](https://github.com/botletics/SIM7000-LTE-Shield/issues/165#issue-568893959) that another user tried that worked for him.

tomlankhorst, 2021-05-26:
Thank you. In addition, I got a response from SIMCOM:

> The chipset for SIM7000E need CA\client crt\client key , 3 files, can not support only CA. But you can input dummy client crt and client key file, just to "cheat" the stack that it already has 3 file totally.  BTW before connection please update date and time by NTP function, AT+CCLK? should return correct data and time. [...]

davegravy, 2021-05-31:
@botletics I tried following the AWS IoT command log [you posted](https://github.com/botletics/SIM7000-LTE-Shield/issues/58#issuecomment-742135234) however I am getting ERROR at AT+SMCONN.

How can I debug this? The same certificates and key allowed me to use their Python SDK to publish successfully.

```SMS Ready
AT+CCLK?
+CCLK: "21/05/31,04:30:24+00"

OK
AT+CPIN?
+CPIN: READY

OK
AT+CSQ
+CSQ: 29,99

OK
AT+CGREG?
+CGREG: 0,1

OK
AT+COPS?
+COPS: 0,0,"ROGERS ROGERS",7

OK
AT+CGNAPN
+CGNAPN: 1,"ciot"

OK
AT+CNACT=1,"ciot"
OK

+APP PDP: ACTIVE
AT+CNACT?
+CNACT: 1,"10.237.129.141"

OK
at+csslcfg=convert,2,AmazonRootCA1.pem
OK
at+csslcfg=convert,1,my_thing.cert.pem,my_thing.private.key
OK
AT+CSSLCFG="sslversion",0,3
OK
AT+SMSSL=1,AmazonRootCA1.pem,my_thing.cert.pem
OK
AT+SMCONF=url,##############-ats.iot.us-east-1.amazonaws.com,8883
OK
AT+SMCONF="clientid","basicPubSub"
OK
AT+SMCONF="KEEPTIME",60
OK
AT+SMCONF?
+SMCONF
CLIENTID: "basicPubSub"
URL: "##############-ats.iot.us-east-1.amazonaws.com:8883"
KEEPTIME: 60
USERNAME: ""
PASSWORD: ""
CLEANSS: 0
QOS: 0
TOPIC: ""
MESSAGE: ""
RETAIN: 0

OK
AT+SMCONN
ERROR
```

I also tried setting a topic and message (same topic that the python client successfully publishes to) with no luck.

davegravy, 2021-06-01:
I **finally** figured this out!

I now have AT+SMCONN succeeding with AWS (haven't tested anything beyond this yet)

There's 2 important non-obvious steps.

1. Whatever url AWS gives you for your IoT Core endpoint, you have to strip out the "-ats" from it.  So "a1k9ecig9j720o-ats.iot.us-east-1.amazonaws.com" becomes "a1k9ecig9j720o.iot.us-east-1.amazonaws.com"
2. You need to use the legacy root certificate provided by AWS [here](https://docs.aws.amazon.com/iot/latest/developerguide/server-authentication.html) under "VeriSign Endpoints (legacy)". NOTE some regions apparently don't support legacy certs, so it seems the SIM7000 won't work in those.

I also updated my SIM7000A to the B04 firmware from B03, not sure if that mattered. 

Contrary to what @tomlankhorst posted, it doesn't seem to matter if your RTC is sync'd.

davegravy, 2021-06-01:
Not quite there yet... subscriptions work but publishing doesn't. 

```
2019-06-26 14:08:06:424[Send->]AT+SMPUB="ryan","8",1,1

2019-06-26 14:08:06:428[Recv<-]AT+SMPUB="ryan","8",1,1

> test626
2019-06-26 14:08:21:365[Send->]test626
2019-06-26 14:08:21:398[Recv<-]test626
2019-06-26 14:08:27:766[Recv<-]
+SMSTATE: 0

OK

```

The connection drops immediately after publish, and in the AWS (Cloudwatch) logs is this:

```
{
    "timestamp": "2021-06-01 04:56:38.276",
    "logLevel": "ERROR",
    "traceId": "6c4ca615-12d0-93d3-6434-f5f85365cc66",
    "accountId": "1234567890123",
    "status": "Failure",
    "eventType": "Publish-In",
    "protocol": "MQTT",
    "topicName": "ryan",
    "clientId": "basicPubSub",
    "principalId": "82b899e4bcb6bfc158f83be904d9a305e2b21ee255f0d7062cbe6ad3eda05f7a",
    "sourceIp": "xxx.xxx.xxx.xxx",
    "sourcePort": 29573
}
```

davegravy, 2021-06-01:
Publishing works with this:

```
AT+SMPUB="test-topic",8,0,0
> testing1
OK
AT+SMPUB="test-topic",8,1,0
> testing2
OK

```
fails with:

```
AT+SMPUB="test-topic",8,2,0
> testing3
+SMSTATE: 0

OK
```

fails with:

```
AT+SMPUB="test-topic",8,0,1
> testing3
+SMSTATE: 0

OK
```

In summary, server retain can't be used and qos=2 can't be used. Not clear to me if this is a SIM7000 client or AWS limitation.

davegravy, 2021-06-02:
Can't seem to get persistent connections working with AWS

1. After `AT+SMCONF="CLEANSS",0` I connect, then subscribe. Confirm I receive messages on the subscribed topic. 

2. Then I kill the power on the shield, publish a new (qos=1) message to the subscribed topic from another client, power the shield back on, reconnect (SMCONN). Queued message not received. Even new messages on the topic are not received. 

3. Resubscribe to topic (SMSUB), queued messages still not received, but new messages **are** received.

After power up and reconnect in step 2 I see this message in AWS Cloudwatch:

```
{
    "timestamp": "2021-06-02 13:54:53.996",
    "logLevel": "ERROR",
    "traceId": "8133c8d9-621f-d392-0b22-5f55b4c22f3e",
    "accountId": "144349053222",
    "status": "Failure",
    "eventType": "Disconnect",
    "protocol": "MQTT",
    "clientId": "ARMS-GF-D01",
    "principalId": "82b899e4bcb6bfc158f83ac904d9a305e2b21ee255f0d7062cbe6ad3eda05f7a",
    "sourceIp": "74.198.90.117",
    "sourcePort": 16187,
    "reason": "DUPLICATE_CLIENT_ID",
    "details": "A new connection was established with the same client ID",
    "disconnectReason": "DUPLICATE_CLIENTID"
}
```

I think this suggests it is not rejoining the existing persistent session but starting a new one? Anyone know what I'm doing wrong?

Also the last will and testament message isn't getting published to the configured topic at any point after disconnect.

davegravy, 2021-06-09:
While I've been successful getting connections to complete over MQTTS, HTTPS is still not working with ERROR response from SHCONN. 

If I use port 443 for the URL, SHCONN succeeds, but then I get 403 errors to all my requests because they lack SigV4 signing. If I use port 8443, then SHCONN times out.  Note that I'm using the same CA , client cert and key for MQTTS, HTTPS (port 443) and HTTPS (port 8443). I can also `curl` the endpoint with no problem using these certs, so there does appear to be a bug somewhere with HTTPS.

JamesHillman87, 2021-07-28:
@davegravy Thanks for all your updates on getting MQTT working with AWS IoT. Can you please share your final commad log? We currently have an error when we connect. Thanks in advance.

psshiner, 2021-08-19:
> 
> 
> @davegravy Thanks for all your updates on getting MQTT working with AWS IoT. Can you please share your final commad log? We currently have an error when we connect. Thanks in advance.

Hi James, I have been playing around with this today and below is the list of commands i used to both publish and subscribe.

//connecting the device to data network

//Is the SIM ready? does it require a pin
AT+CPIN?
<response>+CPIN: READY OK

//check network strength Signal Quality report lower the first number the better 99 = unknown
AT+CSQ
+CSQ: 14,99 OK

//Check network registration status, 2nd number should be 1 to show that its registered
AT+CGREG?
+CGREG: 0,1 OK

//show operator selection 
AT+COPS?
+COPS: 0,0,"O2 - UK giffgaff",7

//Get Network APN in CAT-M Or NB-IOT
AT+CGNAPN
+CGNAPN: 0,"" OK

//Set prefered modem selection 2 Automatic,13 GSM only,38 LTE only,51 GSM and LTE only
AT+CNMP=13

//select the APN 0,deactive,1 active, 2 auto active
AT+CNACT=1,"giffgaff"
OK
+APP PDP: ACTIVE

//Get ip address of modem
AT+CNACT?
+CNACT: 1,"100.71.118.22" OK

//Set the verisign ca cert into device
at+csslcfg=convert,2,verisignca.pem

//set the device connection to use the device certs
at+csslcfg=convert,1,cert.pem,privatekey.pem

//set ssl version to use at least 1.2
AT+CSSLCFG="sslversion",0,3

//view current connection
AT+SMCONF?

//Set up the certs for the SecureMqtt connection 
//rootca and device cert
AT+SMSSL=1,verisignca.pem,cert.pem
AT+SMCONF="URL",a11hzuhptej14u.iot.eu-west-1.amazonaws.com,8883

//Make sure you set the clientid
AT+SMCONF="clientid","SIM7000"

//Connect to the MQTT broker
AT+SMCONN

//Publish a message
//default policy only allows you to publish on clientid topics,number of characters,QOS(only 0 works),Retain (only 0 works) 
AT+SMPUB="SIM7000/test",8,0,0
> 12345678
OK

AT+SMSUB="SIM7000/#",0

AT+SMSUB="SIM7000",1

chillenberger, 2021-09-14:
Here is a quick list of things I had to do to get the SIM7000A to work with MQTTS:

- I had to update the firmware to 1351B04SIM7000A.  There has been discussion on whether these matters.  Yes, these matters.  It will not work with B03 firmware.  When downloading the firmware remember to jumper the BOOT_CFG pin with 1.8V_EXIT to get the module in download mode.  This is already mentioned elsewhere, but I wanted to reiterate.  Also, when in download mode, my module PWRKEY LED did not illuminate, just be aware. 

- I had to put the ca.crt, client.crt, and client.key in the customer file to get AT+CSSLCFG=”convert”… to work.  This is number 3.

- Be sure to check the minimum tls support on your mqtt broker, AWS and Mosquito need at least 1.1.  This step is not shown in the SIM7000 Series_MQTT(S)_Application Note.  Change the protocol to tls version using AT+CSSLCFG=”sslversion”,

- If using AWS IoT core, you need to use the legacy Endpoints and the correct CA for it, just like davegravy said in an earlier post.  If you do not live in a region that supports the legacy endpoint, you can easily change this in your AWS account in the upper right-hand corner where it says your regain. 

I hope this helps someone.

reddy9698, 2022-03-28:
I've been meaning to implement the process mentioned by @davegravy in a program.
This is the program that I'm using in the void loop part.  

`modem.sendAT("+SMCONN");
        if (modem.waitResponse(1000L, res) == 1) {
            res.replace(GSM_NL "OK" GSM_NL, "");
            Serial.println(res);
            Serial.println("connection!!!!");
        }
   
    res="";

    modem.sendAT("+SMPUB = \"BasicPubSub\", 425,0,0");
        if (modem.waitResponse(10000L, res) == 1) 
        {
            res.replace(GSM_NL "OK" GSM_NL, "");
            Serial.println(res);
            Serial.println("datssent");
            res="";
            
            SerialAT.print(jsonBuffer);
          
            
            
        }

    modem.sendAT("+SMDISC");
        if(modem.waitResponse(1000L,res) == 1)
        {
            Serial.println(res);
            res="";
            Serial.println("disconnectedf");

        }   
          
        res="";  
        
   }`

and this is the message that arrives on the topic 

`AT+SMCONN
AT+SMPUB = "BasicPubSub", 425,0,0
AT+SMDISC
AT+SMCONN
AT+SMPUB = "BasicPubSub", 425,0,0
AT+SMDISC
AT+SMCONN
AT+SMPUB = "BasicPubSub", 425,0,0
AT+SMDISC
AT+SMCONN
AT+SMPUB = "BasicPubSub", 425,0,0
AT+SMDISC
AT+SMC `

What is the mistake that I am doing ?

Scrts, 2022-07-09:
I have succeeded MQTTS to AWS using SIM7070G (Firmware B11) and hologram network.
Follow @davegravy flow, BUT be sure to update module time manually or using NTP to the current time! It did not work without updating to the correct time!

polpol, 2022-08-05:
Hoping someone can give me a 101 on generating / collecting the Certs because that's where I'm stuck.

I'm using a HiveMQ broker that's an AWS server. Can someone point me to a detailed how-to on creating / saving the certificates? I think I pulled one, put it in the customer folder, but no matter what I do, I can't get 
`at+csslcfg=convert,2,AWSCert.pem` 
to get accepted by the SIM7000. Always sends an ERROR code back. I can't tell if it's because the file itself is an issue or something else - like wrong file location.

I've opened up the board I have using QPST, and it doesn't contain any of the folders we see in the CFSWFILE documentation:
0 "/custapp/" 
1 "/fota/" 
2 "/datatx/" 
3 "/customer/"
Tried to load a Cert using CFSWFILE but Putty sucks. Anyone got a better tool to use? I know I should probably not be using Win OS.

I created my own >>customer<< folder and put the Cert there. Also put the Cert in about 10 other folders to see if I could find the location it needs to be in, didn't work. I'm using B04 firmware.

reddy9698, 2022-08-05:
> 

Once you open up the QPST tool, open start clients> EFS Explorer.
The EFS Explorer will show you the primary file system by default. On the tool bar you will find an option for an 
"Alternative File System".
You will find the required folders in this and inside the customer folder you can just drag drop the cert files that you want.
(just make sure that the names and extensions are right when you are running AT commands).

polpol, 2022-08-06:
> > 
> 
> The EFS Explorer will show you the primary file system by default. On the tool bar you will find an option for an "Alternative File System".

This worked beautifully, thank you! I thought I was going crazy.

Now I just need the _right_ certificates...

erocam2012, 2022-08-23:
> @botletics I tried following the AWS IoT command log [you posted](https://github.com/botletics/SIM7000-LTE-Shield/issues/58#issuecomment-742135234) however I am getting ERROR at AT+SMCONN.
> 
> How can I debug this? The same certificates and key allowed me to use their Python SDK to publish successfully.
> 
> ```
> AT+CCLK?
> +CCLK: "21/05/31,04:30:24+00"
> 
> OK
> AT+CPIN?
> +CPIN: READY
> 
> OK
> AT+CSQ
> +CSQ: 29,99
> 
> OK
> AT+CGREG?
> +CGREG: 0,1
> 
> OK
> AT+COPS?
> +COPS: 0,0,"ROGERS ROGERS",7
> 
> OK
> AT+CGNAPN
> +CGNAPN: 1,"ciot"
> 
> OK
> AT+CNACT=1,"ciot"
> OK
> 
> +APP PDP: ACTIVE
> AT+CNACT?
> +CNACT: 1,"10.237.129.141"
> 
> OK
> at+csslcfg=convert,2,AmazonRootCA1.pem
> OK
> at+csslcfg=convert,1,my_thing.cert.pem,my_thing.private.key
> OK
> AT+CSSLCFG="sslversion",0,3
> OK
> AT+SMSSL=1,AmazonRootCA1.pem,my_thing.cert.pem
> OK
> AT+SMCONF=url,##############-ats.iot.us-east-1.amazonaws.com,8883
> OK
> AT+SMCONF="clientid","basicPubSub"
> OK
> AT+SMCONF="KEEPTIME",60
> OK
> AT+SMCONF?
> +SMCONF
> CLIENTID: "basicPubSub"
> URL: "##############-ats.iot.us-east-1.amazonaws.com:8883"
> KEEPTIME: 60
> USERNAME: ""
> PASSWORD: ""
> CLEANSS: 0
> QOS: 0
> TOPIC: ""
> MESSAGE: ""
> RETAIN: 0
> 
> OK
> AT+SMCONN
> ERROR
> ```
> 
> I also tried setting a topic and message (same topic that the python client successfully publishes to) with no luck.

You have a working code sample that connects to AWS MQTT via SIM70XX and ESP32 AND listens to a topic?

Scrts, 2022-08-23:
> > AT+CCLK?
> > +CCLK: "21/05/31,04:30:24+00"
Fix the time. It has to be synchronized. You can use NTP for that.

> > URL: "##############-ats.iot.us-east-1.amazonaws.com:8883"
Remove '-ats' from the URL. You also need to use legacy certificate.
See "Supported legacy endpoints" section here:
https://docs.aws.amazon.com/general/latest/gr/greengrass.html

the '-ats' has been stripped from the endpoint address.

> You have a working code sample that connects to AWS MQTT via SIM70XX and ESP32 AND listens to a topic?
My example here works for me when time is synchronized with NTP:
https://iot.stackexchange.com/questions/6347/connecting-cellular-module-sim7070g-to-aws-mqtt

botletics, 2022-09-08:
Hey guys, figured out how to connect with SSL without verifying certs! Not sure if this would work with AWS though...
Please do the following:
- Set "SSL_FONA" to 1 in the .h file (remember to save the file before closing)
- Download the latest .cpp file from GitHub
- Open the unedited LTE_Demo example sketch and change "http://dweet.io" to "https://dweet.io" on line 1035 (under the '2' option)
- Upload to the Arduino
- Run the 'G' command to enable data connection, then run '2' to send data to dweet.io using SSL. And voila! 😊

zibyan123, 2022-09-27:
> I **finally** figured this out!
>
> I now have AT+SMCONN succeeding with AWS (haven't tested anything beyond this yet)
>
> There's 2 important non-obvious steps.
>
> 1. Whatever url AWS gives you for your [...]

I have used 

> I **finally** figured this out!
>
> I now have AT+SMCONN succeeding with AWS (haven't tested anything beyond this yet)
>
> There's 2 important non-obvious steps.
>
> 1. Whatever url AWS gives you for your [...]

I have downloaded the RSA 2048 bit key: [VeriSign Class 3 Public Primary G5 root CA certificate]) but still I am getting error at AT+SMCONN.

zibyan123, 2022-09-27:
> I **finally** figured this out!
>
> I now have AT+SMCONN succeeding with AWS (haven't tested anything beyond this yet)
>
> There's 2 important non-obvious steps.
>
> 1. Whatever url AWS gives you for your [...]

> I **finally** figured this out!
>
> I now have AT+SMCONN succeeding with AWS (haven't tested anything beyond this yet)
>
> There's 2 important non-obvious steps.
>
> 1. Whatever url AWS gives you for your [...]

I am using sim7000g.I have used RSA 2048 bit key: [VeriSign Class 3 Public Primary G5 root CA certificate] but still I am unable to connect with aws. AT+SMCONN returns error. my firmware version is also upgraded.Need help.?

botletics, 2022-12-04:
Please [see this](https://github.com/botletics/SIM7000-LTE-Shield/issues/156#issuecomment-1332732472)

wenjun1972, 2023-04-04:
I was stuck at the same situation of my 7080G that AT+SMCONN was failed by error,

I followed the guidance from https://github.com/botletics/SIM7000-LTE-Shield/issues/58 but still go nowhere.

"Whatever url AWS gives you for your IoT Core endpoint, you have to strip out the "-ats" from it. So "a1k9ecig9j720o-ats.iot.us-east-1.amazonaws.com" becomes "a1k9ecig9j720o.iot.us-east-1.amazonaws.com" You need to use the legacy root certificate provided by AWS here under "VeriSign Endpoints (legacy)". NOTE some regions apparently don't support legacy certs, so it seems the SIM7000 won't work in those."

I also checked AT+CCLK and saw the time was correct.

Any suggestion ?

thanks !

mrbacani, 2023-08-24:
@wenjun1972 I'm having the exact same issue with the 7080G. Were you able to figure out how to correct it?

Scrts, 2023-08-24:
> @wenjun1972 I'm having the exact same issue with the 7080G. Were you able to figure out how to correct it?
Did you check that you are online with the network in the 1st place? Did you check the time and confirm it's synchronized?

Please copy the whole AT command sequence after you power up.

mrbacani, 2023-08-24:
- Yes, network is online, I know this cause I'm able to connect to other MQTT brokers that don't require SSL.
- I did check and set the time manually, I'm not sure if I did it correctly though. I just used the AT+CCLK command and set it to local time.
- I did upload the certificates and I know the module could find them since the result says DOWNLOAD and when I use the command AT+CFSGFIS to get the file sizes it returns correct values.

The AT command sequence I use to connect is this:
AT+CSQ
AT+CFUN=0
AT+CGDCONT=1,"IP","hologram"
AT+CFUN=1
AT+CNCFG=0,1,"hologram"
AT+CNACT=0,1
AT+CFSINIT
AT+CFSWFILE=3,"veriSign.crt.pem",0,1758,5000
AT+CFSWFILE=3,"certific.pem.crt",0,1220,5000 **Certificate obtained from AWS**
AT+CFSWFILE=3,"private.pem.key",0,1675,5000 **Private key from AWS**
AT+CFSTERM
AT+CSSLCFG="CONVERT",2,"veriSign.crt.pem"
AT+CSSLCFG="CONVERT",1,"certific.pem.crt","private.pem.key"
AT+SMSSL=2,"veriSign.crt.pem","certific.pem.crt"
AT+CSSLCFG="SSLVERSION",0,3
AT+SMCONF="URL",xxxx.iot.ca-central-1.amazonaws.com **I did remove -ats from the URL**
AT+SMCONF="CLIENTID","test"
AT+SMCONN

Thanks for any help!

Scrts, 2023-08-24:
My flow [here](https://iot.stackexchange.com/questions/6347/connecting-cellular-module-sim7070g-to-aws-mqtt) is different than yours a little bit. For example AT+CSSLCFG command. I also did not do AT+CFSTERM. Doesn't it close access to the file system, so certs cannot be pulled in?

mrbacani, 2023-08-24:
So I just tried following your flow exactly but still no connection. I also tried using your method of setting the time but when I do so, AT+CCLK doesn't return the correct time so I just set it manually. 

Would it be because of the region I'm using? I'm using CA-Central and it's not listed on Supported Legacy Endpoints but when I try connecting to AWS with the Legacy certificate using Mosquitto it does work.

Scrts, 2023-08-24:
Regardless of Mosquitto operation, I'd really try switching regions just to be sure.

For the time: why don't use use NTP? Just let the module pull the time on its own using AT+CNTP. Just be sure to select the right parameters, so they match your time zone.

Also, how does your device policy look like in AWS?
I'd recommend using a very loose policy to begin with. Something like:
`{
  "Version": "2012-10-17",
  "Statement": [
    {
      "Effect": "Allow",
      "Action": "iot:*",
      "Resource": "*"
    }
  ]
}`

mrbacani, 2023-08-24:
So it was the region that was the issue. I switched it over to one of the ones listed [here](https://docs.aws.amazon.com/general/latest/gr/greengrass.html) under Supported Legacy Endpoints. Thanks for the help!

AlexSecondSun, 2023-09-22:
How did you sign your client certificates @mrbacani ? How do you obtain the certificate and private key from aws, don you have to use the verisign certificate to sign the client certificates?

explosion33, 2023-09-22:
Having, similar problems. I am using the verisgn g5 Root CA, and AWS generated cert / key files. I followed @mrbacani 's flow, double checked the time, etc. But still get an error. I also get an error when using a python pahomqtt script with the same certs. My current thought is a mismatch between root ca and generated cert / key. Any ideas?

tronar, 2024-02-06:
After a couple of days banging my head to the docs... it worked. AT based MQTT over SSL that is, with client cert id.
There are many little things that can go wrong, and all that you get is an "ERROR".
One that was problematic (in my case) is that I had a proxy ESP32, so certs loading was error prone. But the last one was hard. The key has to be in "traditional" format ("RSA PRIVATE KEY"). If you create the client cert with openssl 3, the new format does not work and "convert",1 fails. You need that for the AT+SMSSL command... I was not able to do SSL w/o specifying a client cert,
Hope this helps someone...

Aryan-Morteza, 2024-02-07:
I have encountered with this issue. The problem seems related to the SSL version3. My private.pem.key is like this:
-----BEGIN RSA PRIVATE KEY-----
something
-----END RSA PRIVATE KEY-----

and certificate.pem.crt:

-----BEGIN CERTIFICATE-----
somrthing
-----END CERTIFICATE-----

Does anybody know how I can do pre-convert on the "certificate.pem.crt" and "private.pem.key" in terminal, then upload it to SIM7000e?

I tried this command but doesn't work again:
- "openssl rsa -in private.key -out private-rsa.key -traditional
- "openssl rsa -in private.key -traditional -out private-rsa.key 

The existed mosquitto.key and mosquitto.crt works fine in AT+CSSLCFG="CONVERT",1,"mosquitto.crt","mosquitto.key".

I also tried to copy and paste my own cert and key in the existed file, but it ain't work.

any advice?

reddy9698, 2024-02-08:
> I have encountered with this issue. The problem seems related to the SSL version3. My private.pem.key is like this: -----BEGIN RSA PRIVATE KEY----- something -----END RSA PRIVATE KEY-----
>
> and [...]

I don't think you need to convert the certificates to use them. The .crt and .key are already in the .pem format. You can just use them directly, one good tip would be rename them to something much more simpler like prvt.key and certificate.crt.

[63 more replies]

closed 2026-11-12 by kittinook