botletics/SIM7000-LTE-Shield #59: Fona libary MQTT read commands

bradleytompkins, 2018-12-19:
I see the functionality for reading MQTT topics is missing from you updated FONA library. Is this something you plan to implement? The Adafruit IO library is quite memory intensive.

botletics, 2018-12-19:
Maybe if I get the time later, no guarantees though :(

robotsflo, 2019-03-06:
+1, 
Do you have some news ? 
I also see MQTT Cloud method in IOT_example -> connect and disconnect TCP connection in loop. 

The advantage of MQTT is maintaning TCP connection to pub or receive message with less data consomption. If you close the TCP connection before waiting 10s, we loose benefits to mqtt. (the opening connect / disconnect MQTT TCP eat some data).

[https://cloud.google.com/blog/products/iot-devices/http-vs-mqtt-a-tale-of-two-iot-protocols](url)

I don't know how the adafruit mqtt IO library was built, but i think that the TCP connection is always "ON".

adam6806, 2019-03-07:
That's really interesting and explains a lot about why the data consumption
I was seeing was higher than expected. I didn't realize that opening and
closing the connection consumed data.

botletics, 2019-03-11:
There are dedicated MQTT commands for the SIM7000 now, but I simply don't have time to implement them due to long term personal commitments.

botletics, 2019-07-23:
As a quick update, I've added the dedicated MQTT commands from [the MQTT app note](https://github.com/botletics/SIM7000-LTE-Shield/blob/master/SIM7000%20Documentation/Technical%20Documents/SIM7000%20Series_MQTT_Application%20Note.pdf) and also added a [SIM7000_MQTT_Demo sketch](https://github.com/botletics/SIM7000-LTE-Shield/blob/master/Code/examples/SIM7000_MQTT_Demo/SIM7000_MQTT_Demo.ino) but haven't tested it yet. The next logical step would be finding out how to read the data coming in after using the subscribe feature.

RickGijsberts, 2019-08-05:
If you subscribe to a topic using the embedded MQTT AT commands, the SIM7000 will send the received data from this topic over UART (or USB) to the MCU (or PC).

**For example:**
Subscribe to topic ‘testTopic’ with QoS 1

AT+SMSUB=”testTopic”,1

If this topic contains retained data on the broker or another device is publishing to this topic, it will be received by the SIM7000 and send to the MCU (or PC) as:

+SMSUB: “testTopic”,”data from this topic”

botletics, 2019-08-05:
Thanks for the heads up! I figured it was that way but didn't remember to test it until I saw this. I just updated the [SIM7000_MQTT_Demo sketch](https://github.com/botletics/SIM7000-LTE-Shield/blob/master/Code/examples/SIM7000_MQTT_Demo/SIM7000_MQTT_Demo.ino) to include reading messages from subscribed topics and parsing them to do stuff (like turn an LED on/off)

closed 2019-08-05 by botletics