botletics/SIM7000-LTE-Shield #61: class Adafruit_FONA_LTE' has no member named 'enableNetworkTimeSync

keirthsiva, 2018-12-27:
I a following the LTE_Demo example. I run across this error which says : class Adafruit_FONA_LTE' has no member named 'enableNetworkTimeSync . I comment this section out and run again and the code is able to compile. However, when I go to send a SMS using the 's' command, it allows me to input a phone number and message but replies with, --->AT+CMGF=1 n/ <--- n/ Failed. Please advise me on how to fix this issue. Thank you.

botletics, 2018-12-27:
I fixed [the LTE_Demo sketch](https://github.com/botletics/SIM7000-LTE-Shield/blob/master/Code/examples/LTE_Demo/LTE_Demo.ino) issue yesterday, please check the code again.

closed 2018-12-27 by botletics

botletics, 2018-12-27:
If you can't even read the CCID then that probably means the SIM card could be bad. That happens, and I've seen heard it reported from users as well. Maybe try swapping it out?

keirthsiva, 2018-12-27:
Would switching it out with a T-Mobile sim from my own phone (which i know the sim works on) work for this project?

botletics, 2018-12-27:
Not if you're in the US. The shield requires LTE CAT-M or NB-IoT to work and T-Mobile doesn't have nation-wide NB-IoT yet as far as I know.

keirthsiva, 2018-12-27:
Okay, I will get another Hologram sim and try it out. Thank you for the prompt replies. I appreciate it.

keirthsiva, 2018-12-27:
Quick follow up...if the hologram sim is bad, then it should say so in the hologram.io website/dashboard. However, when I log in, it says that the SIM "was active 'x' minutes ago". Do you still think it is a SIM problem?

botletics, 2018-12-27:
Have you perhaps used up your SMS quota for the SIM card? You say you can't read the CCID of the SIM card. That's either a SIM card issue or hardware issue. What hardware are you using?