botletics/SIM7000-LTE-Shield #62: SIM7000E with B05 No SSL on UART ( second call to "AT+CARECV=0,1" CRASH the modem)

TechSgin, 2018-12-29:
Hi All and thanks in advance.

I am using this modem with great fun on both GSM and LTE CAT M (no NB yet).

Still, I am failing to use SSL on UART, The second call to "AT+CARECV=0,1" CRASH the modem.
Note: this doesn't happen on USB.
I have tried AT&F0 "Factory Reset" and there is no change.
I have disabled AT+IFC=0,0 "Flow Control" 

Are you seeing the same behavior ?

set of commands:
```
AT+CNACT=1, "apn"
AT+CACID=0
AT+CASSL=0,0,0
AT+CAOPEN=0,"www.youraddress.com",8883
AT+CARECV=0,1 <- OK
AT+CARECV=0,1 <- crash after few seconds
```

Thanks
Oren

botletics, 2018-12-29:
Sorry I have no way if testing that (I'm in the US and SIM7000E won't work here).

TechSgin, 2018-12-29:
Thanks, does these command work for you on SIM7000A or SIM7000G ?

botletics, 2018-12-30:
I haven't played with SSL much but I can't even get some other SSL functions to run properly on SIM7000G.

botletics, 2018-12-30:
I tried SIM7000A before and SSL commands worked fine. However, I didn't try the AT+CARECV command specifically.

jaka87, 2022-07-25:
Did you manage to solve this ?
I have similar issue, when trying to read response afterwords in bytes it crashes arduino. If i manually enter AT commands it works but when i try to parse it craches.
I would appreciate if you could share how you parse the response from AT+CARECV