botletics/SIM7000-LTE-Shield #63: library issue with SIM5320

stephenmorton, 2019-01-15:
MQTT Demo on SIM5320

Hi Timothy, thanks for the great library. Your board looks great also.

I have installed your library and trying to run your MQTT demo on SIM5320 hardware (Adafruit 3G card).

I have attached the output of the serial monitor. Basically, It seems to get stuck after outputting the Module IMEI number. I think it's having an issue connecting perhaps? Any help would be greatly appreciated. I have tried lots of libraries for this board and yours seems to be the only one that claims to work with this board.
[image]

Thanks,

Steve

botletics, 2019-01-15:
MQTT on the SIM5320 has not been implemented because it has dedicatedQTT commands.

stephenmorton, 2019-01-15:
Thanks Timothy, I know it's not related to your hardware, but would you know of any samples or examples of anyone using these dedicated Mqtt commands for the sim5320. I'm just having trouble finding anything useable on Mqtt with the 5320. Thanks

botletics, 2019-01-15:
I don't know of any, sorry.

botletics, 2019-01-19:
Yup the library supports HTTP and HTTPS for SIM5320 with Arduino.

botletics, 2019-06-10:
Please see #11

closed 2019-06-10 by botletics