botletics/SIM7000-LTE-Shield #65: Connecting to hardware serial

ammarqs, 2019-01-18:
Timothy,

I am using the arduino mega and noticed that to use the tcpsend, it is better to use the hardware serial for better baud rates. To set up the hardware serial, please confirm the following changes to the program.

1. Remove all softwareSerial fonass function definitions. 
2. in the modulesetup function, change the fonass to Serial3 (or whatever serial port the system is connected to)
3. Change the wiring from D10 to 15 on mega, and D11 to 14 on mega.

botletics, 2019-01-18:
With Arduino Mega if you change the pins from the default D10/D11 software serial pins, the shield won't be directly plug-in compatible.

I am aware of hardware serial allowing faster baud rates, and this is mentioned in a couple places in the Github wiki as well. However, I've had no issues with 9600 baud rate on software serial and it's good enough for most people's applications.

ammarqs, 2019-01-18:
Its ok if its not direct plugin compatible. We will be designing our own controller board to fit this shield.
So the pins D10/D11 are software serial on the shield? it wont work if i connect the D10 to 15 and D11 to 14 on mega. ?

Right, it is. but when trying to send a small image over through the tcpsend function, it takes more than 30secs to a minute to send a 5kb image.

I changed the program to this, and it says couldn't connect:
`Serial.begin(115200);
 Serial3.begin(115200); // Default SIM7000 shield baud rate
  if (! fona.begin(Serial3)) {
    Serial.println(F("Couldn't find FONA"));
    while(1); // Don't proceed if it couldn't find the device
  }
`
It returns back could not find FONA.

botletics, 2019-01-18:
Check out lines 77-89 of the [LTE_Demo sketch](https://github.com/botletics/SIM7000-LTE-Shield/blob/master/Code/examples/LTE_Demo/LTE_Demo.ino)

botletics, 2019-01-18:
[And this](https://github.com/botletics/SIM7000-LTE-Shield/wiki/Pinouts#hardware-serial-note)

botletics, 2019-01-18:
For the code, the [SAMD_LTE_Demo sketch](https://github.com/botletics/SIM7000-LTE-Shield/blob/master/Code/examples/SAMD_LTE_Demo/SAMD_LTE_Demo.ino) will be particularly helpful. It uses hardware serial.

closed 2019-01-18 by botletics

ammarqs, 2019-01-18:
Thank you. that helped. But i am still getting the same couldnt connect error:

`

// Hardware serial is also possible!
HardwareSerial *fonaSerial = &Serial3;

void setup(){
  Serial.begin(115200);
 powerOn(); // Power on the module
  moduleSetup(); 

void moduleSetup() {
  // Note: The SIM7000A baud rate seems to reset after being power cycled (SIMCom firmware thing)
  // SIM7000 takes about 3s to turn on but SIM7500 takes about 15s
  // Press reset button if the module is still turning on and the board doesn't find it.
  // When the module is on it should communicate right after pressing reset
  Serial3.begin(115200); // Default SIM7000 shield baud rate
 // fonaSerial->begin(115200);
 // Serial.println(F("Configuring to 115200 baud"));
 // Serial3.println("AT+IPR=115200"); // Set baud rate
 // delay(100); // Short pause to let the command run
 // Serial3.begin(115200);
  if (! fona.begin(*fonaSerial)) {
    Serial.println(F("Couldn't find FONA"));
    while(1); // Don't proceed if it couldn't find the device
  }
`
any comments?

botletics, 2019-01-18:
Pin 14 on the Mega should be connected with pin 11 on the shield, and pin 15 on the Mega should go to pin 10 on the shield.

botletics, 2019-01-18:
And get rid of "Serial3.begin(115200)" because you don't need it. What you need is

```
// Communicate with SIM7000 module at with hardware serial
// With hardware serial there's no need to change baud rate
fonaSerial->begin(115200); // Default SIM7000 baud rate

if (! fona.begin(*fonaSerial)) {
  Serial.println(F("Couldn't find SIM7000"));
  while (1); // Don't proceed if it couldn't find the device
}
```

If you just follow the example code it will work. Sorry, but I'm spending a lot of time helping people with this kind of stuff and it adds up.

ammarqs, 2019-01-18:
no you are right. I tried that to. If you look in my code, i commented that because it didnt work.

botletics, 2019-01-18:
Is your shield even turning on? Check the green PWR LED and make sure you have a 3.7V LiPo battery connected with the right polarity (with red wire on the left side of the connector), etc.

ammarqs, 2019-01-18:
yes. If i run my Software serial its runs perfectly. no complains other than slower speeds. That is why i am trying to move on to hardware serial. I changed the program and pins to back to what it was, and it runs fine.

botletics, 2019-01-18:
Since your module is probably at 9600 baud rate, try sending the command AT+IPR=9600 to the module first, then run the code I pasted above and it should work.

ammarqs, 2019-01-18:
i did reset the module to revert any changes in baud rates made previously.
I will try what you said regardless.

botletics, 2019-01-18:
Sorry, I meant AT+IPR=115200 to set it back to 115200 so that it will work with hardware serial.

ammarqs, 2019-01-18:
Serial3.println(" AT+IPR=115200"); RIGHT?

ammarqs, 2019-01-18:
no luck. :(

botletics, 2019-01-18:
That won't do anything because Serial3 is set to 115200, and if your module is at 9600 it won't even receive the command. So what you should do is either run the LTE_Demo sketch with software serial and use the "S" command according to [this wiki page](https://github.com/botletics/SIM7000-LTE-Shield/wiki/Testing-AT-Commands) to send AT commands via serial monitor, or you plug in the micro USB to the shield directly and send the command via a serial terminal like PuTTy by following [these instructions](https://github.com/botletics/SIM7000-LTE-Shield/wiki/USB-Drivers)

ammarqs, 2019-01-18:
so i have to connect it back to software serial and run the command using the 'S' function to set it at 115200 and then configure the hardware serial again and try running. Right. Just trying to get it right so that i dont blow up my module.

botletics, 2019-01-18:
Correct, so two options:
* Connect it back to software serial wiring
* Run LTE_Demo with software serial
* Use the "S" command and send "AT+IPR=115200"
* Then use hardware serial wiring and baud rate of 115200

OR (easiest method):
* Plug in USB to the shield directly and send AT+IPR=115200

ammarqs, 2019-01-18:
Done. Now it works. Thank you.

ammarqs, 2019-01-18:
This is very nice. :+1:  
The baud rate changed the game here. less than 14 secs to send the image out. Now i am going to try and go faster. Under 10sec would be ideal for our application. but then again, this is only a 5kb file. 
Would you mind telling me the max SAFE baud rate for the sim7000 module?

botletics, 2019-01-18:
I actually haven't tested the max baud rate for hardware serial. I suppose it depends on the microcontroller itself so I didn't bother. Don't be shy, it won't blow up anything. Feel free to try 912600. According to the AT command manual the SIM7000 can handle up to a baud rate of 4000000.

That being said, I read [from this random article I just googled](https://copperhilltech.com/blog/rs232-to-usb-converter-with-arduino-due-or-mega-2560/) that:

> The additional three serial ports on the Arduino Due and Arduino Mega 2560 run only reliably up to 115200 baud. A baud rate of 230400 is possible but you have to deal with random data loss.

ammarqs, 2019-01-18:
yes thats exactly what i read. So apparently i am limited to only 115200 baud rate. i tried doing a 55kb picture, it took almost 3 minutes to completely receive the file data on the server.

botletics, 2019-01-18:
Use the USB method to send AT commands via USB. With USB you can connect at 9600 or almost any baud rate you'd like.

ammarqs, 2019-01-19:
I had to go in detail with the baud rate setting to match both arduino and sim7000 module.
And according to the atmega2560 datasheet, it can safely operate upto 1M bps at 16Mhz.  But sim7000 has different values. 
The values that match for both above 115200 is only 230400, nothing above that would match both the controller and the sim7000, so i guess i can only go upto 230400.
I havent tried this yet but in theory it seems legit.