botletics/SIM7000-LTE-Shield #66: Update SIM7000E firmware

skeleton0, 2019-02-05:
Hi Tim,

Would you be able to get the latest SIM7000E firmware revision? I know there's at least a B06 version out there that I found on another website, not sure if that's the latest though.

I tried emailing SIMCom but they won't reply to me.

botletics, 2019-02-05:
I'll ask.

skeleton0, 2019-02-12:
They replied today so I have a B07 version. Let me know if you want me to make a pull request adding it.

vAnArhist, 2019-02-13:
> They replied today so I have a B07 version. Let me know if you want me to make a pull request adding it.

Could you provide this firmware to me?
It seems that in B06 version SSL AT commands is not working

botletics, 2019-02-13:
@skeleton0 Please create pull request or email to me, thanks.

closed 2019-02-13 by botletics

skeleton0, 2019-02-14:
@vAnArhist I was looking for an updated firmware revision because I believe the HTTP(S) AT command set (AT+SHREQ etc.) isn't working. Unfortunately from my limited testing of the B07 version, that AT command set is still not working. I'm yet to try the alternative TLS over TCP/UDP AT command set. Let me know if you have more success.

dagoch, 2019-10-03:
Has anyone figured out where to find the latest firmware for the 7000A?  There's nothing on the simcom website.  This github only has the B04 version.  I also need HTTPS support.  ( I have Botletics board v.6 and it only has the B03 firmware.)

skeleton0, 2019-10-04:
@dagoch I got the latest firmware revision by emailing SIMCom requesting it.

dagoch, 2019-10-05:
@skeleton0 Can you send me the email address you used to reach them?  I haven't had any luck so far but I'm not sure I'm emailing the right company.  I can't seem to find a legit website for simcom.

Also, have you figured out which firmware versions actually support HTTPS?

Thanks.

skeleton0, 2019-10-07:
@dagoch Sorry for the slow reply. I received a firmware file by emailing support@simcom.com

I've currently got working HTTPS communications on my SIM7000E. However each module version has different corresponding firmware revisions, and I am not aware of what firmware revision for the SIM7000A supports HTTPS.

Additionally, be aware that while a firmware revision may support the HTTPS AT commands, HTTPS may not actually work properly in practice until a later firmware revision due to bugs etc.

Send an email to the address that I supplied above requesting the latest firmware revision for the SIM7000A and I'd imagine the HTTPS AT commands will be supported and working.

sethivansh6, 2020-07-17:
> They replied today so I have a B07 version. Let me know if you want me to make a pull request adding it.

Can you give me the B07 version?

sethivansh6, 2020-07-17:
> @dagoch Sorry for the slow reply. I received a firmware file by emailing [support@simcom.com](mailto:support@simcom.com)
> 
> I've currently got working HTTPS communications on my SIM7000E. However each module version has different corresponding firmware revisions, and I am not aware of what firmware revision for the SIM7000A supports HTTPS.
> 
> Additionally, be aware that while a firmware revision may support the HTTPS AT commands, HTTPS may not actually work properly in practice until a later firmware revision due to bugs etc.
> 
> Send an email to the address that I supplied above requesting the latest firmware revision for the SIM7000A and I'd imagine the HTTPS AT commands will be supported and working.

@skeleton0 can you share the code used for https request with us.

skeleton0, 2020-07-19:
@sethivansh6 You can find the firmware revisions (one of which is the B07 revision for the SIM7000E module) here: [https://github.com/botletics/SIM7000-LTE-Shield/tree/master/SIM7000%20Documentation/Firmware](url)

For the HTTPS code, have a look at https://github.com/skeleton0/trackerFirmwareHttp/blob/master/Sim7kInterface.cpp and look at the functions with "https" in the identifier. Perhaps look at https://github.com/skeleton0/trackerFirmwareHttp/blob/master/trackerFirmwareHttp.ino to see how those functions are used.

sethivansh6, 2020-07-19:
> @sethivansh6 You can find the firmware revisions (one of which is the B07 revision for the SIM7000E module) here: [...]

@skeleton0  Thanks for your help. I have gone through your code and I have a doubt that doesn't we need an SSL certificate for connecting with https server or we can do it by TLS (AT+CSSLCFG="sslversion",0,3) as in your case. When I was going through SIMCOM AT commands in which they import certificates to flash which made me think that it's important to import certificates. Is it possible to do this without that as well?

skeleton0, 2020-07-29:
@sethivansh6 Apologies for getting back to this so late. You're probably right that root certs or something need to be imported to authenticate the certificates signed by certificate authorites. At the time my concern was getting HTTPS working and not proper security. I assume my code was probably just set to blindly trust my self-signed server certificate I was using.