botletics/SIM7000-LTE-Shield #68: Full sequence to make get or post request

laurentvm, 2019-02-13:
Hi, 

I am not using your library but I am using a SIM7600E with ESP32. Before we were using SIM5360 and we were able to make TCP request using:
AT+CIPMODE=0
AT+NETOPEN
AT+CHTTPSSTART
AT+CHTTPSOPSE

That seems not to work anymore on the SIM7600.
I therefore tried the HTTP layer commands of the SIM7600. Using this sequence it seems to work more or less:
 -|Command issued AT+COPS?
|- Answer :
 AT+COPS?
+COPS: 0,0,"Tango TANGO",7

OK
 -|
 -|Command issued AT+CGATT=1
|- Answer :
 AT+CGATT=1
OK
 -|
 -|Command issued AT+CGDCONT=1,"IP","internet"
|- Answer :
 AT+CGDCONT=1,"IP","internet"
OK
 -|
 -|Command issued AT+CGSOCKCONT=1,"IP","internet"
|- Answer :
 AT+CGSOCKCONT=1,"IP","internet"
OK
 -|
 -|Command issued AT+CGAUTH=1,0
|- Answer :
 AT+CGAUTH=1,0
OK
 -|
 -|Command issued AT+CSOCKAUTH=1,0
|- Answer :
 AT+CSOCKAUTH=1,0
OK
 -|
 -|Command issued AT+CSOCKSETPN=1
|- Answer :
 AT+CSOCKSETPN=1
OK
 -|
 -|Command issued AT+IPADDR
|- Answer :
 AT+IPADDR
+IP ERROR: Network not opened

ERROR
 -|
 -|Command issued AT+IPADDR
|- Answer :
 AT+IPADDR
+IP ERROR: Network not opened

ERROR
 -|
 -|Command issued AT+IPADDR
|- Answer :
 AT+IPADDR
+IP ERROR: Network not opened

ERROR
 -|
ERR, AT+IPADDR.
 -|Command issued AT+HTTPTERM
|- Answer :
 AT+HTTPTERM
OK
 -|
 -|Command issued AT+HTTPINIT
|- Answer :
 AT+HTTPINIT
OK
 -|
 -|Command issued AT+HTTPPARA="URL","http://www.jvmmoto.be"
|- Answer :
 AT+HTTPPARA="URL","http://www.jvmmoto.be"
OK
 -|
 -|Command issued AT+HTTPACTION=0
|- Answer :
 AT+HTTPACTION=0
OK
 -|
 -|Command issued AT+HTTPHEAD
|- Answer :
 125 yamaha  , vendre sa moto , bottes , garage moto liege , ballade magasin , moto a vendre liege , moto yamaha ,   , concessionnaire moto yamaha , , vendeur yamaha  , ballade liege  , yamaha concessionnaires  , , concession yamaha  , piece occasion moto yamaha  , accessoire moto liege  , concessionnaire moto  , magasin accessoires moto  , magasin occasion liege  , magasin scooter liege  , moto liege  , financement moto  , accessoire moto  ,  moto occasion, yamaha,
OK
 -|
 -|Command issued AT+HTTPREAD=0,500
|- Answer :
 AT+HTTPREAD=0,500
+HTTPREAD: 292
div class="clear"></div><script src="http://www.google-analytics.com/urchin.js" type="text/javascript">
</script>
<script type="text/javascript">
_uacct = "UA-3931058-1";
urchinTracker();
</script>
<script type="text/javascript" src="http://www.jvmmoto.be/min/?g=js"></script></body></html>

OK
 -|
 -|Command issued AT+IPADDR
|- Answer :
 AT+IPADDR
+IP ERROR: Network not opened

But it does not work with POST.

Would you have the time to list the right sequence of commands that should be issue to the module?

I cannot find any good information but your library (if working) seems to be able to do the job.

Thanks

botletics, 2019-02-13:
Try the SIM7500 commands [from this wiki page](https://github.com/botletics/SIM7500-LTE-Shield/wiki/Shield-Demo) (SIM7500 is the same as SIM7600, just different number of pins)

closed 2019-02-13 by botletics

laurentvm, 2019-02-14:
Hi Timothy,

Thanks for your hint. I reproduced the command from the wiki but get an
error after CHTTPSTART (see red below).
I'm doing exactly the same as in your wiki but still not working.

What is the difference between CHTTPSTART and HTTPINIT ?

If you have time to look trough this that would really help me as I cannot
figured out how to make this work on the new sim module.

Laurent

Command issued AT+CMEE=2
|- Answer :
 AT+CMEE=2
OK
 -|
 -|Command issued ATE0
|- Answer :
 ATE0
OK
 -|
 -|Command issued ATI
|- Answer :

Manufacturer: SIMCOM INCORPORATED
Model: SIMCOM_SIM7600E-H
Revision: SIM7600M22_V1.1
IMEI: 867584030418060
+GCAP: +CGSM

OK
 -|
 -|Command issued AT+CPIN?
|- Answer :

+CPIN: READY

OK
 -|
OK
 -|Command issued AT+CPSI?
|- Answer :

+CPSI:
LTE,Online,270-77,0x0002,1279845,283,EUTRAN-BAND20,6300,3,3,-116,-1159,-870,9

OK
 -|
 -|Command issued AT+COPS?
|- Answer :

+COPS: 0,0,"Tango TANGO",7

OK
 -|
 -|Command issued AT+CREG?
|- Answer :

+CREG: 0,1

OK
 -|
 -|Command issued AT+CSQ
|- Answer :

+CSQ: 12,99

OK
 -|
 ==> SIMCOM module registered, gonna set parameters...

 -|Command issued AT+NETCLOSE
|- Answer :

OK
 -|
 -|Command issued AT+CHTTPSTOP
|- Answer :

+NETCLOSE: 0

ERROR
 -|
 -|Command issued AT+CGATT=1
|- Answer :

OK
 -|
 -|Command issued AT+CGSOCKCONT=1,"IP","internet"
|- Answer :

OK
 -|
 -|Command issued AT+CIPMODE=1
|- Answer :

OK
 -|
 -|Command issued AT+NETOPEN
|- Answer :

OK
 -|
 -|Command issued AT+CHTTPSTART
|- Answer :

+NETOPEN: 0

ERROR
 -|
 -|Command issued AT+CHTTPSOPSE="www.dweet.io",443,2
|- Answer :

ERROR
 -|
 -|Command issued AT+CHTTPSEND=81
|- Answer :

ERROR
 -|
 -|Command issued AT+CHTTPSRECV=378
|- Answer :

botletics, 2019-02-14:
Sorry, don't know.