botletics/SIM7000-LTE-Shield #69: Failed to connect to any network

frnk94, 2019-02-13:
The shield seems to be working fine ! But when I try to query Network information it doesn't work
-
[image]
[image]
[image]

frnk94, 2019-02-13:
Seems like it can read the SIM just fine im located in Canada btw and as you can see im using hologram sim card

botletics, 2019-02-13:
The Hologram SIM won't work in Canada, sorry. You need Bell's LTE CAT-M SIM card or other compatible SIM.

closed 2019-02-13 by botletics

frnk94, 2019-02-14:
Alright thanks for quick response can you just point me where its saids hologram won't work in canada or how you knew that was the problem I just want to make sure while im looking for new SIMS that the problem wont happen again. Because on Hologram website they say they covered Canada

[image]

frnk94, 2019-02-14:
Also Bell's charge 200$ for their IoT SIMS card... I hate this company ahahah

botletics, 2019-02-14:
I know because other users have confirmed that it won't work there. Also, I know from Hologram themselves that the SIM doesn't yet support LTE CAT-M in Canada. It covers Canada for 2G, 3G, and 4G LTE but not LTE CAT-M or NB-IoT.

frnk94, 2019-02-14:
Alright ! Ty for everything Also nice project