botletics/SIM7000-LTE-Shield #71: How to import certificates on SIM7000

kaizoku-619, 2019-02-18:
I can't seem to find the appropriate AT commands to import certificates to SIM7000
simcom documents only describe how to convert certificates on the SIM7000 and not how to import them on its file system.

tmcadam, 2019-02-24:
@kaizoku-619 I am having the same issue. I have a feeling the commands aren't implemented yet through the AT interface. I have tried a few things, including copying certificates in using the QPST tool. There are folders on the device called 'cert', 'client-cert', 'client-key'. But have tried putting .pem formatted keys in there and trying the 'convert' with AT+CSSLCFG command, doesn't work. I also found some info in the QAPI docs, and tried converting the certs and keys using the SharkSSLParseCert.exe, and then uploading them using QPST again, still the same error. 

https://github.com/Thalhammer/simcom_dam/blob/master/docs/MQTT.md has tried secure MQTT through the QAPI functions and has been having problems too.

botletics, 2019-02-24:
Just see [these AT command logs](https://github.com/botletics/SIM7000-LTE-Shield/tree/master/SIM7000%20Documentation/AT%20Command%20Logs)

closed 2019-02-24 by botletics

tmcadam, 2019-02-25:
Thanks @botletics . Tested these last night and am able to connect to test.mosquitto.org on 8883 and 8884 (with and without client certificate).

captFuture, 2019-03-14:
@tmcadam would you please share an end to end example with us. 
I am trying hard to get it working, but just looking at the logs still does not tell how to get the certificates into the modem (the AT command just reserves the necessary space and defines a filename. 
Thanks
Chris

tmcadam, 2019-03-17:
@captFuture Yes, happy to share. I was using a python script that runs on desktop and shoots off all the AT commands and can also upload the keys if required. I'll clean it up and share it asap (although a new baby here might mean it will take a day or two ;-) ). You can copy the certificates into the device using the QPST tool. The trick is putting them in the right place, there are two file systems on the device, it needs to go in the secondary one, and into a folder there called 'customer'. Keep the key names short and no special characters. I'll try and get a couple of screen shots of QPST if you need help with that method.

captFuture, 2019-03-18:
> @captFuture Yes, happy to share.

Thanks a lot, the python approach sounds very promising. I was trying to do things with qpst but somehow can't connect the modem correctly and therefore i can't upload anything :(

All the best

cmwi2k, 2019-03-20:
I have been playing with this for a while now and using the B07 firmware from this repo managed to get the following.

It seems that setting AT+SHSSL=1,"comodo.crt" to the certificate in the Alternate File System/customer directory as @tmcadam said then setting it to AT+SHSSL=1,"" allows the connection to go through otherwise it throws ERROR.

The comodo.crt is the comodo root ca certificate put on by QPSF EFS Explorer then Alternate File System which is available on the repo under SIM7000-LTE-Shield/tree/master/SIM7000 Documentation/Firmware Updater Tool

I am getting an error 400 but it has come via my apache server via SSL so it may be a configuration issue on there.

My server uses SNI so I put the Host header in AT+SHAHEAD="Host","www.yourdomain.com" once that was in it returned the error 400 data

I will try this on the B05 firmware and see if that makes any difference.

I'm using the SIM7000E in the UK

Hope that helps someone

AT+GMR
Revision:1351B07SIM7000E

OK
AT+CNACT=1,"your.apn.com"
OK

+APP PDP: ACTIVE
AT+CNACT?
+CNACT: 1,"your.ip.address"

OK
AT+SHCONF="URL","https://www.yourdomain.com"
OK
AT+SHCONF="BODYLEN",350
OK
AT+SHCONF="HEADERLEN",350
OK
AT+CSSLCFG="convert",2,"comodo.crt"
OK
AT+SHSSL=1,"comodo.crt"
OK
AT+SHSSL=1,""
OK
AT+SHCONN
OK
AT+SHBOD="SIMCOMBODY",10
OK
AT+SHAHEAD="Host","www.yourdomain.com"
OK
AT+SHSTATE?
+SHSTATE: 1

OK
AT+SHREQ="/index.html",1
OK

+SHREQ: "GET",400,434
AT+SHREAD=0,434
OK

+SHREAD: 434
<!DOCTYPE HTML PUBLIC "-//IETF//DTD HTML 2.0//EN">
                                                  <html><head>
                                                              <title>400 Bad Request</title>
            </head><body>
                         <h1>Bad Request</h1>
                                             <p>Your browser sent a request that this server could not understand.<br />
                                        </p>
                                            <p>Additionally, a 400 Bad Request
                                                                              error was encountered while trying to use an ErrorDocument to handle the request.</p>
   <hr>
       <address>Apache/2.4.7 (Ubuntu) Server at www.yourdomain.com Port 443</address>
        </body></html>

CoolNamesAllTaken, 2019-03-24:
I'm late to the party, but really glad to find other people also working on this!  Does anyone know if transferring certificates is possible over the AT command interface?  I know that @tmcadam aluded to a fabled python script, but I haven't seen anything about how to upload certificates to the SIM7000 in the SIMCOM datasheets.

tmcadam, 2019-03-24:
@CoolNamesAllTaken 
@cmwi2k 
@captFuture

I've uploaded my python script [sim7000-tools](https://github.com/tmcadam/sim7000-tools). It's a work in progress and I have a bunch more commands that I am in the process of cleaning and adding. Anyway it has the uploading certs using AT and also has a working example using test.mosquitto.org using CA Cert and authenticating with client cert and key.

I'm blocked using this module at the minute as I can't connect to AWS IoT. One guess is that even though we are specifying TLS1.2, it isn't happening under the hood. test.mosquitto.org supports all TLS versions (so works), while AWS needs version 1.2. If anyone gets this to work with AWS, please let me know!!

@CoolNamesAllTaken Can you let us know how this works with Azure.

captFuture, 2019-03-24:
Thanks Tom, I am having the AWS issue as well and am investigating. 

Managed to upload the certs using qpst and did a successful upload using c but unfortunately then destroyed my code because I thought it didn't work cause the certificates didn't load afterwards (found out later that the filename I used was too long and it got truncated ) 
:/ 
I will post it when I figured out again how it worked. 

Cheers
C

CoolNamesAllTaken, 2019-03-25:
@tmcadam your sim7000-tools library looks fantastic, I'll try generating some keys and I'll let you know how it goes!

captFuture, 2019-03-27:
@tmcadam I was testing this a bit todayand when it comes to 

AT+CSSLCFG="convert",1,"awscert.crt", "awskeks.key"

i just get a response +CME ERROR: operation not allowed
I wonder why this happens because the certs are valid and i have no issue with the mosquitto ones

CoolNamesAllTaken, 2019-03-27:
I had this issue too, it's caused when the certs and key pair is not an RSA key pair.  Take a look at the issue on @tmcadam's repo that I posted.  It has a log of me encountering the exact same error :p

JGLJGL, 2020-01-22:
> I have been playing with this for a while now and using the B07 firmware from this repo managed to get the following.
> 
> It seems that setting AT+SHSSL=1,"comodo.crt" to the certificate in the Alternate File System/customer directory as @tmcadam said then setting it to AT+SHSSL=1,"" allows the connection to go through otherwise it throws ERROR.
> 
> The comodo.crt is the comodo root ca certificate put on by QPSF EFS Explorer then Alternate File System which is available on the repo under SIM7000-LTE-Shield/tree/master/SIM7000 Documentation/Firmware Updater Tool
> 
> I am getting an error 400 but it has come via my apache server via SSL so it may be a configuration issue on there.
> 
> My server uses SNI so I put the Host header in AT+SHAHEAD="Host","www.yourdomain.com" once that was in it returned the error 400 data
> 
> I will try this on the B05 firmware and see if that makes any difference.
> 
> I'm using the SIM7000E in the UK
> 
> To use the AT commands I used putty on the serial connection to the relevant COM port
> 
> Hope that helps someone
> 
> AT+GMR
> Revision:1351B07SIM7000E
> 
> OK
> AT+CNACT=1,"your.apn.com"
> OK
> 
> +APP PDP: ACTIVE
> AT+CNACT?
> +CNACT: 1,"your.ip.address"
> 
> OK
> AT+SHCONF="URL","https://www.yourdomain.com"
> OK
> AT+SHCONF="BODYLEN",350
> OK
> AT+SHCONF="HEADERLEN",350
> OK
> AT+CSSLCFG="convert",2,"comodo.crt"
> OK
> AT+SHSSL=1,"comodo.crt"
> OK
> AT+SHSSL=1,""
> OK
> AT+SHCONN
> OK
> AT+SHBOD="SIMCOMBODY",10
> OK
> AT+SHAHEAD="Host","www.yourdomain.com"
> OK
> AT+SHSTATE?
> +SHSTATE: 1
> 
> OK
> AT+SHREQ="/index.html",1
> OK
> 
> +SHREQ: "GET",400,434
> AT+SHREAD=0,434
> OK
> 
> +SHREAD: 434
> 
> ```
>                                               <html><head>
>                                                           <title>400 Bad Request</title>
>         </head><body>
>                      <h1>Bad Request</h1>
>                                          <p>Your browser sent a request that this server could not understand.<br />
>                                     </p>
>                                         <p>Additionally, a 400 Bad Request
>                                                                           error was encountered while trying to use an ErrorDocument to handle the request.</p>
[... 36 lines cut ...]
AT+SHAHEAD="Host","www.atesens.com"                                             
OK                                                                              
AT+SHSTATE?                                                                     
+SHSTATE: 1                                                                     
                                                                                
OK                                                                              
AT+SHREQ="/index.html",1                                                        
OK                                                                              
                                                                                
+SHREQ: "GET",400,800                                                           
AT+SHREAD=0,800                                                                 
OK                                                                              
                                                                                
+SHREAD: 800                                                                    
<!doctype html><html lang="en"><head><title>HTTP Status 400 â€“ Bad Request</tit
le><style type="text/css">h1 {font-family:Tahoma,Arial,sans-serif;color:white;ba
ckground-color:#525D76;font-size:22px;} h2 {font-family:Tahoma,Arial,sans-serif;
color:white;background-color:#525D76;font-size:16px;} h3 {font-family:Tahoma,Ari
al,sans-serif;color:white;background-color:#525D76;font-size:14px;} body {font-f
amily:Tahoma,Arial,sans-serif;color:black;background-color:white;} b {font-famil
y:Tahoma,Arial,sans-serif;color:white;background-color:#525D76;} p {font-family:
Tahoma,Arial,sans-serif;background:white;color:black;font-size:12px;} a {color:b
lack;} a.name {color:black;} .line {height:1px;background-color:#525D76;border:n
one;}</style></head><body><h1>HTTP Status 400 â€“ Bad Request</h1></body></html>
`

JGLJGL, 2020-01-23:
@tmcadam  I've achieved the previous steps, but don't know how to add the certificates. 
I want to access to https://www.atesens.com, so if you could explain how to get the certificates I'd appreciate your help, as I'm new and I need to know this for a project.