botletics/SIM7000-LTE-Shield #75: IoT_Example doesn't seem to be working

adam6806, 2019-02-28:
Included the serial output. I am using CloudMQTT. I have the 7000G version. I was able to successfully work through the LTE Demo sketch. I did have to add these two lines.

fona.setPreferredMode(38); // Use LTE only, not 2G
fona.setPreferredLTEMode(1); // Use LTE CAT-M only, not NB-IoT

I added them to the IoT Example sketch as well. I am noticing some off characters so possibly a baud rate issue? I really don't know. It just sits there at the end. I know the cell connection isn't established because the light is blinking at the slower speed.

*** SIMCom Module IoT Example ***
Configuring to 9600 baud
Attempting to open comm with ATs
	---> AT
	<--- 
	---> AT
	<--- 
	---> AT
	<--- 
	---> AT
	<--- 
	---> AT
	<--- 
	---> AT
	<--- 
	---> AT
	<--- 
	---> AT
	<--- 
	---> AT
	<--- 
	---> AT
	<--- 
	---> AT
	<--- 
	---> AT
	<--- AT
	---> ATE0
	<--- ATE0
	---> ATE0
	<--- OK
	---> ATI
	<--- SIM7000G R1529

OK

	---> AT+CPMS="SM","SM","SM"
	<--- +CPMS: 0,10,0,10,0,10
FONA is OK
Found SIM7000G (Global)
	---> AT+GSN
	<--- RRR⸮%5
⸮⸮⸮Module IoT Example ***
Configuring to 9600 baud
Attempting to open comm with ATs
	---> AT
	<--- OK
	---> ATE0
	<--- OK
	---> ATE0
	<--- OK
	---> ATI
	<--- SIM7000G R1529

OK

	---> AT+CPMS="SM","SM","SM"
	<--- +CPMS: 0,10,0,10,0,10
FONA is OK
Found SIM7000G (Global)
	---> AT+GSN
	<---⸮

adam6806, 2019-02-28:
This last time I ran it seems like each time it's trying to get the IMEI from the FONA library it is looping. fona.getIMEI(imei);

adam6806, 2019-02-28:
After looking at it some more and comparing the output to the lte demo the restart is occurring between these two lines. FONA is OK is never showing up. 

<--- +CPMS: 0,10,0,10,0,10
FONA is OK

adam6806, 2019-02-28:
Still can't seem to find the difference between the iot example and lte demo example.

wwkv, 2019-02-28:
I have the exact same issue, the LTE_demo works perfectly but the iot_demo stalls after the AT+GSN command...

wwkv, 2019-03-01:
I found that `Adafruit_MCP9808 tempsensor = Adafruit_MCP9808();` had an effect on the problem we have. When uncommenting this line (and all the lines involving the temperature sensor) the IMEI lookup was succesfull! Not quite sure **why** this is happening, maybe you know what's up? I'm happy though, only found this after hours of troubleshooting!

wwkv, 2019-03-01:
All right, I found it! I installed a previous version of the Adafruit MCP9808 library and now it's working properly!

adam6806, 2019-03-02:
Nice work @wwkv confirmed that does the trick for me as well. Your efforts are much appreciated :) @botletics can you please either update the documentation or implementation so that others don't fall into this issue please?

adam6806, 2019-03-03:
@wwkv which version of the library did you use? I wasn't able to get it working with the adafruit mqtt example on any version.

wwkv, 2019-03-03:
I used version 1.0.1. I'm still having issues too btw, the IMEI lookup works fine but if I enable the temperature sensor it doesn't want to connect to the MQTT. The only fix I now use temporarily is just uncomment all of the code that uses the MCP9808 library... I will look into it some more tomorrow.

botletics, 2019-03-04:
Try commenting out the temperature sleep and wake up lines.

closed 2019-06-10 by botletics