botletics/SIM7000-LTE-Shield #76: SIM7000 MQTT AT commands not working ??

mpablo135, 2019-03-03:
Hi,
According to SIMCOM "SIM7000_Series_MQTT_Application_notes" from here:
https://www.waveshare.com/w/upload/d/d0/SIM7000_Series_MQTT_Application_Note_V1.00.pdf
Also see the attached file.

There are already MQTT commands available for SIM7000 series modules.
Using these commands as opposed to the MQTT libraries will be much easier as shown in their example.
All of the commands related to MQTT return "ERROR" when i send them.
[SIM7000_Series_MQTT_Application_Note_V1.00.pdf](https://github.com/botletics/SIM7000-LTE-Shield/files/2923720/SIM7000_Series_MQTT_Application_Note_V1.00.pdf)

Why i am not able to use these with my shield?

Thanks

botletics, 2019-03-04:
They probably only implemented that on certain firmware versions.

adam6806, 2019-03-04:
@botletics where can we get the firmware files and update tool other than your github? I couldn't find them on simcom's site.

botletics, 2019-03-04:
Everything is on the tutorial: https://github.com/botletics/SIM7000-LTE-Shield/wiki/Updating-Firmware

adam6806, 2019-03-04:
@botletics that page just links back to the github. I specifically was asking about where can I find all of these files NOT here. For instance what if they update the firmware again and I don't want to wait for you to upload the files.

botletics, 2019-03-11:
Then you are free to go contact SIMCom and get it yourself, and they do not post them publicly.

SCWhite, 2019-03-13:
HI, I'm using sim7000E B07
module crash when "AT+SMCONN" is sent

RickGijsberts, 2019-05-09:
Probably the module holds an old, not gentle closed connection in memory. Closing this one with “AT+SMDISC” before (re)connecting solved this problem for me.

botletics, 2019-07-22:
Please see the lastest library files. I have included MQTT commands for SIM7000 as well as a SIM7000_MQTT_Example.

closed 2019-08-01 by botletics

Tasosgi, 2021-01-16:
Hello, which firmware of SIM7000E has the MQTT commands? I try the 1351B07SIM7000E and 1351B08SIM7000E without luck. The strange thing is that after download the new firmware the module become SIM7000A from SIM7000E.

thanks!

dberenguer, 2021-05-06:
Hi @Tasosgi, did you get to make the MQTT commands work on your SIM7000E modem? I'm also using firmware version 1351B08SIM7000E, which is the latest firmware for version TX2 of the SIM7000E. Simcom seems to be now providing support only for TX3.

Tasosgi, 2021-05-06:
I didn't have good lack with firmware update. I bought a new SIM7000E module which had support for mqtt. But at the end I use this library MaJerle/lwgsm and not the native MQTT support.

dberenguer, 2021-05-06:
I was finally able to make the MQTT commands work with firmware 1351B08SIM7000E. It's simply that I had to connect to the wireless network before trying even the most basic MQTT commands.