botletics/SIM7000-LTE-Shield #78: SIM7000G with T Mobile NB IoT ??

mpablo135, 2019-03-05:
I am planning to get a SIM7000G shield and  use with T Mobile Nb IoT network.
Has any body tested to verify that SIM7000G shield works with T Mobile NB IoT and with what brand simcard ?

Thanks

Koaug, 2019-05-25:
Check out https://www.twilio.com/. They are partnered with T-Moble for IoT solutions.

TryingToGITthis, 2019-06-13:
I just noticed my Hologram.io SIM is connected to T-Mobile via 2G.

robbiet480, 2020-03-19:
FYI, I bought a SIM7000G and tried to get it connected. It appears that T-Mobile no longer allows uncertified devices onto their network. At best, I got a 0,3 response to CGREG which means access explictly denied. I was using a Twilio NBIOT SIM.

Humancell, 2020-04-10:
Hello ... @robbiet480 I'd be curious to hear more about this?  I've got a SIM7000A here, newest firmware (AT+CGMR == Revision:1351B04SIM7000A) with a Twilio NB SIM, and was able to get a NB connection to work.

I use the AT Command Tester for SIMCOM, and got the correct settings from there:

```
RDY

+CFUN: 1

+CPIN: READY

SMS Ready
AT+CGMR
Revision:1351B04SIM7000A
OK

AT+CGSN
8652350300xxx59
OK

AT+CIMI
310240200xxxx36
OK

AT+CCLK?
+CCLK: "80/01/06,00:00:13-24"

OK

AT+CPIN?
+CPIN: READY
OK

AT+CGDCONT?
+CGDCONT: 1,"IPV4V6","iot.nb","0.0.0.0.0.0.0.0.0.0.0.0.0.0.0.0",0,0,0,0
+CGDCONT: 2,"IPV4V6","ims","0.0.0.0.0.0.0.0.0.0.0.0.0.0.0.0",0,0,0,0
+CGDCONT: 3,"IPV4V6","sos","0.0.0.0.0.0.0.0.0.0.0.0.0.0.0.0",0,0,0,1
OK

+CSQ: 99,99
OK

AT+IPR?
+IPR: 115200
OK

AT+COPS?
+COPS: 1
OK

AT+CREG?
+CREG: 0,0
OK

AT+CGACT?
+CGACT: 1,0
+CGACT: 2,0
+CGACT: 3,0
OK

AT+CGATT?
+CGATT: 0
OK

AT+CFUN?
+CFUN: 1
OK

AT+CSCA?
+CSCA: "+12063130004",145
OK

AT+CMGF?
+CMGF: 0
OK

AT+CPSI?
+CPSI: LTE NB-IOT,Online,310-260,0xF194,24795850,362,EUTRAN-BAND4,2391,0,0,-9,-69,-59,20
OK

System Mode:  LTE NB-IOT

Operation Mode: Online

Mobile Country Code: 310-260

Mobile Network Code: 0xF194

Location Area Code: 24795850

Cell ID: 362

RF Channel Number: EUTRAN-BAND4

C1 - Coefficient for base station selection: 2391

C2 - Coefficient for Cell re-selection: 0

TAC -Tracking Area Code: 0

SCellID - Service Cell ID: -9

PCellID - Physical Cell ID: -69

Frequency Band: -59

EARFN: 20

AT+CGMR
Revision:1351B04SIM7000A
OK

AT+CGSN
8652350300xxx59
OK

AT+CIMI
310240200xxxx36
OK

AT+CCLK?
+CCLK: "20/04/10,11:31:01-24"
OK

AT+CPIN?
+CPIN: READY
OK

AT+CGDCONT?
+CGDCONT: 1,"IPV4V6","iot.nb","0.0.0.0.0.0.0.0.0.0.0.0.0.0.0.0",0,0,0,0
+CGDCONT: 2,"IPV4V6","ims","0.0.0.0.0.0.0.0.0.0.0.0.0.0.0.0",0,0,0,0
+CGDCONT: 3,"IPV4V6","sos","0.0.0.0.0.0.0.0.0.0.0.0.0.0.0.0",0,0,0,1
OK

AT+CSQ
+CSQ: 23,99
OK

Signal level is -67 dbm. Signal condition is excellent.The signal strength range is -53 dbm (Excellent) to -109 dbm (Marginal).

AT+COPS?
+COPS: 1,0,"T-Mobile",9
OK

Device is currently on "T-Mobile" network.

AT+CGACT?
+CGACT: 1,1
+CGACT: 2,0
+CGACT: 3,0
OK

AT+CGPADDR= 1
+CGPADDR: 1,30.0.162.138,38.7.251.144.83.20.168.224.0.0.0.77.94.105.58.1
OK

IP Address of the connected profile is 30.0.162.138

AT+CGATT?
+CGATT: 1
OK

Device is attached to the network

AT+CFUN?
+CFUN: 1
OK

Device has Full functionality.

AT+CSCA?
+CSCA: "+12063130004",145
OK

SMS service center address is +12063130004

AT+CMGF?
+CMGF: 0
OK

SMS message for is configured for PDU mode

AT+CGREG?
+CGREG: 0,1
OK

The device is registered in home network.

Checking if device is already connected...

AT+CGACT?
+CGACT: 1,1
+CGACT: 2,0
+CGACT: 3,0
OK

Profile  1 with APN iot.nb is already connected
Downloading http://ipv4.ikoula.testdebit.info/1M.iso..

Uploading 1 MB file to http://ipv4.ikoula.testdebit.info/..

Getting Bearer profiles..

AT+SAPBR=4,1

+SAPBR:

CONTYPE: GPRS

APN: 

USER: 

PWD: 

OK
AT+SAPBR=4,2

+SAPBR:

CONTYPE: GPRS

APN: 

USER: 

PWD: 

OK
AT+SAPBR=4,3

+SAPBR:

CONTYPE: GPRS

APN: 

USER: 

PWD: 

OK
Checking registration status...

AT+CREG?

+CREG: 0,0

OK
The device is not registered and is currently NOT searching for a new operator to which to register.

AT+CGREG?

+CGREG: 0,1

OK
The device is registered in home network.

Querying bearer 1 .

AT+SAPBR=2,1

+SAPBR: 1,3,"0.0.0.0"

OK
Bearer  1 is Closed

Bearer 1 is Closed.

Opening Bearer 1...

AT+SAPBR=1,1

OK
Terminate any previous HTTP service...

AT+HTTPTERM

ERROR
Initializing HTTP service...

AT+HTTPINIT

OK
Setting up HTTP parameters..

AT+HTTPPARA="URL","http://httpbin.org/get"

OK
AT+HTTPPARA="CID",1

OK
AT+HTTPACTION=0

OK

HTTP GET is successfull

AT+HTTPREAD

+HTTPREAD: 256
{
  "args": {}, 
  "headers": {
    "Accept": "*/*", 
    "Host": "httpbin.org", 
    "User-Agent": "SIMCOM_MODULE", 
    "X-Amzn-Trace-Id": "Root=1-5e90add1-e9f99fbaf760b574d8dce6e0"
  }, 
  "origin": "172.58.159.84", 
  "url": "http://httpbin.org/get"
}

OK
AT+HTTPTERM

OK
```

robbiet480, 2020-04-10:
@Humancell Weird. I specifically contacted Twilio support and they advised me the 7000G wouldn't work because T-Mobile didn't allow uncertified devices on the network. I moved to using Hologram.

Humancell, 2020-04-10:
Do you know who you talked to in support?  (I work for Twilio ... ha ha)

Here is another test I just did:

```
Checking registration status...

AT+CREG?

+CREG: 0,0

OK
The device is not registered and is currently NOT searching for a new operator to which to register.

AT+CGREG?

+CGREG: 0,1

OK
The device is registered in home network.

Querying bearer 1 .

AT+SAPBR=2,1

+SAPBR: 1,1,"30.1.104.151"

OK
Bearer  1 is Connected.IP address is "30.1.104.151"

Bearer 1 is Connected.

Initializing HTTP service...

AT+HTTPINIT

OK
Setting up HTTP parameters..

AT+HTTPPARA="URL","http://nodered.mytest.com:1880/twilio/nb/test"

OK
AT+HTTPPARA="CID",1

OK
AT+HTTPDATA=21,10000

DOWNLOAD

OK
AT+HTTPACTION=1

OK

HTTP POST is successfull

AT+HTTPREAD

+HTTPREAD: 23
{ "status": "success" }
OK
AT+HTTPTERM

OK
```

robbiet480, 2020-04-10:
@Humancell Nikolay Rusinov, support ticket # 4012120. He advised the only SimCom module authorized was the 7020G.

robbiet480, 2020-04-10:
If you can figure out what went wrong, that would be great because I'd rather be using Twilio!

Humancell, 2020-04-10:
Hello ... yes, I believe that if you look at the "official" T-Mobile website, you'll see that there are only a few modules from SIMCOM that are truly "certified" ... but there is nothing related to the SIM that is stopping you from trying and seeing what works.

I know from working closely with SIMCOM they have the SIM7080G also in certification and I got two of them on line this week (with the latest firmware) for both Cat-M1 and NB. 

If you have your Twilio NB SIM, give it another try with the commands above (Or using the AT Command Tester for SIMCOM) but make sure you have the latest firmware.

robbiet480, 2020-04-10:
I'll try to give this a try later today. I know i'm on the latest firmware for the G, maybe that's the difference.

scootles7, 2020-09-28:
Hey, Robbie.  I'm tardy to this party, but did you ever get your SIM7000 working with the Twilio SIM card?

robbiet480, 2020-09-29:
Sorry, never got back to this.

HamzahVahed, 2024-07-12:
Does anyone have code for sim7000 with firebase connection?