botletics/SIM7000-LTE-Shield #79: Can't enable GPRS

noratx, 2019-03-06:
Hi!

I have tried the botletics Sim7000E shield now for a few days and I am quite happy about it.
However, i seem to be able to enable most of the available services, except for GPRS.
No matter what I do, it just won't work.
I have double checked with my operator and the APN I am using is correct.

// Set modem to full functionality
  fona.setFunctionality(1); // AT+CFUN=1
  fona.setNetworkSettings(F("internet.tele2.se")); // For Tele2/Comviq card
  fona.setPreferredMode(2); // Use GSM only since NB-IOT/LTE CAT-M is not available!
  fona.enableGPRS(false);

    //Enable GPS
    while(!fona.enableGPS(true))
    {
      Serial.println(F("Failed to turn on GPS, retrying..."));
      delay(2000); // Retry every 2 seconds
    }
    Serial.println(F("GPS turned on!"));
    delay(2000);
    while (!fona.enableGPRS(true))
    {
      Serial.println(F("Failed to enable GPRS, retrying..."));
      delay(2000); // Retry every 2 seconds!
    }
    Serial.println(F("Enabled GPRS"));

Gives error:

03:52:38.529 -> 	---> AT+CIPSHUT
03:52:38.598 -> 	<--- SHUT OK
03:52:38.633 -> 	---> AT+CGATT=1
03:52:38.666 -> 	<--- OK
03:52:38.734 -> 	---> AT+SAPBR=3,1,"CONTYPE","GPRS"
03:52:38.801 -> 	<--- OK
03:52:39.039 -> 	---> AT+SAPBR=3,1,"APN","internet.tele2.se"
03:52:39.143 -> 	<--- OK
03:52:39.244 -> 	---> AT+CSTT="internet.tele2.se"
03:52:49.555 -> 	<--- RZR
03:52:49.555 -> Failed to enable GPRS, retrying...

I don't want to put all of my code here (beause of it's length),
But it is pretty much the same as the examples (except I removed things that is not needed for me).

What could be wrong?

Best regards
/Rickard

botletics, 2019-03-11:
Perhaps your SIM card doesn't support 2G in addition to NB-IoT or LTE CAT-M?

Ghangus, 2019-06-02:
Did you ever find a solution to this issue? I have the same issue right now but with telia.

noratx, 2019-06-02:
Yes,

I found it was my LiPo battery that was faulty and needed to be replaced.

Ceck the battery condition and recharge/replace.

I took a battery from annold samsung phone and soldered some cables to the 
battery terminals.
So now I got a 3.7v 2600 MAh battery instead :)

However, I have more or less left the project due to my limited programming 
skills.

I cant make the code small enough :/
So I bought an adafruit fona 808 instead for my Rpi zero w, since I never 
managed to hook the botletics shield up with the Rpi. :)

/Rickard

noratx, 2019-06-02:
And oh, by the way!
It works with bothe Telia, Tele2/Comviq and Hallon/tre in sweden.

Atleast för 2G.
Telia does have support for NB-IoT, BUT, they ONLY sell NB-IoT simcards to 
companies :/

/Rickard

closed 2019-06-10 by botletics