botletics/SIM7000-LTE-Shield #81: SMS not sending and hologram.io sms text to device shows error

BusTrackerPSU, 2019-03-07:
Setup: Arduino Uno board with the Bothletics Sim7000a shield on top with a LiPo 3.7V 2000mah battery connected as well as the USB plugged into the UNO to the computer,

We are using LTE_Demo example code with a hologram sim card. 
Activated the sim card and It is connected Successfully to Verizon IoT network using the Hologram simcard, when on the serial monitor typing "s" to send a SMS message to a phone number in the US and we are located in the US. it says Failed.

We also went on the Hologram.io and send a SMS to the device and it just doesn't receive.
when we press "G"it says everything is okay APN is hologram everything is correct.

I saw that Verizon had a +88 code for outgoing SMS. not sure if its being blocked. Any Ideas?

I can give you more information if you need.

BusTrackerPSU, 2019-03-07:
We got it working after connection to AT&T so that is the issue.

Can we route the sms to use data only or will it charge as a sms as well?

botletics, 2019-03-11:
Yep you can route it: https://github.com/botletics/SIM7000-LTE-Shield/wiki/SMS-Messages#hologram-sms-route

botletics, 2019-03-11:
What you can do is send SMS to Hologram then have Hologram route it, either as SMS or other data form.

closed 2019-03-11 by botletics