botletics/SIM7000-LTE-Shield #83: Won't accept AT commands after reboot

spw1502, 2019-04-01:
I have a python program that works exactly as intended, it writes AT commands to the SIM7000E to send an SMS message under various conditions. My problem is that following a reboot of  raspberry pi (and thus the SIM7000E) the SIM7000E will no longer accept commands. The only way I can get the python program to run again is to attach to the serial port via 'minicom' issue a newline followed by an AT command. This seems to reset the SIM7000E and then the python works as designed. Have written a simpler python program that just issues a simple AT command. This has the same problem.
Ultimately this system is designed to be a remote monitoring system which alerts various conditions including power failure (by monitoring UPS). I intend the program to run as a daemon which will automatically restart on reboot. It's not going to work if I have to manually use minicom to get it to work first. 
Does anybody know how to programatically initialise the SIM7000 on power up without having to use minicom or as suggested elsewhere manually grounding a particular pin.
Thank-you

SCWhite, 2019-04-10:
I'm not sure what are you doing, maybe you forgot to release the port?
or the port change when you reboot?

ser=serial.Serial("COM16",57600,timeout=0.5)
==CODE==
ser.close()

spw1502, 2019-04-14:
Thanks, I am doing a close when program exits normally, but not sure this will get done when ups forces an FSSD following a power cut. However don't think this is the issue, if i go into a simple python shell and manually open serial port and write an 'AT' command to it , it won't work until I have opened serial port using 'minicom' and echoed AT command to it there. After that shell commands work as expected.
Doesn't matter what I do I cannot get SIM7000 to respond until after I have used minicom to write at least one command to it. Once that is done everthing else just works.

closed 2019-04-14 by spw1502