botletics/SIM7000-LTE-Shield #84: HTTPS GET or POST issue

Fredovsky, 2019-04-09:
I'm trying to GET or POST data from/to an URL.

When I use http:// it works, but if I put httpS:// it doesn't work.

I have to use https as it is forced on the server side (http gets redirected to https).

I did set `fona.setHTTPSRedirect(true);` in the code but doesn't make a difference.

Here is the output :

Test 1 ok :
`        ---> AT+HTTPTERM
        <--- OK
        ---> AT+HTTPINIT
        <--- OK
        ---> AT+HTTPPARA="CID",1
        <--- OK

        ---> AT+HTTPPARA="REDIR",1
        <--- OK
        ---> AT+HTTPPARA="URL","http://www.example.com"
        <--- OK
        ---> AT+HTTPACTION=0
        <--- OK
HTTP status: 200
Data length: 1270`

Test 2 failed : 
`
        ---> AT+HTTPREAD
        <--- +HTTPREAD: 1270
        ---> AT+HTTPTERM
        <--- OK
        ---> AT+HTTPTERM
        <--- ERROR
        ---> AT+HTTPINIT

        <--- OK
        ---> AT+HTTPPARA="CID",1
        <--- OK
        ---> AT+HTTPPARA="REDIR",1
        <--- OK
        ---> AT+HTTPPARA="URL","https://www.example.com"
        <--- OK
        ---> AT+HTTPACTION=0
        <--- OK
HTTP status: 603
Data length: 0
Failed to complete HTTP GET...
`

Any hints on the correct AT commands to use ?

botletics, 2019-04-15:
With HTTPS you have to use the SSL commands. Please see the [SIM7000 Documentation folder](https://github.com/botletics/SIM7000-LTE-Shield/tree/master/SIM7000%20Documentation/Technical%20Documents)

closed 2019-04-15 by botletics

jefflikesbagels, 2019-06-22:
I apologize in advance for my ignorance, as this is my first time programming a SIM7000. I am trying to perform the same task as Fredovsky using hologram.io. I created a new function in the Adafruit_FONA.cpp library and called it postDataHTTPS:

```
boolean Adafruit_FONA::postDataHTTPS(const char *request_type, const char *URL, const char *body, const char *token, uint32_t bodylen) {
  // NOTE: Need to open socket/enable GPRS before using this function
  // char auxStr[64];
  
    sendCheckReply(F("AT+GMR"), ok_reply, 10000);
	sendCheckReply(F("AT+CNACT=1,\"hologram\""), ok_reply, 10000);
	sendCheckReply(F("AT+CNACT?"), ok_reply, 10000);
	sendCheckReply(F("AT+CSSLCFG=\"convert\",2,\"hologram.cer\""), ok_reply, 10000);
	sendCheckReply(F("AT+SHSSL=1,\"hologram.cer\""), ok_reply, 10000);

	char urlBuff[strlen(URL) + 22];
	sprintf(urlBuff, "AT+SHCONF=\"URL\",\"%s\"", URL);
	if (! sendCheckReply(urlBuff, ok_reply, 10000))
		return false;
	
	sendCheckReply(F("AT+SHCONF=\"BODYLEN\",100"), ok_reply, 10000);
	sendCheckReply(F("AT+SHCONF=\"HEADERLEN\",100"), ok_reply, 10000);
	sendCheckReply(F("AT+SHCONN"), ok_reply, 10000);
	
	char dataBuff[strlen(body) + 22];
	sprintf(dataBuff, "AT+SHBOD=\"%s\",100", body);
	
	//if (! sendCheckReply(dataBuff, ok_reply, 10000))
	//	return false;
	sendCheckReply(dataBuff, ok_reply, 10000);
	//sendCheckReply(F("AT+SHBOD=\"TEST\",100"), ok_reply, 10000);
  
	sendCheckReply(F("AT+SHAHEAD=\"Content-Length\",\"120\""), ok_reply, 10000);
	sendCheckReply(F("AT+SHSTATE?"), ok_reply, 10000);
	sendCheckReply(F("AT+SHREQ=3"), ok_reply, 10000);
	sendCheckReply(F("AT+SHREAD=0,227"), ok_reply, 10000);
	sendCheckReply(F("AT+SHDISC"), ok_reply, 10000);
  
  return true;
}
```
I've been reading the SIM7000 documentation for the HTTPS commands, and I am struggling to get it working as I am sure I have mistakes somewhere. In my Arduino sketch I have the following:
```
        // Post data to website via 2G or LTE CAT-M/NB-IoT
        // Create char buffers for the floating point numbers for sprintf
        // Make sure these buffers are long enough for your request URL
        char URL[150];
        char body[100];
        char deviceID[] = "######";
        char tagID[] = "[\"_RESTAPI_\", \"WATER_LOW\"]";
        char message[] = "\"Water_Low\"";

        // POST request
        sprintf(URL, "https://dashboard.hologram.io/api/1/csr/rdm");
        sprintf(body, "{\"deviceid\": %s, \"tags\": %s, \"data\": %s}", deviceID, tagID, message);

        Serial.println(F("Attempting to perform HTTPS POST..."));
        Serial.print("URL: ");
        Serial.print(URL);
        Serial.println();
        Serial.print("Body: ");
        Serial.print(body);
        Serial.println();
        if (!fona.postDataHTTPS("POST", URL, body)){
          Serial.println(F("Failed to complete HTTPS POST!"));
        } else {
          Serial.println(F("Successfully performed HTTPS POST!"));
        }
```
When the sketch runs, I get the following on the serial monitor:
```
Attempting to perform HTTPS POST...
URL: https://dashboard.hologram.io/api/1/csr/rdm
Body: {"deviceid": ######, "tags": ["_RESTAPI_", "WATER_LOW"], "data": "Water_Low"}
	---> AT+GMR
	<--- Revision:1351B03SIM7000A
	---> AT+CNACT=1,"hologram"
	<--- ERROR
	---> AT+CNACT?
	<--- +CNACT: 1,"###.###.###.###"
	---> AT+CSSLCFG="convert",2,"hologram.cer"
	<--- ERROR
	---> AT+SHSSL=1,"hologram.cer"
	<--- OK
	---> AT+SHCONF="URL","https://dashboard.hologram.io/api/1/csr/rdm"
	<--- OK
	---> AT+SHCONF="BODYLEN",100
	<--- OK
	---> AT+SHCONF="HEADERLEN",100
	<--- OK
	---> AT+SHCONN
	<--- ERROR
	---> AT+SHBOD="{"deviceid": ######, "tags": ["_RESTAPI_", "WATER_LOW"], "data": "Water_Low"}",100
	<--- ERROR
	---> AT+SHAHEAD="Content-Length","120"
	<--- ERROR
	---> AT+SHSTATE?
	<--- +SHSTATE: 0
	---> AT+SHREQ=3
	<--- ERROR
	---> AT+SHREAD=0,227
	<--- ERROR
	---> AT+SHDISC
	<--- ERROR
```
I used the LTE_Demo example sketch as a building block, so I have all of the other associated code in place and working well to set up the SIM7000. It is also getting a proper IP address when I issue the AT+CNACT? command.

The first problem is the error on the AT+CSSLCFG command, I think that is preventing the AT+SHCONN and AT+SHBOD commands from working. I am also not sure how I should be handling the quotation marks and commas inside the body for the AT+SHBOD command, do I simply prefix them with a \?. Any ideas on what I could be doing wrong?

Thanks!

JGLJGL, 2019-12-20:
Hi Frederic, have you been able to solve the problem? If yes, could you send the code?

sethivansh6, 2020-07-16:
> With HTTPS you have to use the SSL commands. Please see the [SIM7000 Documentation folder](https://github.com/botletics/SIM7000-LTE-Shield/tree/master/SIM7000%20Documentation/Technical%20Documents)

Can you give us an example of how to send these AT commands to sim7000 using esp32 hardware serial?