botletics/SIM7000-LTE-Shield #87: can not connect mqtts with aws

slipiduche, 2019-04-17:
this is my result code:

```
Docklight Log File (ASCII) - Started 4/16/2019 22:02:19.85 

4/16/2019 22:02:45.61 [RX] - 00:22:57<CR><LF>
<CR><LF>
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)<CR><LF>
configsip: 0, SPIWP:0xee<CR><LF>
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00<CR><LF>
mode:DIO, clock div:1<CR><LF>
load:0x3fff0018,len:4<CR><LF>
load:0x3fff001c,len:1100<CR><LF>
load:0x40078000,len:10088<CR><LF>
load:0x40080400,len:6380<CR><LF>
entry 0x400806a4<CR><LF>
Encendiendo el GPRS...<CR><LF>
--->AT<CR><LF>
<SUB>.--->AT<CR><LF>
<SUB>.--->AT<CR><LF>
<SUB>.--->AT<CR><LF>
<SUB>.--->AT<CR><LF>
<SUB>.--->AT<CR><LF>
<SUB><---AT<CR>
<CR><LF>
OK<CR><LF>
--->AT<CR><LF>
<SUB><---AT<CR>
<CR><LF>
OK<CR><LF>
SIM encendido, AT OK!<CR><LF>
--->AT+GMR<CR>
<CR><LF>
<SUB><---AT+GMR<CR>
<CR><LF>
Revision:1529B01SIM7000G<CR><LF>
<CR><LF>
OK<CR><LF>
--->AT+CPIN?<CR>
<CR><LF>
<SUB><---AT+CPIN?<CR>
<CR><LF>
+CPIN: READY<CR><LF>
<CR><LF>
OK<CR><LF>
--->ATE0<CR>
<CR><LF>
<SUB><---ATE0<CR>
<CR><LF>
OK<CR><LF>
--->AT+CREG=0<CR>
<CR><LF>
<SUB><---<CR><LF>
OK<CR><LF>
--->AT+CFUN=1<CR>
<CR><LF>
<SUB><---<CR><LF>
OK<CR><LF>
--->AT+CLTS=1<CR>
<CR><LF>
<SUB><---<CR><LF>
OK<CR><LF>
--->AT+CMGF=1<CR>
<CR><LF>
<SUB><---<CR><LF>
OK<CR><LF>
--->AT+CNMI=0,0,0,0,0<CR>
<CR><LF>
<SUB><---<CR><LF>
OK<CR><LF>
[... 232 lines cut ...]
<CR><LF>
<SUB><---<CR><LF>
+CGATT: 1<CR><LF>
<CR><LF>
OK<CR><LF>
--->AT+SMCONN<CR>
<CR><LF>
<SUB><---<CR><LF>
ERROR<CR><LF>
<CR><LF>
--->AT+SMUNSUB="$aws/things/iotthing/shadow/update"<CR>
<CR><LF>
<SUB><---<CR><LF>
ERROR<CR><LF>
<CR><LF>
--->AT+SMDISC<CR>
<CR><LF>
<SUB><---<CR><LF>
ERROR<CR><LF>
<CR><LF>

Docklight Log File (ASCII) - Stopped 4/16/2019 22:05:06.56 

```
all is ok until SMCONN any suggeshelp?

brunokruse, 2019-05-05:
I notice you have the 1529B01SIM7000**G** version. The https + cert also fails for me on SHCONN using the 7000G version.

The AT Command logs from @botletics are very helpful, but seems like he has the 1351B05SIM7000**E**.

reydelleon, 2019-06-18:
Hey @slipiduche, did you figure this out? Could you connect to AWS over SLL?

botletics, 2020-12-09:
Hey guys, please see [this AWS AT command log](https://github.com/botletics/SIM7000-LTE-Shield/blob/master/SIM7000%20Documentation/AT%20Command%20Logs/SIM7000_AWS_Log.txt) with the 1529B0XSIM7000G firmware.

botletics, 2022-09-08:
Hey guys, figured out how to connect with SSL without verifying certs! Not sure if this would work with AWS though...
Please do the following:
- Set "SSL_FONA" to 1 in the .h file (remember to save the file before closing)
- Download the latest .cpp file from GitHub
- Open the unedited LTE_Demo example sketch and change "http://dweet.io" to "https://dweet.io" on line 1035 (under the '2' option)
- Upload to the Arduino
- Run the 'G' command to enable data connection, then run '2' to send data to dweet.io using SSL. And voila! 😊