botletics/SIM7000-LTE-Shield #88: Connection error

wwkv, 2019-04-19:
I have connection troubles after not using the module after a while. I tried everything the 'Connecting to network' page describes, but the module does not connect. 

This is what the module returns after the i, 1, n and G commands:

`FONA> 1
	---> AT+CPSI?
	<--- +CPSI: GSM,Online,206-10,0x5654,17922,34 EGSM 900,-82,0,24-24
	---> AT+COPS?
	<--- +COPS: 1,0,"Mobistar",0

OK
FONA> n
	---> AT+CGREG?
	<--- +CGREG: 0,2
Network status 2: Not registered (searching)
FONA> i
	---> AT+CSQ
	<--- +CSQ: 16,99
RSSI = 16: -82 dBm
FONA> G
	---> AT+CIPSHUT
	<--- SHUT OK
	---> AT+CGATT=1
	<--- ERROR
Failed to turn on
FONA> `

What could be the error?

botletics, 2019-04-19:
Hmm. Try disabling data with the "g" command first.

wwkv, 2019-04-22:
Disabling data also gives an error message: 'Failed to turn off'...

botletics, 2019-04-22:
Make sure your SIM card supports data with 2G.

wwkv, 2019-04-22:
The SIM card does support 2G. I forgot to mention I used the module for several days without any problems. After not using it for a week I'm now not able to get it working again.

botletics, 2019-04-22:
Hmm that is strange. I'm not entirely sure why ;(

wwkv, 2019-04-23:
As you probably can understand, I would like to have a working module. Is there a possibility to get a spare one or something?

wwkv, 2019-04-23:
I contacted my network provider and they fixed the issue, they didn't say what is was though. Thanks for helping!

closed 2019-04-23 by wwkv