botletics/SIM7000-LTE-Shield #89: Couldn't find FONA

mjam9814, 2019-04-24:
Hey,

So I tried running the LTE Demo Sketch, it compiles properly but when i run the serial monitor, its just outputting random characters and at the end it say 'Couldn't find FONA' as shown in the screenshots below.

[image]

[image]

I'm using the botletics shield (SIM7000G)  with the arduino uno and the hologram simcard (which has been activaed). All the LEDS(the two greens and the blinking blue) are on. I've followed each step so far in the instructables and I've downloaded all the libraries so I'm not sure where I've gone wrong.

botletics, 2019-04-27:
Please make sure you have the proper baud rate set in the serial monitor (9600) and don't change the LTE_Demo code in any way. It should work.

mjam9814, 2019-04-27:
The error shown in the screenshots happened when I used the original code and didn't change anything. So I guess I'll try to improve the solder joints and will let you know how it goes.

mjam9814, 2019-04-29:
So I tried to improve the solder joints as much as I could but it doesn't fix the problem. I have two of your shields (SIM7000G (v6) and SIM7000E (v5)) and the same thing happens to both. Could it be because of the Hologram SIM? I live in Sydney if that makes any difference.

botletics, 2019-04-29:
Do you see the green "PWR" LED turn on?

Please also check if your LiPo battery polarity is correct. The red wire should be on the left side of the connector when you plug it into the board.

mjam9814, 2019-04-29:
Yup, the LED turns on and the battery is in the correct polarity.

[image]

botletics, 2019-04-29:
Try unplugging the USB cable connected to the shield. It might be that you're trying to communicate with it via that USB at the same time the Arduino is trying to talk to it via UART.

mjam9814, 2019-04-29:
I’ve tried that, but still, its giving me the same error

botletics, 2019-04-29:
Hmm. Maybe something is wrong with your Arduino? I've never heard of this issue before.

mjam9814, 2019-04-29:
Yeah i was just trying that out to see if it worked and it did. Using Putty, it seems to responds to the commands

botletics, 2019-04-29:
Then the issue is likely with the solder joints on pins 10 and 11 or the Arduino itself, or a loose fit of the header pins into the Arduino headers.

mjam9814, 2019-04-29:
Ok then. I'll try to borrow another arduino from a friend and see it if that works. If not then its probably the connections. Do you happen to sell the shields with the headers already soldered on by any chance?

botletics, 2019-04-29:
Unfortunately no, they all come unsoldered since some people wire it up with other things and prefer to customize it.

mjam9814, 2019-04-30:
Am I still able to wire it externally to try and get it to work given that I've already soldered the female headers on?

mjam9814, 2019-05-02:
Used a different arduino and got it to work. Thanks!

closed 2019-05-02 by mjam9814